revk/xESP32-Faikout #255: [BUG] System is off line on FTXF60D2V1B

keesvanbemmel, 2024-03-02:
Hi!

I ordered a faikin board from Amazon early July '23 (pictures attached) and just got around to installing it in my FTXF60D. I attached it using a self created s21 cable but also with an original one. Both result in the same behavior: the module setup is fine, mqtt and ssid work perfectly, but "System is off line". I also tried an original wifi module on the s21 port and that works as expected.

Is there anyway to debug what is wrong? Coincidentally I will receive 4 new modules this Monday which I ordered for my other units. Perhaps those are new hardware revisions that work with my unit? I read some units expect 5v signals and older revisions of the faikin board do 3.3?

Perhaps there is anything else I could try or this is a known issue with an easy fix?

Cheers!
[image]
[image]

revk, 2024-03-02:
That looks like the older boards which work on units that are happy with 3.3V, but not all are (both of mine here were fine). The newer boards work by pulling down against a pull up to 5V in the air-con. So yes, very likely to be the issue.

keesvanbemmel, 2024-03-02:
Ok I'll wait for the new boards then and swap one out, ok if I keep this open until then?

keesvanbemmel, 2024-03-04:
Just received them, before I test them: in the new Basic Setting UI, the link to the timezones format page seems missing. What format should I set in the timezone?

closed 2024-03-04 by keesvanbemmel

revk, 2024-03-05:
I'll check the timezone. The link should be there, weird.