revk/xESP32-Faikout #256: Controlling temperature does not work as expected (too hot, too many on/off cycles)

klausj1, 2024-03-02:
**Faikin hardware**
Purchased from Amazon in december, 2023
Not sure if this helps:  Faikin-S3-MINI-N4-R2: 4945ab0 2024-02-28T15:54:01 S21
External temperature sensor is used, measuring the temperatur in the middle of the room, not directly at the Daikin

**Daikin hardware**
Perfera, connected via S21

**Describe the bug**
I set the setValue of the room temperature to 19.5 °C and turn on the heating. Starting temperature is ~18°C
Heating starts, room temperature goes up to almost 22 °C (the orange curve is the room temperature)
[image]

Near the end, the temperature rises faster, probably because I was in the room. Heating turns off, and after a short time heating turns on again, although the room temperature is still far higher than 19.5 °C. And then the heat turns off and on again, starting to cycle. (This figure is just zoomed in at the end of the first one)
[image]

Not sure if this is really a bug, or some settings are wrong. It would be great if you can give some advise.

**To Reproduce**
I cannot really provide steps to reproduce. For me this always happens, when I do not turn off the heat pump manually. At least when its not that cold outside.

**Expected behavior**
I would expect that the heat pump turns off at a room temperature little more than 19.5 °C, and then it turns on again  at a room temperature little less than 19.5 °C

**Additional context**
I connected the Faikin to my home assistant. The graphs are from there.
Outside temperature was around 10°C, and the heatpump (inner and outer unit) used ~140W power (so probably already running at the lower limit of the inverter)

Auto-mode is disabled.

Faikin.local:
[image]

Settings which I think could have an influence:
[image]

revk, 2024-03-02:
There are a lot of settings you can play with.

Here I have my air on within a tight window a fit has small on off cycles but stays in then window.

Try some of the settings to adjust the way it works, maybe the thermostat mode will be closer to what you want.

klausj1, 2024-03-02:
Thanks for the super-fast answer.

I did not use the auto-mode, as I read that it automatically heats and cools as needed, and I wanted to define on my own to only heat (and do not cool). OK, if the auto-mode needs to be enabled to have the faikin do the controlling, then I will change that and see what happens. I'll test and report.

And the behaviour when using the Daikin remote without faikin was pretty bad, maybe mainly because of the missing external temperatur sensor, maybe also because of the controller.

revk, 2024-03-02:
That is also a setting now to swap modes. So just heat.

klausj1, 2024-03-05:
Thx, I changed to Auto-Mode and realized that the Faikin sets the temperature setValue of the Daikin.

When I use a setValue of 19.5 degrees, the faikin sets the setValue to 28. I then lowered the setting `heatover` to 0, then the Faikin used 22 degrees. (Thats a difference of 6, fine, still I do not understand how this is calculated, but this does not hurt ;))

Unfortunately, this still results just in an on/off-mode (like In understood the thermostat mode), not using the inverter functionality to lower the power instead. But (I think, I could not really test because its so warm currently that it does not make sense to heat)) at least the setValue of 19.5 degrees is reached in this way. If you look at my first graph above, the Daikin controller lowers the frequency, but it still heats up to 22 degrees. So I would need to send a setValue lower than 19.5 degrees to the Daikin. I tried to set the `heatover` to -2, but less than zero is not possible.

Do I have any chance to use the inverter functionality, and therefore have less On/Off-cycles? (=  lower the compressor output so you can run it longer at lower duty?) 

Are there any other settings I can play with? What is the "demand" I can set to somewhere between 0 and 100?

What do you mean with "That is also a setting now to swap modes. So just heat."? Can I use the auto mode, but still say "heating only, do not cool"?

Wouldn't it make sense to use a PID-controller as discussed  [here](https://github.com/revk/ESP32-Faikin/discussions/177) directly in the Faikin code?

revk, 2024-03-06:
If you know of a way of either telling the Daikin the reference temperature, or telling it to change the level of heating/cooling it does, let me know. As far as I know the only real control I have is the temperature set value.

It is adjusted using `heatover`, etc, but also considering the difference between what the Daikin thinks the room temperature is at the moment and whatever we are using as a reference, i.e. an offset for how wrong it is. That alone may do what you want if you set `heatover`, etc to `0`.

I meant "not to swap modes" not "now to swap modes", sorry. So you can set it to heating and turn on Faikin auto and it won't change to cooling. The setting is `lockmode`.

PID would make sense if not fighting the internal logic within the Daiklin. Bye trying to finely adjust the target temp to get what I want, it would cycle with much more temperature variance than I wanted because of the inbuilt hysteresis and other functions within the Daikin. As such it works as simply on/off (by setting much higher or lower target), and for that you have fewer options. PID normally assumes you have finer control than just on/off. The result is, for me, a much higher control of temperature. But the settings allow you to change a lot of things if you want.

closed 2024-03-06 by revk

klausj1, 2024-03-07:
Don't get me wrong, I really appreciate your work, and my second Faikin arrived this week.

Thx for the hint to lockmode. There were too many "modes", so I did not understand what it means before. Now its working, and it keeps the temperature. (I did already set heatover to 0)

However, there the Daikin goes On/Off much too often:

[image]

Any tipps? Maybe I'll try the thermostat mode, but I did not understand when it turns off and on, is there a hysteresis?

revk, 2024-03-07:
The thermostat mode does do simple hysteresis, but if you have set everything to 0 it should just be keeping the target temp pretty stable and leaving it to the Daikin to decide when to go on and off.

Dave-Baldwin, 2024-04-06:
Hi, I'm a bit puzzled by the behaviour that I'm seeing, and I think it relates to the above.
I have a simple scheduler in my HA system sending time-based setpoints to Faikin. What I see is that the temperature setpoint is never achieved. Daikin begins short heating cycles about 6 degF above the setpoint.
[image]

This is a bit baffling. When I look at the S21 debug information in MQTT, I do see a parameter (maybe Sx, don't remember at the moment) that when read backwards did seem to indicate a higher 'effective' setpoint may have been reported by Daikin. I was changing as I changed setpoints provided to Faikin via MQTT. I do think I've seen discussion elsewhere here about a bias/hysteresis that Daikin comes up with on its own in some circumstances? From a glance at the Faikin code I don't see any place where biases are added to the provided setpoint other than heatover, coolover, heatback, coolback, and I have set all of those parameters to zero.

When Daikin is heating, it also heats to about 3 degF above the setpoint provided to Faikin via MQTT. Currently I am compensating for this by adjusting the scheduled setpoint issued to Faikin. This is not a big deal (although still confusing).

Do you think it would be possible to mitigate this undesirable "heating resumed too soon" behaviour by setting coolover or coolback to non-zero values?

Thanks for your help. Overall this is a great system. But I think some of the advanced features could benefit from a bit more thorough documentation.

revk, 2024-04-07:
There is a thermostat setting which makes it do hysteresis. Try that maybe.

Dave-Baldwin, 2024-04-07:
Isn't that thermostat setting on the Faikin side? 
Do you believe in my situation above that Faikin is issuing different setpoints to Daikin than it is receiving via MQTT?

revk, 2024-04-07:
I am writing this up more in the Advanced.md now, so give me a minute and it should be clearer how it works.

revk, 2024-04-07:
See https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Advanced.md

schup011, 2024-05-16:
I would like to add to this discussion here that it is actually possible to regulate the heating or cooling power of the Daikin devices by using the difference (SetPoint - InternalTemp). 

For heating, you can achieve the lowest power by setting this to -3K (SetPoint 3 degrees lower than internally measured temperature). This usually has an impact on the inverter setting of the external device and on the fan setting (if set to auto). If you set to -4K, the heating switches off. Highest power is +1K or +2K, above that it stays maximum.

This regulated power can help quite a lot for not only achieving a well-regulated temperature, but also minimum power consumption and wear of the external device because you avoid frequent on-off switching.

I got this information from the German akkudoktor forum, there are people who have set up a quite smooth control based on this behavior, by using the Onecta API (or local API of the old Wifi controllers built into the Stylish devices). However, this is still rule-based and not the classic PID. I myself am trying to achieve a real PID. One big problem are that the measured internal temperature is only reported in 1K steps, and at a set difference of -3K, once the internally measured temperature toggles by +1K, you are not fast enough moving the SetPoint setting accordingly. Thus, heating can switch off unwantedly at that stage.  Maybe, this is better and fast enough by using the S21 protocol. Anyway, with a minimum power setting of "-2K", it should work. 

Could this "maintain -3K difference" be tested with a heatback=3 parameter? I myself don't have a Faikin installed, but I am considering it :-).

revk, 2024-05-16:
I have documented the options now in a lot of detail, so I suspect you can do what you want, but do read though the documentation and check.

schup011, 2024-05-16:
As I don't have a Faikin installed myself so far, I cannot do that right now. 

Just wanted to share this information about how (potentially) to regulate the power of the internal unit. 

If one of the existing Faikin users is interested to test this, this could later be used for implementing a thermostat that regulates instead of switching on and off. Honestly, if this works, that would be a huge motivation for me to set up Faikins for my 4 units as well.

klausj1, 2024-05-17:
@schup011 , you can definitely turn off all the automatic controlling of the Faikin. When NOT using the auto-mode, you can set the temperature setValue e.g. in a Home Assistant pyscript as you want (in steps of 0.5 °C). With the Faikin, you get the temperature measured by the internal Daikin sensor with 0.1 degrees resolution.
I am using this to control my Daikin, and I am adjusting the temperature setpoint automatically depending on the difference between the target temperature and the measured temperature (in the room via BLE, not inside the Daikin). It does not work perfectly, but its somehow OK.

Does this answer your question whether what you want to achieve is possible with the Faikin? I would say it definitely is possible, and I would be very interested in a "real" PID controlling algorithm.

Can you please share the link to what you found in the Akkudoctor? Are the values for cooling the same?

schup011, 2024-05-17:
> @schup011 , you can definitely turn off all the automatic controlling of the Faikin. When NOT using the auto-mode, you can set the temperature setValue e.g. in a Home Assistant pyscript as you want [...]

Thanks, so this definitely answers my question. If the resolution of the internal Daikin temperature is 0.1 °C, it should be possible to achieve a better control, it won't really matter whether this control runs on the Faikin or on the home automation (in my case openHAB).

I need to search in the Akkudoktor forum myself and will then post some links.

schup011, 2024-05-17:
Here is one link:

https://www.akkudoktor.net/forum/heizungssysteme/daikin-perfera-regelung-diskussion/paged/14/#post-188959

klausj1, 2024-05-17:
> Here is one link:
> 
> https://www.akkudoktor.net/forum/heizungssysteme/daikin-perfera-regelung-diskussion/paged/14/#post-188959
> 
> There are several posts by the same user, but this post, to my knowledge, describes his latest status. One addition: in a PM, he confirmed that maintaining the -3K temp difference for lowest power is difficult because the reported internal temperature toggles by 1K. With Faikin, this could be much better.
> 
> Doesn't Faikin also report the actual internal fan RPM? I saw it on screenshots, but I haven't found a documentation on that.

Thanks. Yes, Faikin reports the fan RPM, I am using it in home assistant.