revk/xESP32-Faikout #257: [BUG]

MUN0X, 2024-03-02:
**Faikin hardware**
Purchased from amazon

**Daikin hardware**
FVXM connected via s21

**Describe the bug**
When selecting the “night” fan speed, the unit correctly lowers the fan speed. However after about 10s the button (both web and mqtt) reverts back to auto. The fan speeds remains low though so its most likely a reporting problem and the night setting is remains active.

**To Reproduce**
Steps to reproduce the behavior:
Select the night fan setting coming from any other fan setting.

**Expected behavior**
That the night setting would remain (visually) active.

**Additional context**
Add any other context about the problem here.

revk, 2024-03-03:
I thought I had documented this.

The S21 messages do not report a difference between Night and Auto, so we have to guess (based on reported as Auto and what we last set and fan speed).

If S21 reported Night mode we could know for sure.

But guessing is the best that can be done, unless someone finds some other S21 message/response that provides a clue.

MUN0X, 2024-03-03:
Oh, i missed this in earlier discussions.
But why does it guess wrong after i set night mode via faikin? 
I only have this issue with met FVXM units not with the newer FTXM (perfera). Could it be that the FVXM report back something else than auto?

revk, 2024-03-03:
It looks at fan speed as a clue. So that is probably why, night mode is usually a low fan speed.

closed 2024-03-10 by revk

revk, 2024-08-13:
We may have found an S21 message, try the latest beta and report back.