revk/xESP32-Faikout #258: [FEATURE] Separate Min/Max setting for heating and cooling

MUN0X, 2024-03-03:
**Daikin hardware**
Faikin purchased from amazon connected via S21

**Is your feature request related to a problem? Please describe.**
Yes, the fact that you can only set one min/max in the faikin settings interferes with the faikin auto mode.
My units have valid setpoint range of 10-30°C when in heating mode and 18-32°C when in cooling mode.
In heating mode the perfera model (FTXM-R) will heat to a temperature of 4°C above the desired setpoint. 

This means that I have to apply a heatback-value of at least 4°C, preferably more like 6°C.
So for heating the room to about 20°C, I need to be able to set a minimum value of at least ~14°C.
However, in cooling mode the minimum setpunt that is accepted by daiking is much higher at 18°C.

In cooling  mode, the automode of faikin keeps trying to set values of like 14-15-16°C (already lowered coolover to 2°C).  
This is value is reverted back to 18°C within 5s and faikin keeps trying over and over to set a lower value.

So if I set the valid min/max range (18/32) for cooling my unit doesnt stop heating.
If I set the valid range for heating (10-30) faikin keeps setting invalid temperatures. 

**Describe the solution you'd like**
Create 2 min/max sets in the settings. One for heating and one for cooling just like you do for the heatover/heatback and coolover/coolback combinations.

**Describe alternatives you've considered**
Using home assistant to check and change the min/max to the appropriate values whenever the heating or cooling mode is triggered.
However, these values are rather static. So setting them in the faikin settings makes more sense than adjusting them every time the unit switches operation mode.

**Additional context**
/

revk, 2024-03-03:
Surely one min/max that covers the whole range will work?

MUN0X, 2024-03-04:
It doesn’t. Thats what I tried to explain above.
Setting min/max that covers the entire range will cause faikin to issue invalid setpoints every 5s in cooling mode

revk, 2024-03-04:
OK sorry, I looked at this and it is a nuisance to make the various bits work with changing min/max.

But just limiting the set points may be relatively simple. I'll see what I can come up with.

revk, 2024-03-04:
OK maybe if I add a `tcoolmin` and a `theatmax` as alternative for the set point control logic?

That should do it, yes?

revk, 2024-03-04:
I have made a beta release with that, let me know if it covers what you need.

closed 2024-03-04 by revk

MUN0X, 2024-03-04:
Just tested the beta and it appears to work as you intended. The automode no longer goes bonkers. I understand that it is a compromise to avoid a whole overhaul of your code. 

When documenting these settings it might be worth noting that these settings do not limit the setpoint a user can enter. So a user can request a temperature < tcoolmin but the unit will never be able to reach that.

Thank you for implementing this!

revk, 2024-03-05:
I have said it is for daikin auto mode in the help.