revk/xESP32-Faikout #263: I have bougt two units but one of them worked 2 days the other one works[BUG]

sjaakie09, 2024-03-10:
**Faikin hardware**
Please advise if this is a Faikin you purchased (e.g. from Amazon), had built, or some other device.
yes form amazone

Ideally, please include picture of both sides of the board itself as this will help confirm the version.

**Daikin hardware**
Please advise the model of your Daikin, and how you are connected (e.g. S21 lead, X50 lead, etc).
perfora daikin 
**Describe the bug**
it wil not come online 
you can set up de unit but you dont get hem online?

A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

revk, 2024-03-11:
Have you swapped them around?

This should confirm if the problem is the hardware Faikin module or not.

Also swapping leads to work out if it is one of the leads.

If one of the Faikin modules is not working, let me know Amazon order number and date and I should be able to sort a refund.

sjaakie09, 2024-03-11:
hello Ill send you amazone nr did't you saw it?
her i send it again

206-4331075-7289114 

see below

<html xmlns:v="urn:schemas-microsoft-com:vml"<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/smart/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<link rel=Edit-Time-Data
href="file:///C:/Users/smart/AppData/Local/Temp/msohtmlclip1/01/clip_editdata.mso">

<link rel=themeData
href="file:///C:/Users/smart/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/smart/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=NL link="#007185" vlink="#96607D" style='tab-interval:35.4pt;
word-wrap:break-word'>

Your order will be sent to:    Mark   Dale-lace /J Duerinck Mobexx   WHITCHURCH   United   Kingdom
--
 
Your   delivery option:    Premium   Delivery
 
PO   no.    multra   temp
 
Order   #    206-4331075-7289114
 
 

</body>

</html>

xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:w="urn:schemas-microsoft-com:office:word"
xmlns:m="http://schemas.microsoft.com/office/2004/12/omml"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Word.Document>
<meta name=Generator content="Microsoft Word 15">
<meta name=Originator content="Microsoft Word 15">
<link rel=File-List
href="file:///C:/Users/smart/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<link rel=Edit-Time-Data
href="file:///C:/Users/smart/AppData/Local/Temp/msohtmlclip1/01/clip_editdata.mso">

<link rel=themeData
href="file:///C:/Users/smart/AppData/Local/Temp/msohtmlclip1/01/clip_themedata.thmx">
<link rel=colorSchemeMapping
href="file:///C:/Users/smart/AppData/Local/Temp/msohtmlclip1/01/clip_colorschememapping.xml">

<style>

</style>

</head>

<body lang=NL link="#007185" vlink="#96607D" style='tab-interval:35.4pt;
word-wrap:break-word'>

Your order will be sent to:    Mark   Dale-lace /J Duerinck Mobexx   WHITCHURCH   United   Kingdom
--
 
Your   delivery option:    Premium   Delivery
 
PO   no.    multra   temp
 
Order   #    206-4331075-7289114
 
 

</body>

</html>
Your order will be sent to: 
Mark Dale-lace /J Duerinck Mobexx
WHITCHURCH
United Kingdom

Your delivery option: 
Premium Delivery 

PO no. 
multra temp 

Order # 
[206-4331075-7289114](https://www.amazon.co.uk/gp/f.html?C=139N1T2SKF371&K=1I5FBAK48PG90&M=urn:rtn:msg:202401252032246e0c3a0c920a4c0c88d205920860p0eu&R=3PWRNA0297L5W&T=C&U=https%3A%2F%2Fbusiness.amazon.co.uk%2Fabredir%2Fgp%2Fcss%2Fsummary%2Fedit.html%3Fie%3DUTF8%26orderID%3D206-4331075-7289114%26ref_%3Dpe_27063361_485629781_TE_on_sh&H=664AKQZOP6D4EMWDSCIUQB7CUI4A&ref_=pe_27063361_485629781_TE_on_sh)

revk, 2024-03-11:
It looks like you post to two issues.

Anyway, as I said, refunded.

closed 2024-03-11 by revk

sjaakie09, 2024-03-11:
Beste, Dear,

ok top thank you!

Met vriendelijke groet / With kind regards,
***@***.***<https://comm-co.com/jm2/>
Jacques Duerinck CEO
***@***.******@***.***>

Comm-Co B.V.
Kreekzoom 9
4561 GX  Hulst
The Netherlands
Tel: +31 114 370030
GSM: +31 653 120 225
***@***.******@***.***<https://ewheelz.nl/>[created with MySignature.io]<https://twitter.com/atexshop>[created with ***@***.***<https://nl.linkedin.com/in/jacques-duerinck?challengeId=AQHAi-Fs0-Lh1wAAAYvdylK2ok7NzA6irP0R6i52tF08AkW-uyQBCxRVPtwfspwwReWjC6m5WFdCVpc4-Tuff_-GepJXMa6EQQ&submissionId=abb2045a-0071-9817-4aea-279bfae4233d&challengeSource=AgHJcnaeiOppRQAAAYvdyuWbNO7oSDhbmuPQZCGvfX6Uxe5Sc8Mdb7a4KhBRlzM&challegeType=AgH7_Hj09W8F_QAAAYvdyuWers9w2E30FsryCkBCK1SatbG_yKhyvXU&memberId=AgECOEUb0_Rl1AAAAYvdyuWgmmRlgfXI-eqOta-QLfeZUw0&recognizeDevice=AgEl9duzUhSpagAAAYvdyuWjz6GgpYneL-OCQDE_uC4-rxQ0V2WB>