revk/xESP32-Faikout #271: JST-EH connector does not fit on S21 on Emura 3 board

migueldc73, 2024-03-22:
So, I bought the Faikin module, got it working, but can't seem to find a way to connect it to my S21 connector on an Emura 3
I bought a few JST-EH connectors based on the instructions

> The S21 plug needed is [JST, EH Female Connector Housing, 2.5mm Pitch, 5 Way, 1 Row](https://uk.rs-online.com/web/p/wire-housings-plugs/3116237), which needs pins and crimps as well.

The JST-EH is perfect for the Faikin side, but not for the S21 connector on the Emura board, which is clearly smaller than the JST-EH, with a pitch of probably 1.5 or 2mm. 
[image]

As I can't find any EKRS21 cables in stock for shipping less than 4-5 weeks in Belgium/Netherlands, I am desperate to find the right connector model nr, so I can order it and make my own cable.

revk, 2024-03-22:
Probably better in the discussions section.

migueldc73, 2024-03-22:
Thanks, will close this issue as I have moved it to the Discussions Q&A

tuomassalo, 2024-03-26:
Link to the discussion: https://github.com/revk/ESP32-Faikin/discussions/272

closed 2024-05-29 by revk