revk/xESP32-Faikout #274: Mqtt setup homebridge

Cptmeme, 2024-03-24:
First of all I would like to start by saying that my faikin board has been working great and would like to thank RevK for this awesome project.

I have had a great working system with homeassistant and my faikin board for the last couple of months now, however I dont have the time anymore to keep up my homeassistant system and so it keeps breaking making me want to move to homebridge as that generally requires less maintenance (afaik).

But when it comes to setting up my faikin through mqtt I found that I have no idea how to set it up with the mqtt commands and while I know that they are in the documentation I really dont understand what I should fill-in in each spot. 

I know that this is not a bug but rather a request for help but I didnt know where the proper place to ask was.

Could anyone please help me set up my mqtt settings for the faikin board? 
It would be much appreciated!

revk, 2024-03-25:
I've documented the MQTT. This may be better as a discussion point to see what other people have done.

njh, 2024-03-26:
I am currently using HomeKit with Faikin via [Homebridge](https://github.com/homebridge/homebridge) + [homebridge-daikin-local](https://github.com/cbrandlehner/homebridge-daikin-local). It works reasonably well but it is a shame that Apple don't support the concept of setting a fan speed to 'Auto'.

I did try using [MQTT-Thing](https://github.com/arachnetech/homebridge-mqttthing) with Faikin and had some issued but I should try again with a recent firmware.

See also #91 and #93 (matter support).

44616E, 2024-03-30:
I'm controlling mine in Node-red so unsure if that will be any help. Must admit I found the documentation alittle confusing at first, some how I didn't find the examples which would have made it alot more obvious.

njh, 2024-03-31:
Please add more examples to the Wiki page, if you think there are examples that would help other users:
https://github.com/revk/ESP32-Faikin/wiki/Example-MQTT-commands

I think the thing that confuses people is that the Faikin MQTT topic structure does not match the @mqtt-smarthome proposal:
https://github.com/mqtt-smarthome/mqtt-smarthome/blob/master/Architecture.md

Which lots of other systems have followed.

44616E, 2024-03-31:
Hard thing to change and support backwards compatibility I suppose unless its an option.

revk, 2024-03-31:
It would have to be a setting, yes.

jat80, 2024-05-03:
> I am currently using HomeKit with Faikin via [Homebridge](https://github.com/homebridge/homebridge) + [homebridge-daikin-local](https://github.com/cbrandlehner/homebridge-daikin-local). It works reasonably well but it is a shame that Apple don't support the concept of setting a fan speed to 'Auto'.
> 
> 
> 
> I did try using [MQTT-Thing](https://github.com/arachnetech/homebridge-mqttthing) with Faikin and had some issued but I should try again with a recent firmware.
> 
> 
> 
> See also #91 (discussion about HomeKit) and #93 (matter support).
> 
> 
> 
> 

I'm looking at buying the board from amazon to integrate with homekit via homebridge.  Just wondering if there is anything specific with using this board with the Daikin homebridge plugin that I need to know or does the wifi board work just like the daikin branded one?

revk, 2024-05-06:
I think this needs to be a RevK library issue.

closed 2024-05-06 by revk

revk, 2024-05-06:
Library update in latest beta, new setting, please try.