revk/xESP32-Faikout #275: [FEATURE]  Report "hvac_action" for HA Climate entity in faikin-auto

MUN0X, 2024-03-24:
With the faikin in auto-mode it currently just reports the state that it is set to (off, auto, heat, cool, ..) to the HA climate entity. However, it is unclear as to what the daikin is doing in the background (active or stepped back) which would be extremely useful for automations.

Could you please implement the `hvac_action` within the climate entity? (cfr. [link](https://developers.home-assistant.io/docs/core/entity/climate/))
Basically report `hvac_action` as:
"off" = daikin switched off
idle = daikin stepped back
heating = daikin target temperature raised by faikin-auto in heat or auto mode
cooling = daikin target temperature lowered by faikin-auto in cool or auto mode

revk, 2024-03-25:
Ok I have made a beta, see if it does what you need. It only does for Faikin auto mode logic.

closed 2024-03-25 by revk

MUN0X, 2024-03-25:
I updated to the beta but is doesn't seem to work as intented.
I can see the action coming in via mqtt but it doesn't end up within the HA Climate entity.
Could this be due to a slight misconfiguration of the mqtt-config topic?
`hvac_action_stat_t` ==> `action_topic`
`hvac_action_stat_tpl` ==> `action_template`

Cfr [`climate.mqtt`](https://www.home-assistant.io/integrations/climate.mqtt/)

Not sure if it matters but after reading the documentation I suspect that the actions should be passed in lower case.

MUN0X, 2024-03-25:
Sorry to be a pain but still not working.

I've been playing around by manually sending some mqtt messages.
The configuration has to be explicitly set as  `action_topic` and `action_template`. `action_t` and `action_tpl` don't seem to work.
Also, lower case is indeed required `IDLE` doesn't work while `idle` does

revk, 2024-03-26:
So everywhere else in HA `_t` is the same as `_topic`, but not here, really?

OK, new beta in a moment...

MUN0X, 2024-03-26:
Thats what I saw during testing yesterday. I could report this to HA team.

Anyway, hvac_action is now reported correctly by faikin and HA picks it up.

revk, 2024-03-26:
Yay!