revk/xESP32-Faikout #276: Faikin does not connect to Emura 3

migueldc73, 2024-03-25:
I finally found the right PA 2mm pitch connector and got my Emura s21 port connected to the Faikin.

The Faikin clearly gets power and I can access the webpage, but there seems to be no connection with the FTXJ50A2V1BB indoor unit 👍 
[image]

System is offline and I also cannot turn it on neither via the webpage or Home Assistant
I also don't get temperate info in Home Assistant

I connected the S21 pins 1-5 to the same pins on the Faikin as per the instructions
[image]
[image]

I have suspicion that there is something wrong in the pin layout though
Pin 5 (black) : GND, seems to be OK on both
Pin 4 (red) : VCC 12V seems to be OK on both as the Faikin boots and I get the webpage
Pin 3 (yellow) : should be RX on S21 side, hence TX on Faikin side ? Faikin board just says "48" 
Pin 4 (orange) : should be TX on S21 side, hence RX on Faikin side ? Faikin board just says "34" 
Pin 5 (brown) : I connected it through as per isntructions although many users say it's "unused"

The Emura 3 Wifi seems to work fine as I can turn it on/off and change settings via the Onecta cloud app.

So what did I do wrong ?

revk, 2024-03-25:
It is normal 1:1, you don't normally swap tx and rx at all. It is always an issue with Tx and Rx and shoes side it is viewed from, Pin 1 (not 5) is typically unused, yes. 48 is "from the Faikin" (see little arrow), and 34 is "to the faikin" (again, and arrow). Does that help?

migueldc73, 2024-03-25:
I actually didn't swap the TX and RX, sorry should have said that.

If you look at the pics of both S21 and Faikin side, yellow is on Pin 3 on both sides, same for orange on Pin 4

Do you think I should disconnect Pin5 (cut brown cable ) ?

revk, 2024-03-25:
Depends what voltage, if 5V, should be fine, if GND, then would break it.

migueldc73, 2024-03-25:
I could measure the pins. Do you think yellow and orange should be swapped then on my setup if you look at both pictures ?

revk, 2024-03-25:
I would hope they have not gone for a different wiring.

If you can check voltages from pin 5 (GND) to each other pin when not connected to the Faikin.

Be very careful in case there are mains levels on anything!!!

migueldc73, 2024-03-25:
So I have 
GND to Pin 4 (red) : 14.4V
GND to Pin 3 (yellow, supposedly TX) : 0V
GND to Pin 2 (orange, supposedly RS) : 0V
GND to Pin 1 (brown, unused) : 0V

migueldc73, 2024-03-25:
I cut the brown wire (nothing exploded, LOL), but still same issue 
- Faikin is alive and on the LAN
- system still offline, so no communication over TX/RX

Is there maybe a setting (or dipswitch) on these new Emura 3 to enable/disable communication over S21 ?
But then it's still weird there is 14.4V over pin 2

revk, 2024-03-25:
OK, you have me at a loss now. Sorry, not sure if anyone else in a discussion may have an answer on this.

The fact the rx/tx are 0V not 5V is a concern, and probably why not working. I guess maybe it needs enabling?

migueldc73, 2024-03-25:
The problem is probably my self-made cable with PA connector. I measured the voltage directly on the S21 connector and the TX/RX seem to have 5V there.
Luckily my official Daikin EKRS21 cable just arrived at my Daikin shop, so will try that tomorrow.

revk, 2024-03-25:
Cool.

migueldc73, 2024-03-26:
So I connect the EKRS21 cable and the Faikin started up, but then it completely hung.
The magenta/white flashing light went on solid for a few seconds and then died out completely 
[image]

I suspect something is wrong with my Faikin unit, but why is a mystery.

Checking the voltages on the end of the new EKRS21 cable now to be sure

migueldc73, 2024-03-26:
So it seemed that the 15V I took from the S21 is not stable (called my Daikin dealer to maybe replace the board), so I took the 5V from Pin1 (unused) of the S21 and fed it into Pin 4 (supposedly 12V) on the Faikin.

**Do you think the Faikin will run reliable with 5V ?**

I can now control my untit, turn it on/off, but it does weird things like turning telling the fan settings are wrong, selected temperature is jumping around, ....

revk, 2024-03-26:
The voltage is not the issue, it is the current available. On a normal S21, the 12V (or more) usually has enough power, but the 5V seems to be just a low current reference of some sort. So usually you can't run off the 5V.

If you have a 5V that is running the Faikin at all then it is likely an actual power supply, so is probably fine. The Faikin itself will be fine at 5V (the reg is rated roughly 4V to 36V).

migueldc73, 2024-03-26:
Well, I am actually powering the Faikin from the 5V (so-called reference, pin 1) and it starts up fine.
So you are saying that if it boots up from the 5V, it will probably be running fine all along, right ?

The behaviour of my indoor unit is very strange though. I can turn on/off and select mode, but when I try to change fan speed, it gives this error message that the fan speed is wrong. Will make some screenshots later.

revk, 2024-03-26:
Yes, if that works then it has to be a power feed of some sort, I suspect.

Fans are fun, but we can dump the S21 messages and see.

migueldc73, 2024-03-26:
ok, will do that later. I guess the manual explains how to dump the messages either from the Faikin webpage or HA ?

Something weird though : I powered the Faikin from the 5V of a USB connector on my laptop and sometimes it just stops working after a few mins, but now I restarted it seems to be running 10mins fine...

migueldc73, 2024-03-26:
I think there is actually something wrong with my unit, because even after disconnecting my Faikin I have strange fan behaviour
- in Heating mode, I can only select Auto fan. It is not possible to choose Silent of any of the manual fan modes. Not with the Onecta app, but also not the remote control. I actually noticed this issue already right after the install, because I thought it was Onecta related. 
- in other modes, it does work though
- my other 4 units are Stylish (with the old local API wifi module) and everything works fine there.

So, there is no point to let you debug if there is most probably something wrong with my unit.

But Daikin support will be a tough cookie, I'm afraid

migueldc73, 2024-03-26:
it might also be that because I used Pin 1 (5V, probably reference) as power source for the Faikin, that it somehow makes the TX/RX voltage variations flaky, resulting in flaky serial communication.

So, does anybody know of any other source of 5-15V power on the Daikin board ?
With enough amps obviously,

revk, 2024-03-26:
It should not, we drive low, using the internal pull up on the Tx, and the Rx goes in to a FET, so both should be fine whatever power we run from. Only other possible is maybe also connecting the 5V to pin 1 as that will then add a pull up to 5V on the Faikin side.

migueldc73, 2024-03-26:
OK, so we have now contained and identified that 
1) using 5V instead of 15V ps is not an issue as long as the Faikin boots and runs
2) There is a firmware bug in my Emura 3 that makes any other fan setting than Auto in Heating mode impossible

So now I get the Faikin back connected, I would like to investigate why the set temperatures are jumping around in every mode (tried heat & cool so far).

How can I download the logfile to send to you ?

revk, 2024-03-26:
Lots of others have it working fine with HA. The host is, err, hostname or IP, simple as that. It even handles old school IPv4 addresses. The Faikin auto does set high and low target temps to turn on and off, that is normal. There are some settings that can be used to fine tune that if needed. It should not have been on by default, honest. It could also impact fan settings.

migueldc73, 2024-03-26:
Well, I actually integrated the Faikin directly into Daikin AC local integration into HA.
That's maybe why Mosquitto didn't recognize it, either one or the other.

[image]

[image]

Is the Daikin AC integration logfile useful to you ?

migueldc73, 2024-03-26:
I have to admit that everything seems to be working fine now, apart from the one issue that is when changing from any of the modes to heating mode, this fan error pops up in HA (as well as in the Faikin web interface) 

[image]

migueldc73, 2024-03-26:
Update : 
- this fan mode message actually always appears after changing any mode (so not only heating) and changing temperature.
- You then see the dot (undefined fan mode)
- after 5-10 seconds, HA corrects itself and the actual mode (auto, silent, 1-5) appears and it all seems to work as it should

Seems like a status feedback issue on the fan state.

MUN0X, 2024-03-26:
> I have to admit that everything seems to be working fine now, apart from the one issue that is when changing from any of the modes to heating mode, this fan error pops up in HA (as well as in the Faikin web interface)
> 
> ![Screenshot_20240326_164941_Home Assistant](https://private-user-images.githubusercontent.com/107763234/316943431-70cd4bd7-bd2a-41f8-a50d-1b876fed088e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE0NzczNTAsIm5iZiI6MTcxMTQ3NzA1MCwicGF0aCI6Ii8xMDc3NjMyMzQvMzE2OTQzNDMxLTcwY2Q0YmQ3LWJkMmEtNDFmOC1hNTBkLTFiODc2ZmVkMDg4ZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyNlQxODE3MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMmY0YTdlOWMzMGY3NWJlOWM5YWU2YjFiYzg5OGVmY2RlNDg4ZDdiNWMyZmMwNGE0MGI0Yjg0ZTgwOGE4YmUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.xQL5nDKCuLSpn69DLiznkqMOmAkDjWic-RnV0utVb-o)

Unrelated question bit how did you configure that mushroom card?

migueldc73, 2024-03-26:
> > I have to admit that everything seems to be working fine now, apart from the one issue that is when changing from any of the modes to heating mode, this fan error pops up in HA (as well as in the Faikin web interface)
> > ![Screenshot_20240326_164941_Home Assistant](https://private-user-images.githubusercontent.com/107763234/316943431-70cd4bd7-bd2a-41f8-a50d-1b876fed088e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTE0NzczNTAsIm5iZiI6MTcxMTQ3NzA1MCwicGF0aCI6Ii8xMDc3NjMyMzQvMzE2OTQzNDMxLTcwY2Q0YmQ3LWJkMmEtNDFmOC1hNTBkLTFiODc2ZmVkMDg4ZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyNlQxODE3MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMmY0YTdlOWMzMGY3NWJlOWM5YWU2YjFiYzg5OGVmY2RlNDg4ZDdiNWMyZmMwNGE0MGI0Yjg0ZTgwOGE4YmUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.xQL5nDKCuLSpn69DLiznkqMOmAkDjWic-RnV0utVb-o)
> 
> Unrelated question but how did you configure that mushroom card?

It's the new Tile Card, available since a few months on HA. I am quite happy with it, considering I make most of my cards completely with custom yaml code

MUN0X, 2024-03-26:
Nice! Missed that introduction completely. Might have to redo my heating dashboard now 😀

migueldc73, 2024-03-27:
Everything work fine for 24hours and I thought I was out of the storm, but something weird happened just now 
- can't access the Faikin over the local IP address 
- I can ping it though
- MQTT in Home Assistant is working fine (no fan errors)
- Daikin HA integration is not working though (worked fine yesterday, allthough with the fan error)
- rebooting Faikin (so power off through the mains and back on) solves the issue

It seems like the Faikin state gets corrupted. So I wonder
- known bug ?
- wrong setting (will send screenshots in a min)
- not enough amps on the 5V (but then I guess it would just power off and not even be pingable...)
- something wrong with my Faikin ?

migueldc73, 2024-03-27:
[image]
[image]
[image]

sjaakie09, 2024-03-27:
mine has also this issue can't access the Faikin over the local IP address
I can't  ping it though the last known address? and also not with lan scan?
is there an other way to find the unit it also does not been shown with the host name?

MUN0X, 2024-03-27:
I had the same problem a few days ago with 1 of my faikins. Worked fine via mqtt, webinterface was not available. Unplugging it resolved the issue.

migueldc73, 2024-03-27:
Would be nice if we get fix for this because I actually like the Faikin HA integration 

Pro Daikin HA : 
- responsiveness 
- See the fan states with same naming convention as Daikin : Auto, 12345 and Indoor Silent
- it's elegant to see my Emura integration next to the other Daikin's with the old local API wifi module

Pro MQTT 
- Offers more functionality (actually all of them)
- no annoying fan error bug

revk, 2024-03-28:
I am not sure I understand your fan issue. Does it not work even via the Faikin web interface.

By the way, "silent" is a known issue, and a problem with the underlying S21 protocol, but other fan modes should all work as expected.

revk, 2024-03-28:
New beta.

1. I have changed `temptrack` to mean basing target on Daikin measured temp
2. I have added `tempadjust` to mean applying the adjustment for Daikin measured not being external measured

migueldc73, 2024-03-28:
> I am not sure I understand your fan issue. Does it not work even via the Faikin web interface.
> 
> By the way, "silent" is a known issue, and a problem with the underlying S21 protocol, but other fan modes should all work as expected.

I will make a detailed report later today, but I suppose it's better we start a new thread for this, right ?

closed 2024-05-06 by revk