revk/xESP32-Faikout #278: Connect to the wired remote ?

iSalah, 2024-03-26:
Is it possible to connect Faikin to the wired remote like the [BRC1E52](https://my.daikin.eu/content/dam/MDM/Pictures/Accessories/Controls/Packshots/BRC1E52A_F_heat_detail_tcm135-222146.tif/_jcr_content/renditions/cq5dam.web.1280.1280.jpeg) ?

Thanks

njh, 2024-03-26:
Is this so that you can use both the Faikin and the BRC1E52 to control your air conditioning unit?

I think so far the verdict is that it isn't possible to have multiple devices connected to the S21 port.
But it is possible that there is more than one port on your air conditioner that is Faikin compatible - but I am not sure of the details.

iSalah, 2024-03-26:
Yes I want to use both the remote and the Faikin.
My remote doesn’t have an S21 connector, it’s only 2 wires:
[image]

I see there is an empty X1A connector on the remote though.

revk, 2024-03-26:
P1/P2 is a separate protocol and unrelated to S21, so should be able to use both on the same system at the same time. I do.

iSalah, 2024-03-26:
Oh I see, thank you.
Just in case, is this protocol handled by Faikin ?

I’m asking because the air units are in the ceiling with almost no access
to it, I really don’t see how I could get access to that S21 port
unfortunately.

serimi89, 2024-03-27:
To address the p1/p2 directly, you might want to check out this project:
https://github.com/Arnold-n/P1P2Serial

As revk mentioned, I am using faikin on X50 port with P1/P2 Daikin Madoka display in parallel without problem.

iSalah, 2024-03-29:
Thank you for your comments, I'll check on the P1P2 project then.

My units are embedded in the ceiling, I can't even check if the ports you've mentioned are available.

closed 2024-03-29 by iSalah

AftDawn, 2024-05-03:
Hi, i have a BRC073, it does use the S21 interface (through a "User Interface" aka some small box) but it connects to the wired remote though "S1", is S1 just S21?
If not, can i make an S21 splitter and connect a falkin to that?