revk/xESP32-Faikout #284: invalid fan mode

migueldc73, 2024-03-28:
**Faikin hardware**
Faikin-S3-MINI-N4-R2 bought from Amazon UK
(powered over 5V from Pin 1 as my S21 does not have any voltage over Pin 4)

**Daikin hardware**
Daikin Emura 3 FTXJ50A2V1BB connected over S21 

**Describe the bug**
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:281
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 9:42:48 PM (8 occurrences)
Last logged: 9:54:45 PM
[139737393004224] The fan_mode is not a valid fan_mode: Auto, Silence, 1, 2, 3, 4, 5
[139737477793984] The fan_mode is not a valid fan_mode: Auto, Silence, 1, 2, 3, 4, 5

**To Reproduce**
The " The fan_mode is not a valid fan_mode: Auto, Silence, 1, 2, 3, 4, 5"  error only appears 
- when turning the airco on or changing modes 
- and only in the official HA Daikin integration, NOT in the Faikin local webpage and NOT in MQTT

after the error appears, you can clearly see the state of the fan is null and only becomes the actual value after a few seconds
[image]
[image]
[image]
[image]

When turning the airco on/off or changing modes in MQTT or the local page, the error does not appear on neither side
But you can clearly see that the Daikin integration is about 30secs behind in receiving the state that unit was actually turned out through MQTT
[image]

Full logs attached 
[home assistant_2024-03-28T21-14-07.290Z Daikin.log](https://github.com/revk/ESP32-Faikin/files/14795752/home.assistant_2024-03-28T21-14-07.290Z.Daikin.log)

revk, 2024-03-29:
Hang on, this is using the legacy URLs is it? Isn't using MQTT a better idea?

I can look at the legacy URLs and see if I can see why it is upset, but I had not realised that is what you were using.

migueldc73, 2024-03-29:
After adding the Faikin, I can actually control my Emura unit in Home Assistant with 2 integrations 
- MQTT : 
[image]

- The in HA integrated Daikin integration 
[image]

But there are a few reasons why I prefer the Daikin integration over MQTT in HA 
- More responsive : when I turn the unit on/off, change mode/temp via MQTT, it somehow takes 5secs for the state to be updated. With the Daikin integration it's instanteous. Even the new Daikin Onecta cloud API (which I hate) is faster to update...
- The fan state naming convention in MQTT is non-standard. I prefer the Daikin naming convention Auto, 12345, Indoor Silent
- my other 5 Stylish units are under the Daikin integration, so it just feels more elegant (from a HA nerd pov) to see all 6 units under the same integration

So if I understand your comment well, the Daikin integration doesn't use MQTT (makes sense), but the legacy URLs ?

**Is there a way to force the Daikin integration to use legacy URLs ?**

If not, I am happy to use just MQTT, but I wonder how I can get it more responsive then ?
State naming I can solve by creating a Climate template, which I might do anyway.

revk, 2024-03-29:
I am surprised it is quicker. My understanding was anyone using HA used MQTT and only those using something else would use the legacy URLs. So I'll look.

I also understood the MQTT fan names worked with HA.

I don't use HA so I just take what people tell me.

But I can look at the legacy URLs. They are somewhat more limited, but were meant to actually work.

migueldc73, 2024-03-29:
Here is a video of the delay in MQTT when turning the unti on and off
https://www.dropbox.com/scl/fi/bocexeve2gvc9146upqcx/20240329_085530.mp4?rlkey=fci91yw53nzxcbkrkp5pzdqis&dl=0

In Daikin mode it's instantaneous 
https://www.dropbox.com/scl/fi/1v7f5cj9fc2fjnr9bnp41/20240329_085824.mp4?rlkey=7lzwwqo2y2etyifz1xd1a3f7k&dl=0

MQTT fan names work, but they are a bit confusing if you are used to use the Daikin integration
1 = Low
2 = LowMedium
3 = Medium
4 = MediumHigh
5 = High
Indoor Quiet = Night
But as said, I can probably solve that by creating a climate template in HA (virtual climate entity in yaml)

I found another bug : if you turn off the unit via MQTT, the Daikin integration does not follow and it stays on. The cloud app shows it as turned off with the usual cloud-related delay of course

This is the HA Daikin integration. Plz confirm that this is indeed using legacy URLs and not MQTT 
https://www.home-assistant.io/integrations/daikin

migueldc73, 2024-03-29:
Also more detailed log files attached. 
[home-assistant_daikin_2024-03-29T07-17-30.930Z.log](https://github.com/revk/ESP32-Faikin/files/14801533/home-assistant_daikin_2024-03-29T07-17-30.930Z.log)

revk, 2024-03-29:
OK looks like that uses the legacy URLs, and is looking at `f_rate`

Trying to find what it actually expects.

revk, 2024-03-29:
OK looks like maybe it expects a reply to `set_control_info`.

```
2024-03-28 21:51:07.101 DEBUG (MainThread) [pydaikin.daikin_brp069] Sending query_c: aircon/set_control_info?pow=0&mode=4&stemp=25.0&shum=0&f_rate=0&f_dir=0
2024-03-28 21:51:07.184 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139737477793984] The fan_mode  is not a valid fan_mode: Auto, Silence, 1, 2, 3, 4, 5
```

If you have old units that have these legacy URLs can you access via a web browser.

`http://`*IPAddress*`/aircon/set_control_info?pow=0&mode=4&stemp=25.0&shum=0&f_rate=0&f_dir=0`

And see what response you get?

migueldc73, 2024-03-29:
The Daikin integration also doesn't show the state that the unit is not ready to warm

MQTT integration showing symbol in the upper corner that the compressor isn't read to send warm liquid :
[image]

Daikin integration of the Faikin does not show this
[image]

revk, 2024-03-29:
I have no idea what field that is. The legacy URLs were done to the minimum that seemed to be needed for some other tools.

To add more I need some examples from actual Daikin units I expect (as per previous post).

migueldc73, 2024-03-29:
> `http://`_IPAddress_`/aircon/set_control_info?pow=0&mode=4&stemp=25.0&shum=0&f_rate=0&f_dir=0`

On the old units, I get the reply
 _ret=PARAM NG_

On the Faikin, I get the reply
 _ret=PARAM NG,adv=Invalid f_rate value_

revk, 2024-03-29:
Well that is encouraging, your unit and the Faikin both think `f_rate=0` is wrong, which is what I was expecting. So I wonder why the integration tries that.

Can you do the same (on both) but with `f_rate=A` instead of `f_rate=0`

Thanks.

migueldc73, 2024-03-29:
We are getting somewhere here 

On both the old (Daikin local Wifi) and the new (Faikin) unit I get a positive reply when using `f_rate=A`

_ret=OK,adv=_

revk, 2024-03-29:
OK wonder why the error then?

Try `http://`*IPAddress*`/aircon/get_control_info` on both.

migueldc73, 2024-03-29:
On the old units with power off, the reply
_ret=OK,pow=0,mode=4,adv=,stemp=25.0,shum=0,dt1=25.0,dt2=M,dt3=25.0,dt4=25.0,dt5=25.0,dt7=25.0,dh1=0,dh2=50,dh3=0,dh4=0,dh5=0,dh7=0,dhh=50,b_mode=4,b_stemp=25.0,b_shum=0,alert=255,f_rate=A,f_dir=0,b_f_rate=A,b_f_dir=0,dfr1=A,dfr2=5,dfr3=5,dfr4=A,dfr5=A,dfr6=5,dfr7=A,dfrh=5,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0_

On the new Faikin unit with power off
_ret=OK,pow=0,mode=4,adv=,stemp=25.0,shum=0,dt1=25.0,dt2=25.0,dt3=25.0,dt4=25.0,dt5=25.0,dt7=25.0,dh1=0,dh2=0,dh3=0,dh4=0,dh5=0,dh7=0,dhh=0,b_stemp=25.0,b_shum=0,alert=255,f_rate=A,f_dir=0,dfr1=0,dfr2=0,dfr3=0,dfr4=0,dfr5=0,dfr7=0,dfrh=0,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd7=0,dfdh=0,dmnd_run=0,en_demand=0_

revk, 2024-03-29:
OK, both have `f_rate=A`

I wonderr what `b_mode` and `b_f_mode` are, looks like the same as `mode` and `f_rate`. I can add them maybe.

migueldc73, 2024-03-29:
FYI, some of the other old units reply with this
_ret=OK,pow=0,mode=6,adv=13,stemp=--,shum=--,dt1=25.0,dt2=M,dt3=25.0,dt4=25.0,dt5=25.0,dt7=25.0,dh1=AUTO,dh2=50,dh3=0,dh4=0,dh5=0,dh7=AUTO,dhh=50,b_mode=6,b_stemp=--,b_shum=--,alert=255,f_rate=5,f_dir=0,b_f_rate=5,b_f_dir=0,dfr1=5,dfr2=5,dfr3=5,dfr4=5,dfr5=5,dfr6=5,dfr7=5,dfrh=5,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0,dmnd_run=0,en_demand=0_

slightly different to the very old unit from 2016 which replied
ret=OK,pow=0,mode=4,adv=,stemp=25.0,shum=0,dt1=25.0,dt2=M,dt3=25.0,dt4=25.0,dt5=25.0,dt7=25.0,dh1=0,dh2=50,dh3=0,dh4=0,dh5=0,dh7=0,dhh=50,b_mode=4,b_stemp=25.0,b_shum=0,alert=255,f_rate=A,f_dir=0,b_f_rate=A,b_f_dir=0,dfr1=A,dfr2=5,dfr3=5,dfr4=A,dfr5=A,dfr6=5,dfr7=A,dfrh=5,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0

revk, 2024-03-29:
The all have those additional `b_`... fields, so I've added them now.

migueldc73, 2024-03-29:
Yep, it works ! no more fan more errors :-)

Still don't see the "compressor not ready" state yet, but I can live with that (hopefully my wife too...) : 

MQTT integration showing symbol in the upper corner that the compressor isn't read to send warm liquid :
[image]

Daikin integration of the Faikin does not show this
[image]

revk, 2024-03-29:
Only thing I can suggest is checking that `get_control_info` on units that are not ready/ready and trying to work out which setting. That said, I am not sure I know a "compressor not ready" state from S21, so may not be able to report it anyway.

migueldc73, 2024-03-29:
There is indeed no difference in the `get_control_info` feedback in "waiting" and "ready" state 

Old airco turned off
_ret=OK,pow=0,mode=4,adv=,stemp=25.0,shum=0,dt1=25.0,dt2=M,dt3=25.0,dt4=25.0,dt5=25.0,dt7=25.0,dh1=0,dh2=50,dh3=0,dh4=0,dh5=0,dh7=0,dhh=50,b_mode=4,b_stemp=25.0,b_shum=0,alert=255,f_rate=A,f_dir=0,b_f_rate=A,b_f_dir=0,dfr1=A,dfr2=5,dfr3=5,dfr4=A,dfr5=A,dfr6=5,dfr7=A,dfrh=5,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0_

Old airco turned on, but compressor not ready
_ret=OK,pow=1,mode=4,adv=,stemp=25.0,shum=0,dt1=25.0,dt2=M,dt3=25.0,dt4=25.0,dt5=25.0,dt7=25.0,dh1=0,dh2=50,dh3=0,dh4=0,dh5=0,dh7=0,dhh=50,b_mode=4,b_stemp=25.0,b_shum=0,alert=255,f_rate=A,f_dir=0,b_f_rate=A,b_f_dir=0,dfr1=A,dfr2=5,dfr3=5,dfr4=A,dfr5=A,dfr6=5,dfr7=A,dfrh=5,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0_

Old airco turned on, compressor ready to send heat to the room
_ret=OK,pow=1,mode=4,adv=,stemp=25.0,shum=0,dt1=25.0,dt2=M,dt3=25.0,dt4=25.0,dt5=25.0,dt7=25.0,dh1=0,dh2=50,dh3=0,dh4=0,dh5=0,dh7=0,dhh=50,b_mode=4,b_stemp=25.0,b_shum=0,alert=255,f_rate=A,f_dir=0,b_f_rate=A,b_f_dir=0,dfr1=A,dfr2=5,dfr3=5,dfr4=A,dfr5=A,dfr6=5,dfr7=A,dfrh=5,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0_

Same for the newer, but not Faikin units
_ret=OK,pow=0,mode=4,adv=13,stemp=25.0,shum=0,dt1=25.0,dt2=M,dt3=25.0,dt4=25.0,dt5=25.0,dt7=25.0,dh1=AUTO,dh2=50,dh3=0,dh4=0,dh5=0,dh7=AUTO,dhh=50,b_mode=4,b_stemp=25.0,b_shum=0,alert=255,f_rate=A,f_dir=0,b_f_rate=A,b_f_dir=0,dfr1=5,dfr2=5,dfr3=5,dfr4=A,dfr5=A,dfr6=5,dfr7=5,dfrh=5,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0,dmnd_run=0,en_demand=0

ret=OK,pow=1,mode=4,adv=13,stemp=25.0,shum=0,dt1=25.0,dt2=M,dt3=25.0,dt4=25.0,dt5=25.0,dt7=25.0,dh1=0,dh2=50,dh3=0,dh4=0,dh5=0,dh7=0,dhh=50,b_mode=4,b_stemp=25.0,b_shum=0,alert=255,f_rate=A,f_dir=0,b_f_rate=A,b_f_dir=0,dfr1=A,dfr2=5,dfr3=5,dfr4=A,dfr5=A,dfr6=5,dfr7=A,dfrh=5,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0,dmnd_run=0,en_demand=0

ret=OK,pow=1,mode=4,adv=13,stemp=25.0,shum=0,dt1=25.0,dt2=M,dt3=25.0,dt4=25.0,dt5=25.0,dt7=25.0,dh1=0,dh2=50,dh3=0,dh4=0,dh5=0,dh7=0,dhh=50,b_mode=4,b_stemp=25.0,b_shum=0,alert=255,f_rate=A,f_dir=0,b_f_rate=A,b_f_dir=0,dfr1=A,dfr2=5,dfr3=5,dfr4=A,dfr5=A,dfr6=5,dfr7=A,dfrh=5,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0,dmnd_run=0,en_demand=0_

Then I realized I actually do get the state on the Faikin HA Daikin integration in Heat and Cool mode, but NOT in Auto
And that is the same case for older units
[image]
[image]

I do get the state icon also in Auto in the MQTT integration.

I don't think it's deal breaker, but still interesting.

Which brings me to the other side of the equation, why is MQTT so slow ? 
My HA machine is very fast, so might be an issue with my broker ? Polling interval setting ?
I never really dived into the MQTT broker settings, so not even sure that is configurable.

revk, 2024-03-29:
My understanding is HA makes a change, via MQTT, and shows the change immediately. We periodically update status and show the change then. When I last tested there was no lag, but if the change could not be applied it would revert.

There are settings to report changes quicker if I remember correctly, rather than periodically. But I was under the impression HA always assumed it worked and reported as such instantly.

migueldc73, 2024-03-29:
Changing from MQTT protocol v3.1.1 to 5 improved responsiveness drastically

Thanks for the great support. I learned a lot from this.