revk/xESP32-Faikout #286: [FEATURE] FTXPxxNxxxx compatibility

eskey0, 2024-03-30:
Hello there, I bought 3 faikin to replace my oficial modules, I hope since FTXPxxMxxxx was compatible mine were too, mines are comfora 2023 (not sure about the year, I got them installed last year). I have 1 of the aircons that are another version that I think it is compatible.

Today I dissasembled the unit and I found out that I have no S21, but a S430. The module looks like the ones with the PAP-05V-S (CTXMxxRxV1B and FTXMxxRxV1B) [module](https://imgur.com/ovaNxmS) the connector looks different, it also has 5 leads [S403 connector](https://imgur.com/GayZKkt)

Im down to do whatever to test this out, or provide any more information and make it working (preferably without breaking the unit :)) I'm new to electronics and such but with a bit of guiance I can make it work. At least just for the sake to ditch cloud daikin.

revk, 2024-03-30:
Covered by https://github.com/revk/ESP32-Faikin/issues/208 I think.

But as far as I know S403 is **LIVE**, so be very careful.

closed 2024-03-30 by revk

revk, 2024-03-30:
I think Daikin do use consistent pin-outs for consistent numbered connectors, so if you have an `S403` it should match. But yeh, good luck.

eskey0, 2024-03-30:
Well yeah that was a totall success! Now I have to find a female connector that fits snuggly in the aircon conector and thats it. I guess you can put it in the table of compatible units, the only difference is the s403 is rotated 90 degrees.

Thanks for the work finally out of daikin cloud hands! And thanks to Jerther that found the s403 pinout.

boecks, 2025-05-12:
Sorry for crashing this one year later. I have several FTXP M with Faiking working nicely, where I just hat do swap original Wifi Modules for Faikins.

Now one FTXP N is potentially added and I learned, it will potentially not be as easy as above. Do you have any pictures and part lists for the wiring? 

Thank you.

revk, 2025-11-03:
I don't know, but it is a wiki page, so please do edit you you think so.