revk/xESP32-Faikout #288: Demand control changes of onecta Daikin are not reflected on faikin webpage

hokreb, 2024-03-31:
**Faikin hardware**
2024-01-06, I bought this module from Amazon germany. The faikin module is conntected by the s21 connected with an own made cable with plugs and crimpt with a tool. The cable is about 40cm long.
The faikin software is updated by ota to the latest version. I have not changed any advanced settings

**Daikin hardware**
FTXM42R, multisplit system with five indoor and one outdoor unit

**Describe the bug**
If I change the demand control value in the onecta app, this change is not reflected on the faikin webpage. 
If I change the Fan mode (1, 2, 3, 4, 5), this is imediately reflected on the faikin webpage. It's also the other way around, if I change the demand control value in the faikin webpage, this change is not reflected in the onecta app (it needs to be restarted for getting updated data). This works without any problem with the fan mode.

The same thing happens with the Econo mode, the change in the onecta Daikin app is not reflected, but the Streamer mode is reflected immediately. 

If I can help with activation of debug/dump messages, just give me instructions, I would be happy to help!

hokreb, 2024-03-31:
As an additional finding, although the changes are not reflected the demand control goes active, which I see at the power usage, so this is not a functional problem.

revk, 2024-04-01:
It is reverse engineered, but I think we have it right.

For one way, turning on the dump setting should provide some logs on MQTT.

If you can post those before and after the change, we can see what the aircon is telling the Daikin.

hokreb, 2024-04-01:
So, I now have set dump mode on, and log the mqtt info topic (and also the error, event setting and command topics). When I set econo mode or change the demand control from 40 to 50 %, I do not see any "new" command or which is supposed to be the D7 command. When I set the eco mode by the faikin web ui, it tries to set it by the command "D7" and payload "]200", but after trying it 10 times, it fails and an error will be printed on the error topic (failed-set {"econo":true}).
I have installed a second faikin modul to another indoor unit (I have five indoor units, but till this morning only one faikin installed), there is the behaviour the same, these indoor units only differ by the maximum heating/cooling power.

The setting of the demand control (40% ..100%) does need also several tries, sometimes an error will be send to the error topic. Sometimes after 8 tries the value is set and the state is updated.

Does the dump mode really dumps all commands, the indoor unit receives? Or is it a kind of internal debug mode, which sends all command/query over mqtt? Is there the possibility to eavesdrop the wifi modul to get the commands, it receives by the daikin cloud? I am myself into computer programming and embedded systems (although only from the programming side, not from the electrical side). I could even buy some stuff, to help out if it would be helpful, but I would need a concrete step by step guide for this.

revk, 2024-04-01:
There is a debug as well, which may help.

hokreb, 2024-04-01:
When I activate debug as well, the "D7" commands are not shown in the dump/debug output. Is this by intension?

hokreb, 2024-04-01:
Is it possible to send s21 commands directly to the faikin? I am still not able to set the econo mode with the web ui from the faikin modul, I always got error (error/../failed-set {"econo":true}). Sometimes I see in the G7 Response the second position set to ascii char 8. So it would be interesting to see, what happend, if i send this command to the s21 port.

revk, 2024-04-02:
You can send a command from MQTT. There is something in the manual pages for that.

miniluma, 2024-04-03:
@revk Maybe related to this issue, if I try to modify the Demand value, I see an error in MQTT: 
[image]

It refers X50A protocol but I have S21.

miniluma, 2024-04-03:
I enabled "debug". If I understand the implementation correct, you expect Econo and Demand in S21_G7, right?
I see changes in Econo in G7, but App changes of Demand makes no change in s21_G7

revk, 2024-04-04:
Unless you can find another S21 message that tells us what you want, the faikin won't be able to tell.

Maybe there is a system wide demand control or something that uses different messages?

miniluma, 2024-04-04:
I had to correct my writing. After a Power reset of the Daikin Unit (complete switch off and on), I see G7 messages, when I modify it in the App. But it will not be shown in the faikin demand value. And setting a new value in the faikin does not reflect in a change in the App. Maybe they modified the message or it's a bug in the Daikin 1.30 Firmware.

So again my question regarding the encoding. I'd like to understand what I see.

revk, 2024-04-04:
The S21 messages use a fun range of different codings. I'll check this one shortly and let you know.

revk, 2024-04-04:
OK for this one, the first character in G7 is demand, and is coded as the character '0' (0x30) plus a number 0 to 100.

Econo mode is second character, coded as '0' or '2'

miniluma, 2024-04-04:
Here is what I see:
1. I set demand in Faikin Webapp 
-> rx_G7 shows the set value after a couple of seconds
-> I do not see a change in value in the app

2. I change the value in the app
-> no update in rx_G7 (so no update in the Faikin Webapp or MQTT)

Econo works as expected: A change in the Faikin Webapp or the Daikin App is visible in rx_G7
Side note: Sometimes I see Values different from '0' or '2' in the rx_G7 message. E.g. I noticed a ':'.

-----

In the G7 message, there are 2 more bytes. I see frequent changes there. Do you have an idea what they are for?

-----

To look for unknown messages
- how can I do that?
- Who is 024 and 025 in the S21? Is 024 Faikin or the AC?
- Have you written down your knowledge about the message structure of S21 somewhere so that I can get familiar with it?

revk, 2024-04-05:
Some of the S21 has been documented but not entirely up to date, and it would be a good project to get a lot more up to date.

No I don't know what the other two bytes are (yet).

Where do you see 024/025?

miniluma, 2024-04-05:
rx/tx_dump starts with 024 or 025. E.g 
 "0253643632304F03"

revk, 2024-04-05:
That is the raw packet in hex

0253643632304F03

02 is start
53 S
64 d
36 6
32 2
30 0
4F O
03 is end

(Off the top of my head ASCII)

Easier to look at the decoded messages.

miniluma, 2024-04-05:
Holy shit! 
I just installed a new faikin on a 2MX unit.
I think with Firmware 1.30, Daikin made a major improvement:
It was always unclear how the Multi-Split units behave with different demand values of the indoor units.

**Now, I see the following:**

Unit A and B have Demand 50 set.
Action: Unit A App -> Demand to 40 (lower than Unit B)
Unit B sees no message. After a while, Unit A jumps back to 50.
Action: Unit A App -> Demand to 60 (greater than Unit B)
Unit B gets a message and also jumps to 60.

The same if you set it in the Faikin WebApp.

Why I saw no message: I tested with a 4MX Multi-Split but focussed on a single indoor unit and I had to powercycle the outdoor unit!

@revk Everything is fine ;-)
@hokreb Hast du die Daikin Firmware 1.28 oder über die App 1.30? Hast du auf deinem Faikin in UZpdate gemacht? Kannst du mal deine gesamte Anlage für in paar Minuten Stromlos machen?

hokreb, 2024-04-05:
So, I have firmware 1.28, didnt start the update, as I was to suspicious they change something, which makes the faikin not work anymore (maybe too cautious). What I observed was that the econo mode was not selectable by the faikin, then I needed to switch if the power of the whole climate, than I could use the econo mode by faikin. But before I observed, if I use indoor kitchen system to change demand value, I only can use exact this system to change it again. It was really strange, but maybe this has to do with the global aspect of demand control. Unfortunately this week was so busy, that I didn’t find time to play around more with it. I have the feeling, that if the ONECTA takes control, the faikin commands do not „work“ anymore regarding demand and econo feature. Tomorrow I try to get a better understanding for this.

miniluma, 2024-04-06:
> So, I have firmware 1.28, didnt start the update, as I was to suspicious they change something, which makes the faikin not work anymore (maybe too cautious). What I observed was that the econo mode [...]

Do you have a Single or Multi Split?

hokreb, 2024-04-06:
Multisplit with five indoor units

hokreb, 2024-04-06:
Today I played a little bit, but I don't get the really clues. 
When I try to set the econo mode over faikin website, econo mode does not get set, after a while (10-20 sec) the econo mode get switch off again. When I switched on the econo mode once with the ir remote, I can set on/off the econo mode by the faikin website without any problem. I can repeat this on another indoor system, it's exact the same behavior, before I can control the econo mode by faikin website, I have to set it on/off by the ir-remote at least once. And only

hokreb, 2024-04-06:
And after I switched of the econo mode with the remote the return value of G7 is "g8Xe". This should be "g0Xe".

revk, 2024-04-07:
I've started an S21.md to write up what we find.

closed 2024-05-06 by revk