revk/xESP32-Faikout #290: [FEATURE] Modulate hvac_action (Home Assistant) based on fan and compressor activity

sm-Fifteen, 2024-04-02:
**Daikin hardware**
Two FTXS12LVJU heads (connected to the same multiizone compressor) with one Faikin each, plugged over S21 port

**Is your feature request related to a problem? Please describe.**

[image]

On Home Assistant's climate history graphs, my heatpumps appear to be running constantly so long as they are turned on, even though the fan and compressor activity graph shows pretty clearly that it's not actually running for most of that time. (Note: despite the names, both these controllers are connected to the same compressor)

**Describe the solution you'd like**
Even with no direct marker of demand, fan and compressor activity could be enough to guess what the indoors unit is doing.

* In heating mode
  * Fan & compressor running: Heating
  * Compressor running but no fan: Defrosting/Hot start (`HVACAction.PREHEATING`) or possibly idle on a multizone if a different head is runnning
    * Could pe possible to distinguist between each if demand is available?
    * Unclear (to me) how individual Faikin controllers could be aware of the state of other heads on the same pump.
  * Compressor and fan not running: Idle (`HVACAction.IDLE`)
  * Fan running but not compressor: Unclear, but you can see it happen in the graph. Might just go with `HVACAction.FAN`
* In cooling mode
  * Mostly the same, but with cooling (`HVACAction.COOLING`)
* So on for dehumidify (`HVACAction.DRYING`) and fan-only
* Off is obviously just `HVACAction.OFF`

**Describe alternatives you've considered**
I believe the "demand" field normally provides that information in a more accurate way, but as I found out while investigating  #101, it turns out that my unit not only doesn't have energy metering (or it's at the very least not being reported), but also does not report demand or powerful mode (among others).

```js
{
  "G1": "16FA", // Status, currently Fan-only, target 21C, powerful
  "G2": "=;\u0000\u0080",  // Unknown
  "G3": "0þþ\u0000", // Timer
  "G4": "0\u0000\u0080\u0000",  // Unknown
  "G5": "2?0\u0080", // Swing
// No G6 : Active modes (powerful, quiet, comfort airflow, flash streamer, intelligent eye)
// No G7 : Current demand, econo mode
  "G8": "0\u0000\u0000\u0000", // Unknown
// No G9 : Demand
// No other G
  "SH": "542+",
  "SI": "062+",
  "Sa": "090+",
  "SL": "980",
  "Sd": "000",
  "SN": "440+",
  "SM": "440+",
  "SX": "012+",
  "SD": "000"
}
```

Not being able to read demand is unfortunate (it looks to be about how much each indoor unit is reporting to the compressor that it wants to heat/cool), because  I was hoping to use that in home assistant to more accurately track when each is actually heating or cooling. It might still prove useful for other users, though, assuming the metric says what I think it does.

revk, 2024-04-02:
In this just the hvac action field?

I added that to show what the "Faikin auto" mode is doing, as it was apparently not clear.

It is not trying to actually work out what is going on otherwise.

MUN0X, 2024-04-02:
What you describe will most likely not work for multi-splits.

Also dont forget that units have a “searching” routine when idle. They will turn their fan on for 1,5min every 6 minutes.
In your logic this will trigger “heating” while another unit is active.

sm-Fifteen, 2024-04-02:
In this just the hvac action field?
> 
> I added that to show what the "Faikin auto" mode is doing, as it was apparently not clear.
> 
> It is not trying to actually work out what is going on otherwise.

I prefer leaving the units in default Daikin config rather than mamaged Faikin config, personally. One of the downsides of the Home Assistant HVAC integration is that you can't (AFAICT) make your own custom sensors, so if this sort of introspection isn't provided by the device, you can't make that kind of "algorithm" work with local config.

> What you describe will most likely not work for multi-splits.
> 
> Also dont forget that units have a “searching” routine when idle. They will turn their fan on for 1,5min every 6 minutes.
> In your logic this will trigger “heating” while another unit is active.

Is this what these little fan spikes mean? Can you describe in more detail what that behavior is and why you think it would make such a heuristic not work properly?

sm-Fifteen, 2024-04-02:
I actually tried putting a custom state tracker together, and it looks mostly reliable so far, for what it's worth.

[image]

[image]

It's basically just tracking the variables I mentionned before:
* Is compressor running? (hz > 0)
* Is fan running? (hz > 0)
* Is coolant warm? (temp > 15°C)

And then I try making a truth table of the probable action for the given combo of variables at every HVAC mode (currently just heating, as this was a quick test written on my phone).

It's not perfect, you can see an orange sliver in the second screenshot where the compressor and fan are both running, but the fluid is still colder than 15°C. It also can't distinguish between the "stand by for defrost" and "stand by for warmup" phase, but there's no HVAC action that woukd distinguist the two, so that's probably fine. I think it's looking like a promising heuristic, and certainly better than making it look like it's heating all the time.

sm-Fifteen, 2024-04-02:
> > Is this what these little fan spikes mean? Can you describe in more detail what that behavior is and why you think it would make such a heuristic not work properly? Can't these be accounted with by checking coolant temperature, maybe?
> 
> When a unit is idle it will turn its fan on every few minutes to ensure an accurate temperature measurement.  With another unit actively heating this will translatie in an active fan and compressor while the unit is in fact idle.
> You cant rely on coolant temperature because all heat exchangers get hot when a multi-split is heating.  Even units that are switched off but they descipate little to no heat because there is no airflow.

Interesting, good to know. I would have to check if this is a fan speed that units normally reach down to in any mode, as the service manual (for my unit at least) seems to say there is an "LLL speed" that the automatic fan setting will never reach (unclear whether that includes night/quiet/motion sensor away mode, though "SL speed" seems more likely).

[image]

If that's what the units use for that "searching routine", and nothing else, there might be a way of identifying and discarding these, though I understand that this may probe tricky, as this is the sort of behavior that can vary between units. Even if it can't be done, though, and these short fan bursts to circulate the air intake are erroneously seen as heating activity, I would still consider it an improvement over the current state of things.

sm-Fifteen, 2024-04-04:
> > One of the downsides of the Home Assistant HVAC integration is that you can't (AFAICT) make your own custom climate sensors, so if this sort of introspection isn't provided by the device, you can't make that kind of "algorithm" work with local config.
> 
> I use this via HACS: https://github.com/jcwillox/hass-template-climate to make scripted climate wrapper entities with custom logic

Actually, having looked into it a bit more, since these are MQTT devices, I may be able to declare custom devices that point towards the same topics as the originals and replace the `action_template`, at least for testing while figuring out the logic. Haven't had any luck so far, though, and testing detection of the defrost cycle is going to get trickier as we move into spring.

Also, HAss' [HVAC MQTT entities](https://www.home-assistant.io/integrations/climate.mqtt/) don't currently have the "preheating" action, so defrosting would have to be labeled as either cooling or drying.

sm-Fifteen, 2024-04-05:
The truth table of states I had in mind goes like this:

|                      |                     |                 | Off mode | Heat mode | Cool mode | Auto mode   |
|----------------------|---------------------|-----------------|----------|-----------|-----------|-------------|
| **Comp not running** |                     |                 | off      | idle      | idle      | idle        |
| **Comp running**     | **Fan not running** | **Liquid cold** | off      | preheat\* | idle      | idle\*\*    |
| "                    | "                   | **Liquid warm** | off      | idle      | ???       | idle        |
| "                    | **Fan running**     | **Liquid cold** | off      | ???       | cooling   | cooling     |
| "                    | "                   | **Liquid warm** | off      | heating   | ???       | heating     |
| "                    | **Single zone**     | **Liquid cold** | off      | preheat\* | cooling   | cooling\*\* |
| "                    |                     | **Liquid warm** | off      | heating   | ???       | heating     |

* \* Home assistant MQTT HVAC doesn't have a "preheat" state, so those would pobably have to be reported as "cooling", at least for now
* \*\* Could mean preheat in auto mode heating operation, but again HAss MQTT has no state for that.

Given how it's pointless at this time to distinguish between preheat and cooling operations, that could be simplified as follows:

|                      |                     |                 | Off mode | Heat mode | Cool mode | Auto mode |
|----------------------|---------------------|-----------------|----------|-----------|-----------|-----------|
| **Comp not running** |                     |                 | off      | idle      | idle      | idle      |
| **Comp running**     | **Fan not running** | **Liquid cold** | off      | cooling\* | idle      | idle      |
| "                    | "                   | **Liquid warm** | off      | idle      | idle      | idle      |
| "                    | **Fan running**     | **Liquid cold** | off      | cooling?? | cooling   | cooling   |
| "                    | "                   | **Liquid warm** | off      | heating   | heating?? | heating   |
| "                    | **Single zone**     | **Liquid cold** | off      | cooling\* | cooling   | cooling   |
| "                    |                     | **Liquid warm** | off      | heating   | heating?? | heating   |

Or even more simply, regardless of mode:
* Heat, cool or auto mode:
  * Comp not running: Idle
  * Comp running:
    * Single-zone:
      * Fluid warm: Heating
      * Fluid cold: Cooling (possibly for defrost)
    * Not single zone
      * Fan running warm: Heating
      * Fan running cold: Cooling
      * Fan idle cold in heating mode: Cooling (defrost)
      * Fan idle otherwise: Idle
* Fan-only mode:
  * Fan running: Fan
  * Fan not running: Idle?? (I don't think the fan is meant to stop)
* Off mode: Off
* Dry mode:
  * Not sure

It's not *that* bad. A threshold of 10-15°C do distinguish between warm and cold refrigirant seems to work pretty well from my tests, and a threshold of ~7 Hz for fan speed seems to properly reject these air volume mixing pulses that @MUN0X was talking about, though it's probably acceptable if these pulses make it appear like idle units are short-cycling every 6 minutes. At any rate, like I said in https://github.com/revk/ESP32-Faikin/issues/293#issuecomment-2039751047, that could probably just be a setting.

closed 2024-04-05 by revk

trullock, 2024-04-05:
@sm-Fifteen Please share your HA implementation :)

sm-Fifteen, 2024-04-05:
> Ok I am not planning on putting that in Faikin, feel free to do in HA.
> 
> I've made `hvac_action` follow "Faikin auto" mode, but now is a setting to turn on which says it is "Faikin auto" mode only.
> 
> If there is an S21 message/status to get this, I'll add more generically for non Faikin auto mode. Let me know.

Would you accept a PR if were to get a Faikin implementation working?

> @sm-Fifteen Please share your HA implementation :)

It's just a "Data entry" template object that uses Jinja logic to return a string matching the current state. I haven't been able to make an HVAC entity reflect that yet.

revk, 2024-04-05:
I'd rather some S21 that provides the state.

sm-Fifteen, 2024-04-05:
> I'd rather some S21 that provides the state.

Understandable, and I would rather go with that myself, but the power meter thing (specifically how my unit doesn't report it, nor powerful mode), combined with how Daikin's own thermostats don't seem to display that info is all making me start to think that there could very well not be any S21 flag for that.

Assuming there isn't, you would rather report nothing at all, outsite of Faikin's auto mode?

revk, 2024-04-05:
Well for now, it seems a lot of guess work to get anything remotely correct, to me.

There is a lot in S21 we do not understand and one thing I am planning it more tests and documentation.

sm-Fifteen, 2024-04-05:
> Well for now, it seems a lot of guess work to get anything remotely correct, to me.
> 
> There is a lot in S21 we do not understand and one thing I am planning it more tests and documentation.

That's fair. I did notice the wiki page documentaing S21 seems to be somewhat behind what the code knows how to handle, so that would definitely be a good thing. It might also be a good thing to list what official devices were tested to find these signals, in case there are some interesting ones that have yet to be tried (like Daikin One adapters + thermostats or ~~the [wired navigation remote](http://4ae617e62d.nxcli.net/content/commercial/accessories-and-controllers/navigation-remote-control-brc1e73brc1e72brc1e52a7/)~~ nevermind, that's P1P2-only).

The logic I'm proposing isn't as complicated and "guessy" as it may look, since at its core it's mostly "if stuff is spinning and the liquid's hot, then it's heating, and if it's cold we're cooling", with a few edge cases and the added ambiguity of dealing with multi-zone systems.

revk, 2024-04-05:
As long as you are not trying to Jia Tan me, I'll maybe take a look.

hubrikt, 2024-04-20:
Is there any way to stop reporting "HVAC_Action" in HomeAssistant?

@sm-Fifteen have you been able to prepare the template in HA to define your truth table?
Thanks!

revk, 2024-04-20:
Setting `nohvacaction`

sm-Fifteen, 2024-04-22:
> Is there any way to stop reporting "HVAC_Action" in HomeAssistant?
> 
> @sm-Fifteen have you been able to prepare the template in HA to define your truth table?
> Thanks!

@hubrikt : I tried, but no luck. The correct hvac_action does show up in the state viewer in the dev tools, but the UI amd sensors don't reflect it. Looks like this is for display purposes only, and the actual state needs to come from the MQTT topic.

For good measure, here's what I was using in my HAss config:

```yaml
homeassistant:
    customize:
        climate.faikin_entree_mqtt_hvac:
            action_topic: DC9999EEEEEE
            action_template: |
                {% set fan_running = value_json.fanfreq > 0 %}
                {% set comp_running = value_json.comp > 0 %}
                {% set liquid_warm = value_json.liquid > 10 %}
                {% if value_json.mode == 'off' -%}
                    "off"
                {%- elif value_json.mode == 'fan_only' -%}
                    {% if fan_running -%}
                        fan
                    {%- else -%}
                        idle
                    {%- endif %}
                {%- elif value_json.mode == "heat" -%}
                    {% if not comp_running -%}
                        idle
                    {%- elif not fan_running -%}
                        idle
                    {%- elif fan_running and liquid_warm -%}
                        heating
                    {%- else -%}
                        cooling
                    {%- endif %}
                {%- endif %}
```

It doesn't do anything useful (it sets the entity's state attributes according to the dev tools, but that's not recognized by the HVAC integration) and doesn't implement cooling and dry modes, but it shows the logic I was going for.

sm-Fifteen, 2024-09-05:
I hope we can find a "native" way of doing this as part of #408. Home Assistant just added a "defrost" action as  complement to the existing "preheat" action, and while it's not mentionned in the [MQTT climate documentation](https://www.home-assistant.io/integrations/climate.mqtt/), there are tests in home-assistant/core#121448 which show that it would be recognized.