revk/xESP32-Faikout #292: Replace or suppliment the emojis

trullock, 2024-04-03:
Can we either replace the emojis on Faikin's web interface or add a text description of what they mean?

I can't even guess what "hands on head" or "teddy" mean :)

Awesome product otherwise, bought 3 👍

revk, 2024-04-04:
Hmm, the table cells all have tool tips to explain.

I suspect that is not as obvious on a mobile.

revk, 2024-04-04:
OK try latest beta, has a new `no.icons` setting.

closed 2024-04-04 by revk

trullock, 2024-04-04:
> Hmm, the table cells all have tool tips to explain.
> 
> I suspect that is not as obvious on a mobile.

Ahh on desktop yes there are tooltips, but only on the controls and not on the icons. Instinctively I was hovering the icons to see what they were but nothing happened. Might be worth tooltipping those too?

revk, 2024-04-04:
Done, try latest beta

trullock, 2024-04-04:
Lightening turnaround