revk/xESP32-Faikout #293: Is it possible to tell what the unit is doing when in Auto mode?

trullock, 2024-04-03:
**Daikin hardware**
2024-01-06 via S21

**Is your feature request related to a problem? Please describe.**
Coming from the Daikin Residential (HACS) integration for HA, it was impossible to know what the HVAC_Action (i.e. what the unit was currently doing - heating or cooling) when in Auto mode. Is this information available, and can we expose it?

**Describe the solution you'd like**
HVAC_Action to be set to heating/cooling/idle reflecting the unit's current operation when in Auto mode

**Describe alternatives you've considered**
I currently wrap the HA climate entity in a templated one which guesses the action based on the temperature setpoint vs current temperature.

revk, 2024-04-03:
For the Faikin auto mode it does that already.

If you know what the S21 coding is to get from the aircon itself I could add that.

trullock, 2024-04-03:
You mean it already does the temperature difference "guess"?

I don't know the S21 coding, is there something I can read to get up to speed on how to sniff it?

revk, 2024-04-04:
When you set the Faikin to control the temperature, it turns on or off heating or cooling by setting a set point higher or lower. When it does that it sets the hvac action to match what it is doing.

revk, 2024-04-04:
Yeh, others have suggested this, but it would be guess work, and fooled by fan running for other reasons and so on.

44616E, 2024-04-04:
Can this logic not be implemented in HA?

trullock, 2024-04-04:
Thats where i do it at present, but can be improved since Faikin now exposes me the liquid temp.

I was hoping the actual state (no guesswork) would be exposed to faikin for it to report 1-1

revk, 2024-04-04:
It may be, there is a lot of S21 status? But not one I know as yet.

sm-Fifteen, 2024-04-05:
> Thats where i do it at present, but can be improved since Faikin now exposes me the liquid temp.
> 
> I was hoping the actual state (no guesswork) would be exposed to faikin for it to report 1-1

The best lead I know is that the official wired "navigation controller" can report the "Stand by" state to indicate that the indoor unit is not running either because the outdoor unit is warming up or defrosting, but that could be done via a heuristic as well (as the indoors refigirant plunging to -25°C with the compressor runnung fut the fans off is otherwise pretty unusual in a heating cycle). Even Daikin One doesn't seem to (from the user manuals, anyway) display when a unit is actively heating or cooling, though, so I'm not holding much hope.

Still, there could be some hidden commands we haven't found yet that do all those things.

sm-Fifteen, 2024-04-05:
> Yeh, others have suggested this, but it would be guess work, and fooled by fan running for other reasons and so on.

I'd personally rather have a heuristic that's 80% accurate than the current situation of not reporting anything, as is currently the case. There are also a few things we can do to make sure everyone's happy:

* Allow users to turn off the hvac_mode heuristic in the settings
* Allow users to specify a unit is single-zone, so that compressor activity can be considered a reliable activity metric and not have to be cross-referenced with fan speed
* Allow users to set a fan speed theshold below which the fan is considered idle, to ignore the pulses idle indoor units do to run some air over their temp sensor.

We should probably merge this issue with the other one, though.

trullock, 2024-04-05:
I agree with @sm-Fifteen's approach.

Re "single zone", my compressor hosts two indoor units, but they both have to be in the same mode at once, one cant be heating and the other cooling. Only certain advanced outdoor units support that I think so the option needs not to be about "single zone" but whether the compressor supports dual mode (probably has a proper name)

revk, 2024-04-05:
I'm beginning to wish I had not added this reported value, and maybe I'll make it only appear if there is an explicit setting in future.

44616E, 2024-04-05:
> I'm beginning to wish I had not added this reported value, and maybe I'll make it only appear if there is an explicit setting in future.

Maybe only appear when the faikin auto mode is enabled. I think there are too many different types and setups to worry about a heuristic mode at the moment. Thoose that want it can do it off board in HA, NodeRed, python etc.

revk, 2024-04-05:
Tricky as we send HA config at start and don't update when you turn on/off faikin auto mode. Not sending once we have told HA we have it just causes error messages.

I'll make it a setting.

44616E, 2024-04-05:
> Tricky as we send HA config at start and don't update when you turn on/off faikin auto mode. Not sending once we have told HA we have it just causes error messages.
> 
> I'll make it a setting.

Fair enough, haven't used HA so sorry not aware of these issues.

closed 2024-05-06 by revk