revk/xESP32-Faikout #302: [FEATURE] set the actual temperature in a room by mqtt or command

hokreb, 2024-04-15:
I use the home assistant system for controlling my prefera daikin multisplit system with five indoor units and one outdoor unit (with faikin modules :-)) . I see, that there is already a kind of "custom" faikin mode available, where the heating/cooling is controlled by an BLE temperature device, so that not the in device temperature sensor is use, which does not give you the actual room temperature. 

Is it possible, to send the actual temperature of the room by an mqtt message? I have already some 866 mhz temperature sensors, where I get the acutal temperature every 10 seconds, so instead of using the ble devices, I could also easily send the actual temperature eg as mqtt. I looked into the wiki, but doesn't find any hint regarding this.

revk, 2024-04-15:
Yes, there is a message for that, documented. "control"

hokreb, 2024-04-15:
I did not found the command for this, maybe I have not good explained, what I would like to set. Currently there will be reported a temperature value "home", which reflect the temperature the aircon currently sees as the room-temperaure. I would like to overwrite this value with an own measured value of the room temperature. Then this value will be use to check, if target temperature is reached. I tried to set the home value by command/faikin-name/home, but this does not change any value.

revk, 2024-04-16:
Sorry, yes, this is in https://github.com/revk/ESP32-Faikin/blob/main/Manuals/Advanced.md but was not that clear. I have added some more detail, but it did say this:-

[image]

Basically sent a `control` message with a JSON payload containing the temperature reference to use in `env`.

closed 2024-04-16 by revk