revk/xESP32-Faikout #305: FXDQ32A3VEB  (FXDQ-A3) doesn't have X50A or S21 port (wiring diagram attached)

alexcf, 2024-04-18:
I've got a FXDQ32A3VEB (i think 4 actually!), which are connected to a Madoka (piece of shit*) controllers. Looks like they're using the Diii-Net method to connect them.

Looking at the board, I've got an X35A for power, but no X50A for comms. The notes say: X24A (wireless controller), X33A (adapter for wiring) and X38A (multi-tenant connector) - is X24A interchangeable with the S21? Pic attached.

\* why anyone in their right mind would think building a BLE based controller was a good idea needs a good talking to.
PS. RevK - great work :)
[image]

geni0602, 2024-04-18:
Did you try to measure the voltage on the x24a? 
https://github.com/revk/ESP32-Faikin/wiki/Wiring#connector
Do you get 5v pin1 and 12-14v Pin 4 GND is pin 5?

alexcf, 2024-04-18:
OK, so closer inspection on the board (also finding out that my multimeter is broken), I noticed there is an X50A but it's not listed on the wiring diagram. I'm still going to test the voltage on those pins though, but... After a bit more digging, it looks like the X24A is designed for the infrared remote control, so generally just a receiver :(

alexcf, 2024-04-27:
Confirmed all working!

closed 2024-04-27 by alexcf

alexcf, 2024-04-27:
I was wrong. I couldn't get this working. It seems like the X50A isn't wired in. :(

yiannosi, 2024-10-19:
Did you find a solution? I think I am facing a similar problem..