revk/xESP32-Faikout #311: [BUG] MQTT control temp changes other settings as well

miniluma, 2024-04-26:
I have Faikin up and running. AC is on in heat mode. Daikin fan is set to night mode and Faikin shows Fan in "Auto" (as known).

In the Webinterface: When I change the temperature, the temp is modified only - as expected.

The Bug:
When I send "control {"temp":[value]}" with MQTT, Faikin changes the temp, but seems to (at least) also send the fan mode to Daikin. (I see "Auto" in the Daikin app)

Is this intended behavior? I'd expect it modifies only the control sent in the mqtt command.

revk, 2024-04-26:
That is unexpected. I'll investigate.

revk, 2024-04-26:
Ah hang on.

I'll look, but because the failing can't tell it is night, if it has to send a message that includes the fan mode, it would have to send auto, as that is all it knows.

Now, I am not sure without looking, but it may be temp and fan mode are in the same message.

If so, it would be unavoidable.

We have looked to see if any way to tell "night" from auto, and I wish there was. Just weird of Daikin.

miniluma, 2024-04-26:
Ok, I double checked. It's the same when I so it from the Webinterface. It also sends "auto".

What if you check the fan RPM? To handle different indoor units properly, maybe with a parameter like "fan rpm in night mode"? (e.g. my Perfera 2kW indoor units has 590 +/- 10 in night mode (and 690 in Level 1) and my 5kW indoor has 730 in night mode, so I think it can't be hard coded but it's also hard to guess...)

But this may lead to unwanted behavior if auto mode is going down to the same rpm as night mode...

revk, 2024-04-26:
I tried fan rpm as a guide, in fact I think that is in the code now, but it's not reliable at all.

closed 2024-04-26 by revk

miniluma, 2024-04-26:
One more question just a bit related to this: Have you sniffed only the S21 to reengineer the communication or did you also use the connection between the AC and the original Wifi module?
It looks like this is also just a serial connection and the original wifi module can handle night/auto fan.

revk, 2024-04-26:
I did monitor the original and did not find any way it indicated night mode back to the control.

I assumed the control just assumed it had set night and stayed like that, which would not work if you used some other means too change the mode.

If someone can confirm if the wifi remote Daikin do can tell it is in night mode, i.e. change to night mode on IR remote, and wifi module say it is night mode, then it is well worth investigating more.

miniluma, 2024-04-26:
I can confirm. When I change the fan speed on the IR remote, the App detects properly if it's auto or night acc. to what I set.

revk, 2024-04-26:
Hmm, OK there is a `dump` setting which should show all S21 registers we know of, on MQTT.

If you can do that and see if any value changes when you change to auto and to night, that would be helpful.

miniluma, 2024-04-26:
If I see it correctly at the end of the app, the app holds 3 values:
- Mode: auto, quiet and fixed
- Speed: auto, quiet and 1-5
- Level: 1-5 (depending on the last speed while mode was "fixed")

miniluma, 2024-04-26:
> Hmm, OK there is a `dump` setting which should show all S21 registers we know of, on MQTT.
> 
> If you can do that and see if any value changes when you change to auto and to night, that would be helpful.

I tried, but nothing new. It sends 'A' in G1 also in night mode as you already found out.
That's why I asked for the connection/communication of the original wifi module. ;-)

revk, 2024-04-26:
Only other way is to snoop the original module. Are you in the UK?

miniluma, 2024-04-26:
[L1 to Quiet.log](https://github.com/revk/ESP32-Faikin/files/15129758/L1.to.Quiet.log)
[L1 to Auto.log](https://github.com/revk/ESP32-Faikin/files/15129759/L1.to.Auto.log)

I recorded the output after changing from Level 1 to auto and L1 to quiet mode.
I cut the part of the last G1 with L1 until I saw G1 with A for the second time.
I notice more messages after changing the fan mode compared to normal... But it's hard to "read" for me.
Do you see something interesting maybe?

miniluma, 2024-04-26:
Could you remove all known messages and payload? If there is something remaining, I can make more tests and focus on the unknown parts.

revk, 2024-04-26:
Wow, my mqtt logging usually shows on one line, which makes the changes way easier to see!
Ah, I may have told you `dump` not `debug`.

Debug shows like this, which is easier to compare. Try that. Turn off `dump`.

```
2024-04-26 12:39:37.890041:info/GuestAC/s21 {"G1":"03@A","G2":"=;\u0000\u0080","G3":"000\u0000","G4":"0\u00B0\u00A00","G5":"000\u0080","G6":"0008","G7":"00@I","G8":"0200","GA":"A2D0","GB":"0330","GC":"57D0","GG":"0000","GK":"q{51","GM":"3127","GN":"0000","GP":"5A00","GQ":"1D00","GS":"0000","GT":"5000","SH":"012+","SI":"522+","Sa":"511+","SL":"000","Sd":"000","SN":"500-","SM":"500-","SX":"081+","SD":"000"}
```

https://github.com/revk/ESP32-Faikin/blob/main/Manuals/S21.md shows what we know.

I am not seeing anything obvious.

I have a module with S21 both sides, designed to snoop on protocol, but would be easier to send to UK...

miniluma, 2024-04-26:
Ok. I repeated with dump and a more single line style.
But I see nothing as well. :-(
[image]

revk, 2024-04-26:
I think you will find `debug` way easier than `dump`, try it.

MUN0X, 2024-04-26:
G7 would make sense wouldn’t it? 
Its where demand control and powerful are reported as well. All related to the power uptake of the unit…

revk, 2024-04-26:
Well yes.

My tests do not show a change.

Do yours, consistently?

miniluma, 2024-04-26:
[fan mode and switch off.log](https://github.com/revk/ESP32-Faikin/files/15134724/fan.mode.and.switch.off.log)

I retried a bit and switched fan modes. Can't see something relevant. I eliminated unchanged messages already.

The AC turned off at 21:30. The last byte of G7 is doing something interesting here. It's maybe related to pressure or rpm or something. It goes from 'R'/'S' (82/83) to '@' (64) within ~3:30 minutes.

What is the second byte of "GG"? I see it counting up when I change fan mode with the IR Remote.

revk, 2024-04-27:
That S21 document on the main page is what we know/use at present.