revk/xESP32-Faikout #313: AC specialists in UK who can install Faikin?

robimart, 2024-04-27:
Hi all

Slightly off-topic but I was wondering if anyone knows of AC specialists (preferably in London, UK) who can install and help configure Faikin units? I have 6 Daikin units and would love to integrate them with my Homebridge setup but I’m wary of electrocuting myself and need some paid help. 

Grateful for any tips!

robimart

alexcf, 2024-04-28:
It's honestly not that difficult once you've had a look around. Find out the model of your Daikin units, then take the panel off and take a photo of the wiring/circuit diagram, and the board itself. :)

robimart, 2024-04-28:
Thanks Alexcf, a bit of positive encouragement can go a long way! :)

The main problem is that the units are very difficult to access so I can’t easily poke around. On one of the units I was able to locate the sticker on the third photo. Appears it’s FXDQ20M8V3B // R410A. Hmm….this doesn’t appear on the “known working models” list on this website.  Anyone know whether this _should_ work and is worth a try?

[image]
[image]
[image]

alexcf, 2024-04-28:
Hard to say without taking off that panel where the model number is (carefully!) to see if there's an X50A or an S21. 

From experience, and a quick google I reckon you might be out of luck. 😐 

That said, I found this, which would be a stretch even for me! https://mattala.com.au/2016/08/14/airduino-network-enabling-the-daikin-vrv-with-the-brc1d61-or-similar-thermostats/

Here's the wiring: http://www.daikintech.co.uk/Data/VRV-Indoor/FXDQ-M/2007/FXDQ-M8V3B/FXDQ-M8V3B_Databook.pdf

robimart, 2024-04-28:
Thank you so much for helping out, it does indeed look like I’m out of luck. Those Arduino PCB hacks are definitely a challenge too far for me!

I think what I’ll need to do, is replace the the wall thermostats with ones that have inbuilt WiFi or Zigbee and enable remote control through that route. I’d previously ordered some weird stuff from China through AliExpress which didn’t work but I’ll give it another go.

One thing I’ve never understood is ModBus. Should this allow any thermostat/controller that’s connected via P1/P2 cabling to control not just the AC it’s connected to but also ALL other AC units? All very confusing…

Thanks again!

alexcf, 2024-04-28:
One thing I did look at, although this might be out of scope - https://github.com/Arnold-n/P1P2Serial if you email Arnold with your model he might be able to provide some input.

robimart, 2024-04-28:
On first glance this looks exactly like what I need!! Hopefully compatible with my Daikin units and hopefully I can sleeve this in alongside my existing wall controllers. 

Really appreciate you pointing me in the right direction!

closed 2024-05-06 by revk