revk/xESP32-Faikout #315: [FEATURE] Esp ethernet instead of wifi?

devironglove, 2024-04-29:
**Daikin hardware**
Currently running 4 units of FTXM20R with remote only (no app)

**Is your feature request related to a problem? Please describe.**
No problems related I could find :)

**Describe the solution you'd like**
For parenting and health reasons, we would like to try to control our units via ethernet (sockets present) Wifi is too unreliable in the house (big house) and another solution would be tonextend the current network (meaning one more AP + energy cost) 
This isn’t a solution as the HVAC need to run any time in the day and night (night = wifi down)

**Describe alternatives you've considered**
Currently running a zigbee/zwave mesh network but couldn't find any way to control the units via any of them…
For my solar production, I’ve used a w32-eth01 with two uarts for reading my production and pushing to local network (wired)
Do you think this could be possible with this project?

MassiPi, 2024-05-02:
it is not likely to work "out of the box", but sometime ago i built a controller based on ESP8266 and a W5100/W5200/W5500-based device with great results. I think arduino code also did a lot of step forward, so i think you can try. the problem i see is that faikin hardware does not expose other ESP32gpio, and the ethernet adapter is SPI based, probably it will be easier for you to build your own hardware..

revk, 2024-05-06:
I think ESP32 can do ethernet, but I have never done that.

Why is WiFi unreliable? I have a reasonably big house, with 6 access points and all works well.

Sonic-Amiga, 2024-05-07:
WiFi can be unreliable, i confirm. It depends on configuration of your living space plus on your wish to put access points here and there. Personally i have a corner in my apartment, which is barely reachable; and one just may be too greedy to buy another AP just to cover a single remote corner to support a single device. On the contrary, cables cost almost nothing; and if you're modeling your space from scratch, cable is a good option, nice to have.

devironglove, 2024-05-07:
This is exactly my situation :) Just one unit is barely reachable by current AP, and moving the AP ain't a solution because other side of the house would be covered.
Second AP seems overkill for just so small coverage added.
I might eventually look at making my own hardware based on this repo and try it out :) But first things first (renovation :D)

njh, 2024-05-08:
My preference is to always use wires, where possible, and to fall back to wireless, if the device is mobile 🙂

I had considered making a Modbus network out of my air conditioning units, and then having a some kind of device bridge from Modbus to Ethernet.

There are third party S21 to Modbus adaptors available (for a price!):
https://www.bms-systems.co.uk/Content/Images/uploaded/Datasheets/Moralle/DA001-Data%20sheet.pdf
I wonder if they reverse-engineered the S21 protocol too, or had help from Daikin.

DVTrast, 2024-08-01:
I´m also very interested in an ethernet-variante of the faikin (without WiFi).

revk, 2024-08-01:
I'd need to sort an ethernet board (test/dev board) and then a design for one. Will take a while, sorry.

closed 2024-08-01 by revk