revk/xESP32-Faikout #317: FTXM28RVLT, FTXM36RVLT, and FTXM60SVLT Experience

bai-yi-bai, 2024-05-01:
Hello,
I wanted to start an issue to share my experience of using ESP32-Faikin with Daikin models which I have not seen documented yet, likely because these models are not targeted towards North American or European markets. 

The models in question are:

* FTXM28RVLT (x2)
* FTXM36RVLT
* FTXM60SVLT

I recently began building up a expansive home automation system and wanted to unlock their potential, especially to better control humidity.
Yesterday, I received a batch of five ESP32-Faikins (2024-03-28) I had assembled by JLC PCB and successfully installed the firmware images using an old ESP8266-01 UART programmer with a 10k resistor from GPIO0 to ground.
`python3 esptool.py -p /dev/ttyUSB0 write_flash 0x0 Faikin-S3-MINI-N4-R2-bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 Faikin-S3-MINI-N4-R2.bin`

I did not verify that these units were actually compatible with ESP32-Faikin, but this community seems very supportive, so I went ahead anyway.
I have had these units for approximately four years and found that sourcing the official wireless adapter from third-party websites was no longer possible.
I did find a thread in Chinese discussing these units and the Wi-Fi adapter, which dates back to 2018, [mobile01](https://www.mobile01.com/topicdetail.php?f=731&t=5395516)
I still have the original documentation which came with the units.

The FTXM28RVLT and FTXM36RVLT installation manual, 3P364521-4L M15N030C, mentions how to attach a "HA System" to the S21 port.
This manual also applies to FTXM22NVLT, FTXM28NVLT, FTXM36NVLT, and FTXM22RVLT.
I found no issues regarding any of these six units.
I'm mentioning them in this issue if someone in the future comes looking for information on these other units.

Now, I must begin the process of sourcing the required JST connectors and crimping tools.

### FTXM60SVLT

This leads me to the trouble of the FTXM60SVLT's installation manual, 3P492258-14F M17N027a.
This manual also applies to FTXM41SVLT, FTXM50SVLT, and FTXM71SVLT.
I found no issues regarding any of these four units.
This manual mentions attaching a "HA System" to the S403 port with an "HA board (option part)" which has an S21 port. 
There is no part number for this HA board mentioned in the installation manual.

I also carefully detached the main control board to get a better photo of it and to inspect it for possible alternative connection points.
I had a limited amount of time, and I was getting quite hot going up and down my ladder disassembling the unit, so I did not meter out where all the connections went.
[image]
[image]
I did re-assemble the unit and it worked fine.

Luckily, there are many issues on this project which details the S403 pin out and potential connectors to use.

[S403](https://github.com/revk/ESP32-Faikin/issues?q=S403)

For now, there's no real "issue" to report. 
I simply wanted to document this information for others.
Hopefully, they will work with the already massive amount of functionality.

Now to source the connectors and get them all up and running.

Sonic-Amiga, 2024-05-07:
Hello! https://github.com/revk/ESP32-Faikin/issues/208 tells everything. But !!! BE CAREFUL !!! and know what you're doing because S403 port is NON-ISOLATED, this means that there's an AC mains voltage between any pin and ground, and THAT GROUND MAY BE YOU .

bai-yi-bai, 2024-05-24:
FTXM28RVLT confirmed working. Thanks to @reedy for updating Wiring.md with the exact part numbers.

bai-yi-bai, 2024-05-25:
My FTXM36RVLT is confirmed working.

I guess I will add FTXM28RVLT and FTXM36RVLT to [https://github.com/revk/ESP32-Faikin/wiki/List-of-confirmed-working-air-con-units](https://github.com/revk/ESP32-Faikin/wiki/List-of-confirmed-working-air-con-units).
Both used EHR-5 connectors with SEH-001T-P0.6 crimps.

My FTXM60SVLT not working yet, because I bought the wrong connector for the S403. This model does not accept an **XAP-10V-1 2.5mm pitch 10PIN** connector.

- JST PAP-10V-S connector
-- SPHD-001T-P0.5 crimp (26～22 AWG)
-- SPHD-002T-P0.5 crimp (28～24 AWG)

bai-yi-bai, 2024-06-08:
FTXM60SVLT confirmed working.

JST PAP-10V-S is not the correct connector for the FTXM60SVLT. I had to trim off the retaining clip. It fits snugly enough.

[image]
[image]
[image]

closed 2024-06-08 by bai-yi-bai