revk/xESP32-Faikout #318: [FEATURE] Configure state_class on sensors for HAss stat collection

sm-Fifteen, 2024-05-03:
**Daikin hardware**
S21, FTXS12LVJU

**Is your feature request related to a problem? Please describe.**
The Home Assistant does not collect long-term stats on any of the sensors by default because the MQTT sensor discovery definitions do not specify a [state class](https://developers.home-assistant.io/docs/core/entity/sensor#available-state-classes).

Stat collection enables a number of nice features in Home Assistant, like moving window averages on discrete value charts, loading graph data at lower density than native, and visualization of long-term trends on sensor readings.

**Describe the solution you'd like**
Simply add the [`state_class` key](https://www.home-assistant.io/integrations/sensor.mqtt/) to the liquid and outdoor temperature sensors (`measurement`), power usage (`total_increasing`) and fan and compressor frequency (`measurement`), to indicate that HAss may collect statistics on these measurements.

**Describe alternatives you've considered**
Not doing anything isn't a total dealbreaker, it just means no statistics are kept beyond the full-resolution window of 10 days. For power usage especially, that could be annoying.

The components of the HVAC/Climate entity (indoor temp, target temp, hvac mode, hvac action, etc.) are not considered "sensors", and I'm not entirely sure of how statistics for these work.

revk, 2024-05-03:
OK is that `"state_class":"key",` in the JSON?

sm-Fifteen, 2024-05-03:
> OK is that `"state_class":"key",` in the JSON?

It exists at the same level as `device_class` and `unit_of_measurement`, yes.

revk, 2024-05-03:
Ok I have issued a beta, try that.

sm-Fifteen, 2024-05-03:
> OK is that `"state_class":"key",` in the JSON?

Oh, I misinterpreted what you meant by that, `state_class` needs to be valued as `measurement` for the outdoor temp, liquid temp, humidity and frequency sensors, and as `total_increasing` for the power metering sensor (since it's understood to be monotonic, being the lifetime power consumption of the unit). Battery percentage, I would assume would also count as `measurement`.

revk, 2024-05-03:
OK, that is kind of why I asked!

I'll make a new beta now, let me know.

sm-Fifteen, 2024-05-03:
Will test when I get back home.

Note that I don't have power monitoring available on my units, so I won't be able to test that part myself, but based on HAss doc, it should behave properly. `total_increasing` assumes that the number can only go down if it means the counter has reset or wrapped around, which is how FM seems to work so far. The [documented 10% tolerance](https://developers.home-assistant.io/docs/core/entity/sensor/#state-class-total_increasing) should be enough to avoid issues around [slight dips like the one in the top right of this chart](https://github.com/revk/ESP32-Faikin/issues/101#issuecomment-1722154007).

sm-Fifteen, 2024-05-03:
All four sensors I have (liquid, fanfreq, comp and outside) correctly registered as statistics as soon as the controller rebooted post update, and from a quick test in fan mode, they all get logged as I would expect them to.

That solves the issue for me. Thank you, revk.