revk/xESP32-Faikout #319: HomeAssistant: fanrpm -> fanfreq

Sonic-Amiga, 2024-05-03:
Hello there! Long time no see...
My user has complained about missing fan speed reporting to HomeAssistant in my recent update. I myself don't use HA, but i suspect this has to do with changing reported item name from "fanrpm" to "fanfreq". Yes, i see that the number is also treated differently.

The change originates from commit cc59320a895206aa3a7500ab34bb05ee8bc75a7d , with the ultimate explanation of "checkpoint". Could you explain a bit more ?

I know that send_ha_config() also report the item under a new name. I guess this is only queried by the HA once, when the hardware is configured. Is there a quick way to update this information in HA ? Sorry for possibly lame question; i'm not a HA user.

revk, 2024-05-04:
Not sure how HA caches things, maybe restart HA or the MQTT server.

Sonic-Amiga, 2024-05-05:
How did this go on your system ? Or don't you use HA ?

revk, 2024-05-05:
I don't use HA

Sonic-Amiga, 2024-05-05:
Okay... In the meanwhile i found some history in issues... Is this change related to https://github.com/revk/ESP32-Faikin/issues/84#issuecomment-1796272745 ?

revk, 2024-05-06:
Yes, it was changed, and that thread shows why. You may have to clear something in HA or MQTT to get it to see the change properly.

Sonic-Amiga, 2024-05-06:
Okay, thanks for explanation