revk/xESP32-Faikout #322: [BUG] "on/off" icon isn't displayed correctly by a web browser

Sonic-Amiga, 2024-05-05:
I see you've switched from text to icons again; I understand that it should be a standard on/off symbol, but at least on Windows browsers can't render it:
[image]
And that's not Firefox-specific issue; i tried Edge; it also looks like a square

Sonic-Amiga, 2024-05-05:
Tested on Firefox for Android; the same thing

revk, 2024-05-06:
There is a setting now. Have a look.

closed 2024-05-06 by revk

Sonic-Amiga, 2024-05-06:
You mean "noicons" setting ? I know, but it looks boring without all the icons.

revk, 2024-05-07:
You want some of the icons but not all!!!

Isn't Unicode fun.

Sonic-Amiga, 2024-05-07:
Yes. And i'd say showing text instead of this particular icon should be the default; because i suggest there are plenty of users with not-so-cutting-edge systems around; who will see the empty square. It definitely feels like a bug, even to me, who knows the drill.

I replaced icon back with text in my port; but yours is subject too; plus it's yet another unwanted diff. I'd like to be as upstream as possible, this simplifies things a lot.  I even have a dream of complete merge; but currently amount of #ifdef's would still be rather large; you wouldn't be happy with such a merge, neither it would be easy to maintain.