revk/xESP32-Faikout #323: Room Temperature with I2C

nemonau, 2024-05-05:
Hello and thank you for your Work with the nice Faikin Project.

At the weekend, I cleaned my Daikin indoor units. 
I noticed that an ALPS HSHCAL101 is used to measure the room temperature.
This uses 4 wires (0V, 5V, SCL and SDA) and therefore supplies the measured values via I2C.

https://tech.alpsalpine.com/cms.media/product_spec_hshcal101b_en_a900757bb4.pdf

As far as I have found out, this sensor is used in many Daikin devices.
Would it therefore be possible to transmit the room temperature (e.g. Xiaomi or another value from HomeAssistant) to the Daikin device via the I2C connections of the Faikin/ESP32?

Unfortunately I am not really familiar with ESP32 and I2C, but I have an oscilloscope with decoding function for I2C, if that helps in any way:-)

Thank you very much

Greetings Nino

Translated with DeepL.com (free version)

revk, 2024-05-06:
That is interesting...

The current hardware design would not do it, and we need to allow for the voltages. And also confirm if the I2C is isolated from mains levels.

I'll have to check if ESP32 can be an I2C slave. I have no idea. If it can then that is an interesting possibility.

revk, 2024-05-06:
The first good news is "A single I2C controller can be a master or a slave" which suggests it is capable of being a slave.

nemonau, 2024-05-25:
Please excuse the delay

Today I measured the voltage of the individual pins against earth (PE).
Each of the four measurements showed exactly the same voltage of around 80VAC and -12VDC.

[image]

With the Low Z setting (3kOhm impedance) the multimeter showed 0VDC and 0VAC for the same measurement.

I think I'll take a chance next week and connect the oscilloscope to the i2C data line.
Unfortunately I don't have a differential probe for the oscilloscope, so wish me luck :-)

nemonau, 2024-06-06:
ok it didn't blow up:-)

I temporarily extended the I2C cable by 2m to make it easier to measure.
From the waveform on the oscilloscope, it is clear that I2C is only designed for short distances.
But at least we recognize a 120kHz clock signal and some values.
And then the same numbers always follow, R&W 18 and 68.
Unfortunately I don't know what this means. The display range on the oscilloscope is also very small, it would be great if someone had a logic analyzer and could take a closer look.

[image]
[image]

wurmtal868, 2024-06-15:
Just for my understanding
You are want to disconnect the local APLS sensor and use the input to feed in BLE temp of the FAIKIN module via I2C?

nemonau, 2024-06-18:
> Just for my understanding You are want to disconnect the local APLS sensor and use the input to feed in BLE temp of the FAIKIN module via I2C?

if that were somehow possible, that would be awesome:-)

unfortunately i have no experience with the programming of an esp32 or how the i2c communication is structured.

i have now ordered an I2C extender from adafruit for little money, so i should be able to extend the cable between the temperature sensor and the indoor unit by a few meters.
I live in an apartment with a flat roof and it gets quite warm directly under the ceiling where the indoor unit is mounted, for this reason the control is quite inaccurate.

As a test, I extended the temperature sensor with a simple cable and it now hangs 1m below the indoor unit.
The temperature control is now much better.
As soon as the I2C extender arrives, I will mount the sensor a little further away.

Many thanks

revk, 2024-08-18:
I think this is something of a separate project, not out of the question, but probably not an issue to leave open here.

closed 2024-08-18 by revk