revk/xESP32-Faikout #324: Rename 'ESP32' to 'ESP'

Sonic-Amiga, 2024-05-05:
Would you mind renaming the folder within your source tree ? I did that when updating my port back in February; because it felt wrong to reference ESP32 in 8266 port. But now i cannot cherry-pick because of path difference; i have to compare files side-by-side and pick changes manually. It's very inconvenient process.

closed 2024-05-06 by revk

Sonic-Amiga, 2024-05-06:
Sorry, what does this mean ? "No i won't rename" ?

revk, 2024-05-07:
Did you look at the name of the project now?

revk, 2024-05-07:
Oh...

revk, 2024-05-07:
Done.

Sonic-Amiga, 2024-05-07:
Thank you, i reply in evenings, just made it to the keyboard. Yes, i meant exactly this - renaming a directory inside the project, not the project itself. Because git only sees and cares about the former. Now i can do cherry-picks back and forth again.