revk/xESP32-Faikout #325: [FEATURE] MQTT Topic name

lorbetzki, 2024-05-05:
Hello,

at the moment we have this mqtt situation:

command/<hostname>
setting/<hostname> 
event/<hostname> 
info/<hostname> 
error/<hostname>
<mac of faikin> # livestatus
homeassistant/*/<mac>

I would appreciate it if you could specify a sendtopic by yourself instead of hostnames or mac. This would simplify filtering immensely.

for example

<topic>/command/
<topic>/setting/ 
<topic>/event/<hostname> 
<topic>/info/<hostname> 
<topic>/error/<hostname>
<topic>/livestatus/
homeassistant/*/<topic>/

what would you think of this?

revk, 2024-05-06:
This would be a RevK library feature not Faikin specific.

Not impossible. Would need to be a setting.

closed 2024-05-06 by revk

revk, 2024-05-06:
Library update in latest beta, new setting, please try.