revk/xESP32-Faikout #329: Powerful mode setting/reporting does not work

kjoconnor, 2024-05-15:
I bought my Faikin from Amazon pre-built and I'm currently connected to a CTXS07WVJU9 via S21.
[image]
[image]

**Bug**

I can not set Powerful mode, nor is Powerful mode changed in the web UI or MQTT when it is manually set on with the wireless remote.

revk, 2024-05-15:
Can you enable debug messages to MQTT and that may show if we are sending the right S21 messages.

kjoconnor, 2024-05-15:
I think was the command that was sent:

```
{"protocol":"S21","cmd":"D3","payload":"30303032","text":"0002"}
```

This popped up as well in MQTT:

[image]

And finally I can see powerful being `0.35` here, but not sure what that means.
```json
{"protocol":"S21","ts":"2024-05-15T05:23:01Z","online":true,"home":[20.00,20.37,20.50],"heat":true,"fanrpm":[480,504,690],"comp":23,"outside":15.50,"liquid":[26.00,29.21,32.00],"anglev":5,"power":true,"mode":"H","temp":21.00,"fan":"A","swingh":false,"swingv":false,"powerful":0.35}
```

revk, 2024-05-15:
That is a periodic summary and means 35% of the period it was in.

MassiPi, 2024-05-15:
are you sure your unit's S21 supports setting and reporting powerful (and eco, i'd say) mode?
not all units support that, for example with my units i can set them via the remote controller, but not via S21 (nor they are reported via S21)

Utopia69, 2024-05-15:
Same here, powerful is not working on unit: FTXS35G2V1B

MassiPi, 2024-05-15:
> FTXS35G2V1B

so i'm not the only one out there with this old-but-gold unit :)
sadly it also lacks power consumption.. but on the other side if you ask Daikin it isn't even supported by any wifi module..

jfinnie, 2024-05-17:
I've got an ancient FDXS50C7VMB (13year old!) that also doesn't seem to have working POWERFUL reporting or enabling from the Faikin, Faikin-S3-MINI-N4-R2: 729c466 2024-05-14T08:53:28.  

Additionally I note that setting nopowerful in the advanced settings doesn't seem to remove Powerful button from the UI.  Noswingh and noswingw did do as expected, and remove those options from UI (it's a ducted split, so no swing vents).

Anything useful I can provide?

Utopia69, 2024-05-17:
tested also on FTXS25DAVMW (2007) & FTXS35DAVMW (2007), they both have powerful as feature, but isn't working on Faikin
Same problem as my FTXS35G2V1B (2008)

mmmgggttt, 2024-07-08:
same here, here is mqtt dump
[image]

revk, 2024-07-08:
If the daikin does not accept it, is there a different S21 command you know of?

mmmgggttt, 2024-07-08:
how can i see / change the command ? I have a remote , should i start by logging mqtt when sending the code from remote ?

revk, 2024-07-08:
The protocol is all guess work, and works on lots of units.

It may be yours does not do powerful via S21, or is somehow different.

Trial and error.

There are debug and dump settings that send details to MQTT, and a command to make new messages to send.

revk, 2024-08-18:
Did we get anywhere with this, we have code to do `D3` and `D6`

However, there is ongoing work to debug more of S21 protocol which may help resolve this. I'll close this as there is another issue covering it now.

closed 2024-08-18 by revk