revk/xESP32-Faikout #332: Automatic control means some functions are limited.

eshroom, 2024-05-16:
**Faikin hardware**
Purchased on Amazon Faikin-S3-MINI-N4-R2

**Daikin hardware**
X50 lead with FNA25A

**Describe the bug**
I have 7 units and 7 faikins. One of them has caused some issues. At first when on firmware b3be82a+ it was hard to get it to connect. It was accessible on the local address, but showing as disconnected with the unit (I tried on multiple other working units). I did get it to recognise the unit eventually, it would turn it on, but whatever temperature was set, it would immediately drop back down to 16 degrees.

On upgrade to firmware 729c466, I saw a big improvement, except the reporting is off. In Home Assistant it is always showing as in Auto Mode, when it is not. On the local page there is a message saying "Automatic control means some functions are limited.", which does not appear with any of the other faikin controllers.

I guess this must be faulty unit rather than a bug, since I have 6 other identical set ups and have tried switching the controllers around, but maybe you can advise?

revk, 2024-05-16:
That message normally means it either has "faikin auto" turned on, or is receiving messages to control it from something remote over MQTT (unlikely if you did not set that up). In such case it would report auto and it would set high and low temperatures to control to the external reference.

Is "faikin auto mode on" - maybe screenshot the main control page?

eshroom, 2024-05-16:
Problem solved. I must have turned on faikin auto mode while I was trying to get it working initially. Thanks.