revk/xESP32-Faikout #334: [BUG] Current case seems to have a float/detached part

reedy, 2024-05-17:
**Faikin hardware**
N/A

**Daikin hardware**
N/A

**Describe the bug**
Using https://github.com/revk/ESP32-Faikin/blob/48c1db8/PCB/Faikin/Faikin.stl / https://github.com/revk/ESP32-Faikin/blob/48c1db8/PCB/Faikin/Faikin.scad it seems there's floating plate that's not attached to anything

OpenSCAD:
[image]

BambuStudio:
[image]

PrusaSlicer:
[image]

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Import the STL into a slicer, or the scad into OpenSCAD.
2. Look at the generated object..

**Expected behavior**

The part not to have an apparent floating/not attached part.

I could print with supports, but that seems wasteful, as it's not attached (is it supposed to be?). If it's necessary, it could be printed directly on the bed. But either way, it's unclear where it goes/what it would be used for.

The other part has something similar, but it is attached at either side of the model.

**Additional context**
n/a

Thanks!

reedy, 2024-05-17:
It seems to be a regression from 48c1db8e6531d6a6cf116dbf3815d301777cbac3

If I use the file from before that, https://github.com/revk/ESP32-Faikin/blob/aa194f032697c8fcc73cf78b59b8a794f2c04b7c/PCB/Faikin/Faikin.scad or https://github.com/revk/ESP32-Faikin/blob/aa194f032697c8fcc73cf78b59b8a794f2c04b7c/PCB/Faikin/Faikin.stl it seems to be attached as one would expect:

[image]

[image]

revk, 2024-05-17:
I think it is an anomaly, infinitely thin, and my slicer does not even see it, just delete if yours does.

It is automatically generated so the "bug" would be in the PCB case tools, not here.

closed 2024-05-17 by revk

reedy, 2024-05-17:
Fair enough.

If we split it into parts, we get something like this, which is trivially removeable from the print.

[image]

[image]