revk/xESP32-Faikout #335: Ota updates don't work when fixed ip is enabled

Utopia69, 2024-05-18:
Hi,
I'm running 5x Faikin-S3-MINI-N4-R2. 729c466 2024-05-14T08:53:28 S21 
When I insert a fixed ip ota updates check fails.
{"url":"http://ota.revk.uk/Faikin-S3-MINI-N4-R2.bin?XXXXXXX","fail":-3}

[image]

With dhcp there is no problem.
I also tested to change 1 dns server to an ipv6 dns, this gives no solution.

Grtz,

revk, 2024-05-18:
Interesting, I wonder if there is an issue with DNS being set, I'll have to look.

Utopia69, 2024-05-18:
> Interesting, I wonder if there is an issue with DNS being set, I'll have to look.

Without dns servers problem is also there (but this can be normal without dns  servers)

revk, 2024-05-18:
Yes, looks like DNS issue. I have done a beta release, please test.

closed 2024-05-18 by revk

Utopia69, 2024-05-18:
Yup, beta is working
Thx