revk/xESP32-Faikout #336: [FEATURE] FTXM-R streamer

luca-iodice, 2024-05-19:
Is it possible to set Streamer setting via S21? Anyone could guide me in order to dump the proper setting?

Thank you

luca-iodice, 2024-05-23:
For now I use this code in configuration.yaml, but I think that MQTT auto discovery could be implemented for streamer, control, etc.

mqtt:

  - switch:
        unique_id: "Sala1_streamer"
        name: "Streamer"
        command_topic: "command/Sala1/control"
        state_topic: "state/Sala1/status"
        value_template: "{{ value_json.streamer }}"
        state_on: true
        state_off: false
        payload_on: '{"streamer": true}'
        payload_off: '{"streamer": false}'
        device:
            identifiers: "34B7DA587C00"
            name: "Sala1"

Nick-Sills, 2024-06-13:
@luca-iodice  I have noticed the same issue for my AC units.

@revk  I've manually added the following MQTT discovery payloads to achieve the switches below. 
**Note** These switches match the native Daikin integration.

[image]

**Note:** Streamer doesn't work without faikin changes. See below:

We need two switches to be added to Climate entity in Home Assistant. Streamer and Power:

**Example discovery for power:**

`homeassistant/switch/34B7DA588E44pwr/config`
```json
{
  "name": "Power",
  "unique_id": "34B7DA588E44_power",
  "command_topic": "Faikin/Villacana AC Back Bedroom/command/power",
  "state_topic": "Faikin/Villacana AC Back Bedroom/state/ha",
  "value_template": "{{ value_json.power }}",
  "payload_on": "1",
  "payload_off": "0",
  "state_on": true,
  "state_off": false,
  "retain": true,
  "device": {
    "identifiers": [
      "34B7DA588E44"
    ],
    "name": "Villacana AC Back Bedroom",
    "model": "FTXM20R",
    "sw_version": "579c531",
    "manufacturer": "RevK",
    "configuration_url": "http://Villacana AC Back Bedroom.local/"
  }
}
```

**Example for Streamer:**

`homeassistant/switch/34B7DA588E44strmr/config`

```json
{
  "name": "Streamer",
  "unique_id": "34B7DA588E44_streamer",
  "command_topic": "Faikin/Villacana AC Back Bedroom/command/streamer",
  "state_topic": "Faikin/Villacana AC Back Bedroom/state/ha",
  "value_template": "{{ value_json.streamer }}",
  "payload_on": "1",
  "payload_off": "0",
  "state_on": true,
  "state_off": false,
  "retain": true,
  "device": {
    "identifiers": [
      "34B7DA588E44"
    ],
    "name": "Villacana AC Back Bedroom",
    "model": "FTXM20R",
    "sw_version": "579c531",
    "manufacturer": "RevK",
    "configuration_url": "http://Villacana AC Back Bedroom.local/"
  }
}
```

So...

1. Two new discovery payloads need to be added as the above examples

2. Both the `power` and `streamer` states (`true`, `false`) should be added to: `<BASE TOPIC>/state/ha`.  **Copy the states from: `<BASE TOPIC>/state/status`**

3. Faikin needs to be able to subscribe to: `<BASE TOPIC>/command/streamer` to control the Streamer state over S21 with `"0"` or `"1"` **as strings** .

**Note:** Recent changes have meant that Faikin is already correctly subscribed to: `<BASE TOPIC>/command/power`

If you need any more info thenlet me know.

revk, 2024-06-15:
I'll look in to switches, as an option.

Nick-Sills, 2024-06-17:
@revk Thanks. Let me know when there's a Beta to test.

revk, 2024-06-17:
Might not be today, I have quite a lot of real work!

Nick-Sills, 2024-06-17:
Not a problem.

revk, 2024-06-18:
Just doing a beta, do test.

Nick-Sills, 2024-06-18:
Thanks. 

First pass: 

The new `availability_topic` payloads are causing the `climate`, `power` and `streamer` entities to be unavailable.  If I remove the `availability_topic` from the HA discovery payloads, then the controls are available, but the status for Power and Stream don't seem to be getting read. 

I've got to go offline for most of the rest of the evening, but I'll do more debugging when I'm back.

Nick-Sills, 2024-06-18:
@revk 

1. I can't see any status for `power` or `streamer` in  `<BASE TOPIC>/state/ha`
2. The  `availability_topic`  isn't working. Entities are not available unless I remove the `availability_topic`. I'm not sure why.
3. I don't think the Streamer switch is working, though it's difficult to tell without a status.

Nick-Sills, 2024-06-18:
I think the discovery payload needs to be like this:
(Should be quotes on "true" and "false")
```json
  "availability_topic": "Faikin/Villacana AC Back Bedroom/state/online",
  "payload_available": "true",
  "payload_not_available": "false",
```

**Not:**

```json
  "availability_topic": "Faikin/Villacana AC Back Bedroom/state/online",
  "payload_available": true,
  "payload_not_available": false,
```

Plus:  `<BASE TOPIC>/state/ha` needs `power` and `streamer`

```json
{
  "online": true,
  "target": 21,
  "temp": 21.5,
  "outside": 20.5,
  "liquid": 10,
  "comp": 20,
  "Wh": 881800,
  "fanfreq": 12,
  "mode": "heat_cool",
  "fan": "auto",
  "swing": "off",
  "preset": "none"
  "power": true,        <----- ADD
  "streamer": true,    <----- ADD
}

```

revk, 2024-06-19:
I'll get on to it this morning.

revk, 2024-06-19:
Ok step 1, beta issue in a moment, `ha` becomes `haenable`, and added `haonline` and `haswitches` which turn on these new features. This means beta will work as before unless you turn these on.

Nick-Sills, 2024-06-19:
FYI:  Boolean `true` and `false` can only be interpreted from a json node, however,  `Faikin/<TOPIC>/state/online` is a string, not json, hence use `"true"` `"false"` in the config. 

I think it's more usual to use:  `"Online"`  `"Offline"` anyway

revk, 2024-06-19:
OK, availability should be working in next beta in a few minutes, I'll look at switches next.

Nick-Sills, 2024-06-19:
Faikin-S3-MINI-N4-R2: 4d829a2 2024-06-19T09:41:16 S21

Availability seems good

revk, 2024-06-19:
OK I think I have the logic. What is extra special is that the interpretation of a JSON `true` and `false` for the `payload` works, and understands a value of `true` and `false` in the JSON status, but sends `True` and `False` not `true` and `false`. That is HA being, shall we say, *special*. But accommodated now. `Try next beta.

closed 2024-06-19 by revk

Nick-Sills, 2024-06-19:
Faikin-S3-MINI-N4-R2: 590de8d 2024-06-19T10:04:55

We're very nearly there. MQTT switches are available, and appear to work. 👍 

**MQTT Cosmetic issues:**

**Streamer:**

Change text from 'streamer' to 'Streamer'
Icon should be: mdi:air-filter

**Power**

Change text from 'streamer' to 'Streamer'
Icon should be: mdi:power

This will make this:
[image]

Look like this:
[image]

revk, 2024-06-19:
Hmm.... Ok in next beta, may be a few minutes though. We working state and availability to be cleaner.

Nick-Sills, 2024-06-19:
Let me delete and redicover the entities

Nick-Sills, 2024-06-19:
Still the same I'm afraid

revk, 2024-06-19:
I said its would be a few minutes, making now.

Nick-Sills, 2024-06-19:
> I said its would be a few minutes, making now.

That is not relevant. Please see the previous posts.
<hr/>
Now updated
Faikin-S3-MINI-N4-R2: b9ef904 2024-06-19T11:28:03 S21

Unfortunately, we've lost all status. 
[image]

revk, 2024-06-19:
I tested here and all worked. Can you reload the config somehow in ha? It has changed the topic for ha status, but will have sent a new config.

Nick-Sills, 2024-06-19:
Configs:
**climate**
```json
{
  "unique_id": "34B7DA588E44",
  "dev": {
    "ids": [
      "34B7DA588E44"
    ],
    "name": "Villacana AC Back Bedroom",
    "mdl": "FTXM20R",
    "sw": "b9ef904",
    "mf": "RevK",
    "cu": "http://Villacana AC Back Bedroom.local/"
  },
  "icon": "mdi:thermostat",
  "avty_t": "Faikin/Villacana AC Back Bedroom/state",
  "avty_tpl": "{{value_json.up}}",
  "pl_avail": true,
  "pl_not_avail": false,
  "~": "Faikin/Villacana AC Back Bedroom/command",
  "min_temp": 16,
  "max_temp": 32,
  "temp_unit": "C",
  "temp_step": 0.5,
  "temp_cmd_t": "~/temp",
  "temp_stat_t": "Faikin/Villacana AC Back Bedroom/state",
  "temp_stat_tpl": "{{value_json.target}}",
  "curr_temp_t": "Faikin/Villacana AC Back Bedroom/state",
  "curr_temp_tpl": "{{value_json.temp}}",
  "mode_cmd_t": "~/mode",
  "mode_stat_t": "Faikin/Villacana AC Back Bedroom/state",
  "mode_stat_tpl": "{{value_json.mode}}",
  "payload_on": "1",
  "payload_off": "0",
  "power_command_topic": "~/power",
  "modes": [
    "heat_cool",
    "off",
    "cool",
    "heat",
    "dry",
    "fan_only"
  ],
  "fan_mode_cmd_t": "~/fan",
  "fan_mode_stat_t": "Faikin/Villacana AC Back Bedroom/state",
  "fan_mode_stat_tpl": "{{value_json.fan}}",
  "fan_modes": [
    "auto",
    "low",
    "lowMedium",
    "medium",
    "mediumHigh",
    "high",
    "night"
  ],
  "swing_mode_cmd_t": "~/swing",
  "swing_mode_stat_t": "Faikin/Villacana AC Back Bedroom/state",
  "swing_mode_stat_tpl": "{{value_json.swing}}",
  "swing_modes": [
    "off",
    "H",
    "V",
    "H+V",
    "C"
  ],
  "pr_mode_cmd_t": "~/preset",
  "pr_mode_stat_t": "Faikin/Villacana AC Back Bedroom/state",
  "pr_mode_val_tpl": "{{value_json.preset}}",
  "pr_modes": [
    "eco",
    "boost"
  ]
}
```

**streamer**
```json
{
  "unique_id": "34B7DA588E44streamer",
  "dev": {
    "ids": [
      "34B7DA588E44"
    ],
    "name": "Villacana AC Back Bedroom",
    "mdl": "FTXM20R",
    "sw": "b9ef904",
    "mf": "RevK",
    "cu": "http://Villacana AC Back Bedroom.local/"
  },
  "avty_t": "Faikin/Villacana AC Back Bedroom/state",
  "avty_tpl": "{{value_json.up}}",
  "pl_avail": true,
  "pl_not_avail": false,
  "name": "Streamer",
  "stat_t": "Faikin/Villacana AC Back Bedroom/state",
  "cmd_t": "Faikin/Villacana AC Back Bedroom/command/streamer",
  "val_tpl": "{{value_json.streamer}}",
  "pl_on": true,
  "pl_off": false
}
```

**power**
```json
{
  "unique_id": "34B7DA588E44power",
  "dev": {
    "ids": [
      "34B7DA588E44"
    ],
    "name": "Villacana AC Back Bedroom",
    "mdl": "FTXM20R",
    "sw": "b9ef904",
    "mf": "RevK",
    "cu": "http://Villacana AC Back Bedroom.local/"
  },
  "avty_t": "Faikin/Villacana AC Back Bedroom/state",
  "avty_tpl": "{{value_json.up}}",
  "pl_avail": true,
  "pl_not_avail": false,
  "name": "Power",
  "stat_t": "Faikin/Villacana AC Back Bedroom/state",
  "cmd_t": "Faikin/Villacana AC Back Bedroom/command/power",
  "val_tpl": "{{value_json.power}}",
  "pl_on": true,
  "pl_off": false
}
```

**status** - `state/ha`
```json
{
  "online": true,
  "power": false,
  "target": 21,
  "temp": 23.5,
  "outside": 31.5,
  "liquid": 22,
  "comp": 0,
  "Wh": 882400,
  "fanfreq": 0,
  "mode": "off",
  "fan": "auto",
  "streamer": false,
  "swing": "off",
  "preset": "none"
}
```

revk, 2024-06-19:
The config is all looking at the same topic, and looks right.

Where do you see state/ha ?

Nick-Sills, 2024-06-19:
Faikin-S3-MINI-N4-R2: 2ecf871 2024-06-19T13:06:56 S21

Seems to be working now. Thanks

Can we add the following as well:

Streamer config, add:
```json
  "icon": "mdi:air-filter"
 ```
 
 Power config, add:
 ```json
  "icon": "mdi:power"
 ```

revk, 2024-06-19:
Fair enough, next beta in a minute...

Nick-Sills, 2024-06-19:
Faikin-S3-MINI-N4-R2: ffb0c19 2024-06-19T15:16:01

Everything is working and looking good. 👍  Thanks
I'll do some more checks to make sure, but I think we're good for MQTT

However, I've found a bug with the **HTTP API**. The same switch exists in the Daikin native integration for Faikin, and it tracks the streamer status, but the switch doesn't work:

Turning on the Streamer fails:

`/aircon/set_special_mode?en_streamer=1`

`ret=PARAM NG,adv=Unknown mode`

revk, 2024-06-19:
That's a new one on me, I'll look.

revk, 2024-06-19:
I have that as `spmode_kind=3&set_spmode=X`

Nick-Sills, 2024-06-19:
Faikin-S3-MINI-N4-R2: 01533d0 2024-06-19T15:50:01

Yep, that fixed it. I think we're all done.   Thanks for your help.