revk/xESP32-Faikout #339: [BUG] Demand control cannot be set anymore

eupesco, 2024-05-23:
**Daikin hardware**
S21, 3MXM40A9 + 2x CTXM15R + 1x FTXM20R

**Describe the bug**
When I try to change the demand control to another value (HA or directly on Faikin webpage), it is alway reset to previous value.  In MQTT, message failed-set demand is displayed.

revk, 2024-05-23:
That means the Daikin is not accepting the message. You can dump S21 messages if you like.

This is not a bug, it is correctly logging that it failed to set.

The fix is turn off demand control feature in the settings.

eupesco, 2024-05-23:
I am just wondering, because it worked properly for a long time and then suddenly it stopped working.
Now when I set a new demand control value in the Onecta app, the new value is not synchronized in Faikin (and vice versa) and it seems that no S21 message is sent. Is it maybe related to the latest firmware update of the Daikin WiFi module (1_31_0)?

revk, 2024-05-23:
OK, then something is odd.

Can you get a dump of the S21 messages when you make a change?

eupesco, 2024-05-23:
Message G7 is correct? I did some changes in the Onecta app and here are the G7 messages (after change, there were several G7 messages):
Demand off:  N0A= / N0A>
Demand 60: N0A; / N0A:
Demand 50: N0A9 / N0A8 / N0A7
Demand 95: N0A6 / 50A6 / 50A5
Demand 50: 50A4 / N0A4 / N0A5 / N0A4

revk, 2024-05-23:
Ok and changing on Faikin, is it sending the messages to make a change?

We should be sending a D7

What is interesting is that my code has D7 where demand is in the first byte. Your logs suggest the last byte changing mostly. Can you provide the exact logs from MQTT of the faikin sending D7 and getting G7 responses?

revk, 2024-05-23:
That status is a min/ave/max value over a reporting period, it thinks it has changed until it gives up, so a red herring.

I have not changed how `D7`/`G7` works so not sure how this worked for you before.

The issue looks like we change the first byte, but you are seeing changes in last byte. And the last byte is not changing by anything like the amount I would expect. So this looks like a protocol difference in your model.

eupesco, 2024-05-23:
Yes, it worked perfectly before - I used it a lot with HA automations for example. 

The only change that was made was the mentioned fw update of the WiFi module (BRP069C4x to version 1_31_0).

eupesco, 2024-05-23:
I tried it with another indoor unit.
Demand control is 70 according to Faikin (in Onecta 50%): G7 N0ED

I set in Faikin to 50: D7 b000
Several changes to G7: N0EE N0ED N0EE N0ED --> failed-set

eupesco, 2024-05-23:
Sorry for confusion - the firmware update was done on the Daikin internal WiFi-module - nothing to do with Faikin.

revk, 2024-05-23:
OK, does that update firmware in your Daikin itself as well? Could it be that your Daikin now has different messages than before. That would explain it. I mean I thought Faikin replaced the Daikin WiFi module.

It may be we can work out what messages it now expects, but they don't look as obvious at all.

eupesco, 2024-05-23:
I would expect that only WiFi has been updated (due to the changes to Daikin cloud) - I can't say for sure, but I think the SW version of the indoor units hasn't changed (SW 20003002, EEPROM 12FC). And it looks like only demand control is affected. Why would Daikin change just that? 

Does anybody have an comparable configuration of Daikin (so WiFi BRP069C4x with 1_31_0)?

eupesco, 2024-05-24:
More weird things are happening:
I set all units to 100% in the Onecta app and I can see in Faikin MQTT messages 100%. But in the info status MQTT messages, the demand control has NOT changed to 100% (stays at 70%).

Now when I change the demand control of one unit to 40% in the Faikin web page, the following happens:
- info MQTT demand message changes to 40
- Faiking MQTT messages show some strange values like 68 until it changes to 40
- Strangely **all units** are set 40

When I change one back to 100:
-  status message changes to 100 at the changed unit
-  Faikin messages change from 40 to 44 to 100
-  All other units change to 100: Faikin messages change on one from 40 to 86 to 100 - other from 40 to 88 to 100
- Info state messages of the other units: one stays at 40, the other changes to 70.

The error message is now only shown occasionally.

revk, 2024-05-24:
There are different messages you can turn on and off. The one you are looking at is almost certainly the periodic summary, I think usually every minute, and shows min/ave/max for the period. There is a setting for live status messages if you want.

eupesco, 2024-05-24:
I activated live livestatus but I still see the interim values (like 86, 82, etc.).  I also disconnected the units from the mains - and now the behaviour is worse again (currently I only get failed-set errors).

Do you have any idea what might be wrong with my setup? Is there anything else I can check?

eupesco, 2024-05-24:
This is a great pity :(.
Is there an easy way to flash an older version of the Faikin fw, for example from February (so a time when it certainly worked properly)?

revk, 2024-05-24:
You have to check out old version from GitHub, and either manually flash the binary from that (there are instructions somewhere) or build and flash.

eupesco, 2024-05-24:
Ok, doesn't seem to be that easy. Maybe this is a stupid question, but is it possible that you upload the 2024-02-28 release for S3 (https://github.com/revk/ESP32-Faikin/commit/0cfe2ac6384c0fdf76b788646552f14c5bc69643) to beta for a few minutes so that I can just OTA flash it?

revk, 2024-05-24:
People auto load code so I would rather not. You can change the URL it uses for OTA to something where you put that binary...

eupesco, 2024-05-24:
Good idea, tried it but I get error upgrade check failed. I'll give up for now :(

eupesco, 2024-05-27:
Yes, that's right. The demand control only applies to the outdoor unit.
However, the description in the Onecta app is different: it says, that the least restrictive demand control value set in an indoor unit is applied to all the indoor units (e.g. if you set the units to 50%, 60% and 80%, 80% is used for the outdoor unit).

Unfortunately, I still have the problem that I can no longer set the value correctly via Faikin.

revk, 2024-05-27:
As I say, if you can work out there required S21, I can add it. Sadly I do not have every model of Daikin on hand here to try and debug such things myself here.

eupesco, 2024-05-27:
Yes, that is absolutly clear. However, I am still not able to get an older sw version of Faikin running to check if the problem is on Faikin or Daikin side :(

wurmtal868, 2024-06-15:
I had a similar issue with my 3MXM40 a while ago. 
I fixed it by sending every change in demand control to all indoor units connected to the same outdoor unit. When the indoor units request different demand values the outdoor unit seemed to reject changes until all request the same value.

MUN0X, 2024-06-15:
> I had a similar issue with my 3MXM40 a while ago. I fixed it by sending every change in demand control to all indoor units connected to the same outdoor unit. When the indoor units request different demand values the outdoor unit seemed to reject changes until all request the same value.

Thats how the old daikin app used to do it

hokreb, 2024-06-15:
I had the same problems in spring this year (~march). I was only able to resolve this by switching the fuse to make the complete units currentless. After that, everything worked again, I tried several things, but never have any hint what was wrong.

eupesco, 2024-06-19:
I still have the problem.
@wurmtal868 : I have an automation in HA, that sends the new demand control value to all indoor units. Like said, this worked for a long time until an unknown time (maybe related to the Daikin update so 1_31_0).
@hokreb : I have tried this several times without success.

The thing that confuses me is the random behavior. Sometimes it works, sometimes it doesn't (currently it doesn't work at all).

fichtenmoped82, 2024-07-04:
I observe the same like MUN0X:

This is the same behaviour with my system of 2MXM50 and 2 indoor units (both one of the newer 2023 models incl. the 1_31 FW update). I also control it by setting all units initially to 100%, then using just one after a few seconds to set the demand percentage. The other unit follows automatically.

revk, 2024-08-18:
Did we resolve this - it sounds like it is a tad complicated with multiple units.

More work is being done on S21 protocol (separate issue), which may help with this.

eupesco, 2024-08-18:
Unfortunately not - I still have the problem with the latest beta. If I set it to all the indoor units at the same time, it still does not work.

revk, 2024-08-18:
OK, may be going round in circles. `dump` shows what we send. Ultimately we have to understand what the Daikin does to handle it. I can say that there is a lot of work being done to understand the S21 protocol now, which may help if what we are sending is wrong somehow. But this is all reverse engineered.

eupesco, 2024-08-18:
Yes, the main problem is that it sometimes accepts the new demand value. Although it is not synchronized with the Onecta app (Onecta does not show the new value) - but it is active when I look at the power consumption.
So I would say that the S21 message is definitely correct, but there must be a constraint that I do not understand.

eupesco, 2024-09-14:
With the latest beta, the problem seems to be gone. No idea why, but it works a lot of times now. I hope it stays that way 😀

closed 2024-09-14 by eupesco