revk/xESP32-Faikout #342: [FEATURE] Support of intelligent eye sensor enabling/disabling

nikbyte, 2024-05-26:
**Daikin hardware**
FTXA35C2V1B, S21 connector

**Is your feature request related to a problem? Please describe.**
This AC has "intelligent eye" sensor which is not controlled by "sensor" option in faikin.
It would be good to figure out why and fix

**Describe the solution you'd like**
Working "sensor" switch in UI.

revk, 2024-05-26:
If you find the S21 messages, let me know and I'll make it work.

nikbyte, 2024-05-26:
I enabled messages dump to mqtt and I see no changes in any of registers when I enable/disable this mode from IR remote. :(

nikbyte, 2024-05-26:
How else I can debug it?

Does faikin publish into mqtt all S21 messages or only part of them? For example, G9 or GD which I don't see in output.

revk, 2024-05-26:
You can send any S21 to test.

nikbyte, 2024-05-26:
Yes, I also wanted to try, but I cannot figure out where and how publish it to mqtt. Could you please provide me an example?

nikbyte, 2024-05-27:
Just for info.

I tried 0x00 0x00 0x00 0x00 and 0xff 0xff 0xff 0xff into all possible D-registers, it doesn't change value of intelligent eye. :(

(of course, I ensured that I see my packets in /tx queue and I ensured that I can enable/disable another features this way).

I'm out of ideas yet.

basak, 2024-05-27:
I can confirm this isn't working for me either with the same model (both FTXA25C2V1B and FTXA35C2V1B). As it happens I was also trying to debug this yesterday and saw nothing relevant changing with `"debug": true`.

geni0602, 2024-07-17:
Hi, 
I can also confirm it. The sensor is not functioning over the webGUI  Daikin FTXM50R2V1B. On the remote there is one button to control confort and sensor mode.
[image]
1. press = comfort mode 
2. press = Sensor mode
3. press comfort + sensor mode 
4. off
 when I press the remote button from comfort to sensor mode I can see this change
[image]
[image]

and this ist the whole massage:
{"G1":"13PA","G2":"=;\u0000\u0080","G3":"0\u00FE\u00FE\u0000","G4":"0\u00B0\u00800","G5":"000\u0080","G6":"0004","G7":"00Q@","G8":"0200","GA":"CA30","GB":"0330","GC":"0031","GG":"0D00","GK":"q{51","GM":"5231","GN":"0000","GP":"2A00","GQ":"BB00","GS":"1000","GT":"4000","SH":"542+","SI":"081+","Sa":"561+","SL":"000","Sd":"000","SN":"150+","SM":"640+","SX":"562+","SD":"000"}

geni0602, 2024-07-17:
Comfort mode is working properly 
[image]

revk, 2024-08-18:
I think we have `sensor` and `comfort` working now, don't we?

There is more work on S21 being done which may allow more options soon.

nikbyte, 2024-08-18:
@revk no, for AC with intelligent eye it still doesn't work.

revk, 2024-08-19:
Is that not what `sensor` mode is, or am I getting confused?

nikbyte, 2024-08-19:
No, it's some different mode. I'm not sure it's supported in S21, but let's see, maybe we'll find it sometimes.
For me the sensor switch does nothing, just ignored.

revk, 2024-08-19:
OK then I think we need to wait for more S21 research. It is going well.

geni0602, 2024-08-19:
I am not at home right know, I'll test it next week

Take-Off90, 2024-08-29:
I have a Emura3 (FTXJ35A2V1BB) which has also the intelligent eye sensor.
Maybe the follwing information is helping:

When I activate the **Sensor Mode** on Faikin it is reducing the LED light of the A/C.
Disabling the Sensor Mode ist not turning it back on.

The Faikin **LED Control Switch** is doing nothing. 

Sensor mode is different from intelligent eye.
Sensor mode can recognize if there are people in the room and then adjust the settings. (used for energy saving)
The intelligent eye can "see" **where** in the room a person is. (Used to "readjust" the airflow away from the person)

revk, 2024-08-31:
I think we need to just wait for some of the reverse engineering that is going on. Then this can be adjusted.

Sonic-Amiga, 2024-09-22:
@nikbyte Hello! Still here ?

We have a python script https://github.com/revk/ESP32-Faikin/blob/main/Tools/Simulators/collect_profile.py can you run it on your A/C ? The script will send some commands via mqtt and gather responses. I need the resulting profile which it dumps. Then i'll put this profile into our A/C simulator and see how original Daikin controller handles it.

nikbyte, 2024-09-22:
Sent back by email.

Sonic-Amiga, 2024-09-22:
Cool news:
before falling asleep, i put together a .settings file and run it in the sim. m_dtct=1 !!! My controller recognizes it. By comparing with other profiles i'll be able to figure out settings bits.

Bad news: neither old app nor Onecta support this feature. Without some app, that controls it, we can't figure out control commands.

MartB, 2024-11-13:
@Sonic-Amiga I can toggle the intelligent eye from my onecta app on the original wifi controller on my new Perfera unit. Can i help somehow? I can confirm that the sensor mode is not doing the same thing on my controller too.

Sonic-Amiga, 2024-11-14:
Hello! What kind of wifi adapter does it natively use? Would be very nice to get high-quality pcb photos

Sonic-Amiga, 2024-11-14:
Hm, so my C41 should indeed talk to your aircon. Okay, thank you for the profile, I'll try pushing it for some more.

Sonic-Amiga, 2024-11-24:
@MartB  Hi! Keep fighting; and even with your profile it doesn't want to work. After i disconnect it from (simulated) X50 aircon (which works fine) and connect it to (simulated) FTXM35A5V1B with your profile; the adapted disconnects from cloud and sits in "config" mode, responding this to /config/adapter/product_info:
```
{"rsc":2000,"model_name":"","serial_number":"","region":"eu","device_type":1,"device_kind":"NA"}
```
 I noticed that the adapted also queries for Rz52 and Rz72 commands. Responses from my aircons (both) are '40' and '23' correspondingly. I programmed these, result is the same.
 In order to complete this research, could you tell (using Faikin) what are responses to these two commands from your A/C ? Maybe it tells something different, and the controller does some identification based on these values.

Sonic-Amiga, 2024-11-24:
> I can toggle the intelligent eye from my onecta app on the original wifi controller on my new Perfera unit. Can i help somehow?

 Unfortunately we seem to be stuck with emulating an S21 A/C with this controller; but it's possible to intercept raw traffic.
 I'm not sure how to use Faikin's snoop mode; never tried/tested it; i instead just connected a PC via a usb-serial adaptor in parallel to one of serial port lines, fired up RealTerm (https://sourceforge.net/projects/realterm/) and simply captured hex dump.
@revk Can you help with snoop mode maybe ?

Sonic-Amiga, 2024-11-25:
Hello everyone! Some news, bad and good.
Bad: I still cannot befriend BRP069C41 with S21. It just refuses to talk to cloud and sits in "unknown device connected" state.
Good: Thanks to https://github.com/revk/ESP32-Faikin/discussions/220#discussioncomment-11312795 i've downloaded Thai app ("Daikin mobile controller") and reverse engineered it a bit. It mentions at least 3.40 protocol, so i expect BRP072C42 to know some extra features. And from common/basic_info i see HW platform is the same, so i should be able to upgrade my BRP069B41 to 072C42 and get the new functionality

The app refuses to find my BRP069, so i need some tricks and some time. Perhaps i'll write my own firmware download tool.

Sonic-Amiga, 2024-11-26:
All fixed, all tricks learned. Now i have an upgraded controller, fully recognized by the app. Let's see what new stuff it gives us.

MartB, 2024-12-06:
Sorry was not really paying attention, is there anything you still require from me?

Sonic-Amiga, 2024-12-06:
Hi! No, nothing to ask you for. I still cannot replicate your setup, i.e. mt BRP969C41just refuses to connect to the cloud and do anything if i use S21 protocol. It only fully works with X50.
 I am currently trying a different way. Also not doing a whole lot, no time, sorry.