revk/xESP32-Faikout #343: [FEATURE] Respect fan night mode

nikbyte, 2024-05-27:
**Daikin hardware**
FTXA35C2V1B, S21

**Is your feature request related to a problem? Please describe.**
I'd like to have fan night mode compatible with faikin-auto mode.

**Describe the solution you'd like**
Faikin-auto mode always switches fan to auto mode.
It would be awesome if it switches mode to auto, only in case if fixed speed enabled.
If auto or night mode already enabled, then just don't touch them.

**Describe alternatives you've considered**
If solution above is not suitable, then add night mode switch in faikin-auto mode menu.

revk, 2024-05-27:
There is no way in S21 to know the fan is in night mode, sadly.

Also, it does not set auto, it sets low/med/high and adjusts to keep temp in range. I think it may be reported as auto in HA on basis that Faikin auto mode is controlling it.

nikbyte, 2024-05-27:
It manages fan speed even in simple thermostat mode?

Also... It can't read, but it can set (I checked). It means, "alternative" way is possible – to add night mode to faikin-auto and in this case not manage fan and just set night mode.

revk, 2024-05-27:
It can set night mode, yes, I could look in to a way to force that.

At present it only ever goes down to "low" fan in faikin auto.

I don't think it ever sets "auto" mode itself though.

revk, 2024-05-28:
Ah, OK, there is already a `nofanauto` which stops Faikin-auto mode touching the fan. Set this and set `quiet` and it will not touch it.

nikbyte, 2024-05-28:
Unfortunately, it doesn't work. At least in simple thermostat mode. Even if this option enabled, night mode is disabling all the time.

revk, 2024-05-28:
I can check code? But that seems very odd.

revk, 2024-05-29:
OK what is the fan being set to.

The issue is that `D1` sets power, mode, temp *and* fan, so no way to control anything without sending a fan.

So even if we are "not setting the fan" as part of Faikin auto mode, it will have to send the current fan setting in the `D1`.

As it cannot tell *auto* from *night* it may think it is *auto* at present, and so send that, not *night*.

If that is what you are seeing, I may have to come up with a change of logic somehow, or a new setting.

nikbyte, 2024-05-29:
If I understood properly, it means that nofanauto can't work. Maybe just replace it with setting to set night mode each time when faikin auto changes D1?

revk, 2024-05-29:
Well it can work, hence asking what fan is set. Is it setting `auto`?

E.g. if you want fan set `low` it will stay `low` and not change if `nofanauto` is set.

The issue is more than night mode can't work properly as S21 does not report its status.

closed 2024-05-29 by revk

nikbyte, 2024-05-29:
Thank you! Let me try.

nikbyte, 2024-06-21:
Thank you again for the feature, it works perfect!

I only have a question: is this possible to announce this setting as a switch into mqtt or probably you could advice how to switch it via http request. Thank you in advance.

revk, 2024-06-22:
I'll think about it.