revk/xESP32-Faikout #346: [BUG] - MQTT Prefix - App/Host not working as expected - Also add a custom Prefix

Nick-Sills, 2024-06-02:
**Faikin hardware**
Amazon purchased, 04/24.
Faikin-S3-MINI-N4-R2: 729c466 2024-05-14T08:53:28

**Daikin hardware**
Daikin FTXM42R
Daikin FTXM20R

Connect to Faikin using EKRS21 Wiring Harness + Dupont cables

After much squinting at photos connecting the EKRS21 to Faikin boards, and not being 100% sure about the orientation of the cable, I can confirm to anyone else attempting to use an EKRS21 cable that **Brown = Ground** 😄 

Thanks again @revk for this project. 👍 

**Describe the bug**

I have multiple (3) locations using Home Assistant and MQTT. My network uses site-to-site VPN connections with an MQTT bridge and Home Assistant at each site. My Home location also has an additional  **Master** Home Assistant docker installation and an additional central Mosquito broker on a Synology NAS. MQTT traffic is segmented using a unique prefix for each site, and the MQTT bridges are configured to send everything to the central broker, but filter MQTT traffic for the site unique prefix. This works really well using the ESPHOME `topic_prefix` and Zigbee2MQTT `base topic`. There's a local Home Assistant install at each site, and this is restricted to discovering and managing it's own local MQTT devices, whereas the master can see and manage all devices. 

e.g.  for Site 3

ESPHOME `topic_prefix`  = `site3`
Zigbee2MQTT `base topic` =  `site3`

mosquitto.conf
```
connection bridge-03
address 192.168.1.201:1883
bridge_protocol_version mqttv50
topic # out 0
topic site3/# in 0  
```

Unfortunately, there's no way in the faikin firmware to set a custom topic prefix, so this is what I tried:

**Setting custom prefixcommand, prefixsetting, etc.** 

`prefixcommand = site3/command`
`prefixsetting = site3/setting`
`etc`

Faikin
[image]

Unfortunately, the Home Assistant discovery payload seems to ignore this, and results in HA Climate entities in the local and master locations being unable to control any function the AC units 

MQTT
[image]

> **Note:**
> Based on a previous discussion, **I also tried the latest beta build, but the results were the same.**

The end result is that the AC units can't be controlled in Home assistant.

**Setting `prefixhost` and/or `prefixapp`**

I couldn't  work out how it could be possible to set both at the same time, so I tried combinations as well as both together, but the result was always `<device name>/faikin`, plus commands still had no prefix.

My idea was to add Faikin to the bridge protocol

```
connection bridge-03
address 192.168.1.201:1883
bridge_protocol_version mqttv50
topic # out 0
topic site_3/# in 0  
Faikin/# in 0
```
But unfortunately, I couldn't get `Faikin` as the primary prefix, and `command` still has no prefix.  Just to re-iterate, I have also tried the latest Beta build. 

There are some quirks in the HA Climate entity that I won't mention at this time.

**To Reproduce**
See above

**Expected behaviour**
I expected to be able to control the base faikin topics to facilitate MQTT segmentation.  

**Additional context**
After these problems I tried to use the Home Assistant native Daikin local control using the Faikin IP addresses. This can connect both locally and over VPN, which is fantastic!, However, there's an additional problem with the climate entities that I'll open another issue for later if necessary.

Thanks for reading, and let me know if you need further info.

revk, 2024-06-02:
Latest faikin code allows / in command and setting names.

Nick-Sills, 2024-06-02:
I've updated to the latest

revk, 2024-06-02:
I'll look in the morning. The latest code really should cover all of this quite well. But I'll take a look.

Nick-Sills, 2024-06-02:
Thanks!  I appreciate it.

Just in case I'm wrong on this...

I was expecting the HA discovery payload to change from:
```
"~": "command/Villacana AC Back Bedroom",
```

To:

```
"~": "site3/command/Villacana AC Back Bedroom",
```
As it didn't, I assumed that's why AC commands aren't working. Similarly with the App/Host, it renders the AC control inoperable.

revk, 2024-06-02:
Yeh, it could be an error in my HA config stuff.

revk, 2024-06-03:
OK, yes, HA confirm messages did not allow for `prefixapp` or `prefixhost` at all.

I have changed to fix this in latest beta.

Also fixed the order of *appname* and *hostname* when `prefixapp` and `prefixhost` are set.

Also changes HA status messages to be a `state` message rather than top level MAC, so they follow the same rules for `prefixapp` and `prefixhost`.

Please test latest beta. Thanks.

closed 2024-06-03 by revk

Nick-Sills, 2024-06-03:
Thanks @revk, does this also fix the `prefixcommand` etc ?

I'll be able to test this afternoon.

revk, 2024-06-03:
It is called `topiccommand` now, but I have not changed how that works for this, just what is sent to HA.

So let me know if that does not sort it, and I'll look more in to that. Basically all the HA confirm assumed `command/`*hostname*`/...` which is wrong if different `topiccommand` or `prefixapp` or `prefixhost` set. Sloppy, sorry.

revk, 2024-06-03:
The buttons not allowing a save is weird. But yeh, check latest beta.

Nick-Sills, 2024-06-03:
Yes, latest beta: Faikin-S3-MINI-N4-R2: https://github.com/revk/ESP32-Faikin/commit/b1a9db5e304709094d454a0d5fa5553646e4d0b3 2024-06-03T08:35:42

revk, 2024-06-03:
I'll do some tests here and confirm.

revk, 2024-06-03:
OK complicated.

We have a periodic report that is sent as `Faikin/` as the prefix. When using without `prefixapp` that would be `Faikin/hostname`, but with `prefixapp` it becomes `Faikin/appname/hostname`.

I am making a special case for not duplicating the appname if appname is used as the main topic, which should address that.

Beta in a moment.

Nick-Sills, 2024-06-03:
Thanks...

Does that fix `Command > Faikin >  {HOST}` as well:

[image]

revk, 2024-06-03:
OK `command/Faikin/hostname` would be if `prefixapp` set.

Where do you see that, as the Faikin does not *send* commands, it receives them.

(Sorry not familiar with whatever your are using to log things)

Nick-Sills, 2024-06-03:
I'm using MQTT Explorer, and the command is coming from Home Assistant when I change the climate state. 

The climate entity is publishing to `command/Faikin/hostname` so HA can't sent the command over a bridge when the remote bridge has `topic Faikin/# in 0`.  I think it should publish to `Faikin/hostname/Command`, that way all statuses and commands can be segmented with `Faikin/#`

I believe this is what happens with Zigbee2MQTT  `base topic` and ESPHome `topic_prefix` .

revk, 2024-06-03:
OK, the `prefixapp` and `prefixhost` settings allow a number of combinations which can be adjusted to suite your needs.

The latest beta should consistently report a config to HA matching what the Faikin is set to do.

[image]

Nick-Sills, 2024-06-03:
**Bingo!** The messages are now routed correctly for me. Thanks

[image]
<br/>
<br/>
And if I set `topiccommand` etc.. with a prefix and deselect `prefixapp` and `prefixhost` that works as well, and means I can use my site name `z2m_villacana` to segment: 👍 

[image]

Thanks for your help

Nick-Sills, 2024-06-04:
Sorry to raise another problem. 

With the latest beta:  Faikin-S3-MINI-N4-R2: b1a9db5 2024-06-03T08:35:42
I've lost the current temperature for both Faikin units:

[image]
</br>
[image]
</br>
</br>
It looks like the MQTT HA Discovery payload is wrong:

```json
"curr_temp_t": "34B7DA588E44",
```

Should be:
```json
"curr_temp_t":  "Faikin/Villacana AC Back Bedroom/state/ha",
```

"Faikin/Villacana AC Back Bedroom/state/ha"

```json
{
  "unique_id": "34B7DA588E44",
  "dev": {
    "ids": [
      "34B7DA588E44"
    ],
    "name": "Villacana AC Back Bedroom",
    "mdl": "FTXM20R",
    "sw": "b1a9db5",
    "mf": "RevK",
    "cu": "http://Villacana AC Back Bedroom.local/"
  },
  "icon": "mdi:thermostat",
  "~": "Faikin/Villacana AC Back Bedroom/command",
  "min_temp": 16,
  "max_temp": 32,
  "temp_unit": "C",
  "temp_step": 0.5,
  "temp_cmd_t": "~/temp",
  "temp_stat_t": "Faikin/Villacana AC Back Bedroom/state/ha",
  "temp_stat_tpl": "{{value_json.target}}",
  `"curr_temp_t": "34B7DA588E44"`,
  "curr_temp_tpl": "{{value_json.temp}}",
  "mode_cmd_t": "~/mode",
  "mode_stat_t": "Faikin/Villacana AC Back Bedroom/state/ha",
  "mode_stat_tpl": "{{value_json.mode}}",
  "fan_mode_cmd_t": "~/fan",
  "fan_mode_stat_t": "Faikin/Villacana AC Back Bedroom/state/ha",
  "fan_mode_stat_tpl": "{{value_json.fan}}",
  "fan_modes": [
    "auto",
    "low",
    "lowMedium",
    "medium",
    "mediumHigh",
    "high",
    "night"
  ],
  "swing_mode_cmd_t": "~/swing",
  "swing_mode_stat_t": "Faikin/Villacana AC Back Bedroom/state/ha",
  "swing_mode_stat_tpl": "{{value_json.swing}}",
  "swing_modes": [
    "off",
    "H",
    "V",
    "H+V",
    "C"
  ],
  "pr_mode_cmd_t": "~/preset",
  "pr_mode_stat_t": "Faikin/Villacana AC Back Bedroom/state/ha",
  "pr_mode_val_tpl": "{{value_json.preset}}",
  "pr_modes": [
    "eco",
    "boost",
    "home"
  ]
}
```

revk, 2024-06-04:
Oops, let me fix that now.

Nick-Sills, 2024-06-04:
Thanks! All sorted

Faikin-S3-MINI-N4-R2: 57fcc76 2024-06-04T09:06:53