revk/xESP32-Faikout #347: [BUG] - HA Climate entity loses target set temp

Nick-Sills, 2024-06-03:
**Faikin hardware**
Amazon purchased, 04/24.
aikin-S3-MINI-N4-R2: 729c466 2024-05-14T08:53:28

**Daikin hardware**
Daikin FTXM42R
Daikin FTXM20R

Connect to Faikin using EKRS21 Wiring Harness + Dupont cables

**Describe the bug**
Climate entities in Home Assistant occasionally lose the target set point and display nothing.

At first I thought this was an MQTT retention issue, but this happens for bother MQTT and the Home Assistant native Daikin integration:

**Example using MQTT:**
[image]

**Example using the Home Assistant native Daikin integration**
[image]

**To Reproduce**
I just have to leave the set point unchanged for a day or so.  I adjusted the set point in the Onecta app  to get the set temp back 

**Expected behaviour**
Retain the last set temp

Thanks for reading!

revk, 2024-06-03:
Can you dump MQTT messages for this?

Nick-Sills, 2024-06-03:
It's difficult, because it can't be reproduced on demand, and both climate entities are now OK.

It's also happening without MQTT using the native Daikin integration, so I was thinking that MQTT messages won't help.

I'll see if I can reporoduce on demand somehow and dump the messages.

Nick-Sills, 2024-06-03:
Good question. I'll check this when it happens and report back

Nick-Sills, 2024-06-03:
It's happened again, and the web interface shows an empty target temperature:

[image]

Nick-Sills, 2024-06-03:
I think I've now worked out what's happening.  I can reliably make the target temp null by rebooting either Faikin board. I've been making multiple changes and causing reboots lately,  and didn't correlate the two until I specifically looked for it. Apologies for the long winded discovery here. 

So, after a reboot I have to go into the Web UI and move the set temperature slider to enable the Climate entity in HA to operate.

revk, 2024-06-03:
Ok that is, none the less, somewhat odd, but makes slightly more sense.

closed 2024-06-08 by revk

revk, 2024-06-08:
I need debug go the S21 messages really, to see why we are not getting temperature until is changed.

Nick-Sills, 2024-06-08:
I just reboot both units after updating to the latest firmware, and guess what, the target temps were fine!  It doesn't seem to be as easy to reproduce as I thought. 

You might as well leave this issue closed, and if I ever notice this problem or a pattern to reproduce, I'll try and dump the messages.