revk/xESP32-Faikout #349: [BUG] in HA, the aircon autostarts in heat mode

Swalabrr, 2024-06-04:
**Faikin hardware**
this is happening with Faikin purchased

**Daikin hardware**
i'm connecting with S21

**Describe the bug**
whenever the aircon is requested to start from within HA, regardless of how the request has been made -- manual or through scripting -- the aircon will start in Heat mode

```
  - service: climate.set_hvac_mode
    data:
      hvac_mode: cool
    target:
      device_id:
        - 1e7002676ea1b51a562a14e2de5b01c3
  - service: climate.set_temperature
    data:
      temperature: 23.5
      hvac_mode: cool
    target:
      device_id:
        - 1e7002676ea1b51a562a14e2de5b01c3
  - service: climate.turn_on
    data: {}
    target:
      device_id: 1e7002676ea1b51a562a14e2de5b01c3
```

**To Reproduce**
start AC, switch to dry or cool,
stop AC, 
start AC. It will start in heating mode

**Expected behavior**
start in the last selected mode

**Aditional info**

also, just setting the HVAC mode or the temp through scripting will start the AC unit (but it will start with the selected HVAC / temp)

revk, 2024-06-04:
Is this through HA?

Is this legacy URLs or MQTT?

Does it do the same using the web interface (unlikely)?

Can you dump the mqtt messages being sent or http requests sent?

Swalabrr, 2024-06-04:
yes, it is through HA using MQTT. It does NOT behave the same using the webinterface.

I'd like to dump the mqtt, but I have no idea how to do that. Do you mean "debug" on webinterface?

revk, 2024-06-04:
There are tools to show what is happening on MQTT. I think even HA can do it. But not sure which to recommend.

Nick-Sills, 2024-06-04:
@Swalabrr Try MQTT Explorer and monitor the messages. 

https://mqtt-explorer.com/

Nick-Sills, 2024-06-08:
I notice exactly the same issue, and checking the MQTT messages and it looks like HA is sending a 'Heat'  command MQTT message when the 'On' button is clicked.  Is there even an exposed MQTT 'On' command in Faikin? 

It's also behaving exactly the same with the 'Legacy' HTTP integration, and always starting in Heat mode. However, the official Daikin WiFi board **does start with the correct HVAC mode**, i.e, it starts with the last HVAC mode before it was turned off. I can debug this and get back with my findings. 

@revk Do you want me to create an additional issue for the the HTTP integration, or handle it here?

revk, 2024-06-08:
We have had issues with reporting hvac mode, as we don't know from S21. We do have an "on" command but HA sees off as just one of the modes it can set I believe. I assume it uses hvac mode somehow.

Nick-Sills, 2024-06-08:
@revk I'm not sure if I was mistaken, or if the last Beta update fixed some things, but now with the **'Legacy' HTTP integration**, turning the AC Off then On retains the last HVAC mode. So, if I set to `Cool`, turn Of then On, the AC unit returns to `Cool`. 👍 

However, the Fan mode and Preset values aren't maintained, they revert to 'Auto' and 'None'. 

Comparing the HTTP response with the Daikin WiFi board I can see the issue.  

**Daikin & Faikin**
1. Turn  On - Cool, Silence, Eco
2. /aircon/get_control_info
3. Turn Of
4. /aircon/get_control_info

For **Daikin** there are no differences in any parameters returned by the first and second  `get_control_info` apart from `pow`, which is to be expected.

For **Faikin**, as well as `pow`, the following parameters are reset in the **second**  `get_control_info`, when the AC is off  ->  `f_rate`, `b_f_rate`, `dfr3` and `adv`.

So, it looks like the `dfr` array, `f_rate`, `b_f_rate`, and `adv` need to be persisted after turning off the AC with a `pow=0`, so they are returned by the next, and subsequent,  `get_control_info` calls.

Does that make sense?

As far as the MQTT climate entity not persisting sate between `Off` and `On`, I'll play with publishing some MQTT topics to see if  anything jumps out.

Nick-Sills, 2024-06-08:
I think with MQTT we have similarly persist, and return the values in the `state` topic?

revk, 2024-06-09:
The latest release (and beta) have a number of improvements for legacy URL handling, yes.

This includes some level of `dfr` handling. This is not persistent over restart of the Faikin, it simply persists the per mode fan value returned with `get_control_info`.

But the `f_rate` and `b_f_rate` (what does the `b_...` mean?) should persist anyway, as before.

Do you have a clear sequence of command (URLs) I could try and test here?

Nick-Sills, 2024-06-09:
For my bedroom AC, the fact that 'Silence' fan mode is lost on power on is the most concerning issue here. 

If you need anything else then let me know.

**Fan Rate - Silence**
/set_control_info?pow=0&mode=7&stemp=20.0&shum=0&f_rate=B&f_dir=0

**Preset - Eco**
/set_special_mode?spmode_kind=2&set_spmode=1

Swing Mode Vertical and **Set Mode Cool** 
/set_control_info?pow=1&mode=3&stemp=20.0&shum=0&f_rate=B&f_dir=1

**After Power On with Set Mode Cool**
/get_control_info 
ret=OK,pow=1,mode=3,`adv=12`,stemp=20.0,shum=0,dt1=20.0,dt2=20.0,dt3=20.0,dt4=20.0,dt5=20.0,dt6=20.0,dt7=20.0,dh1=0,dh2=0,dh3=0,dh4=0,dh5=0,dh6=0,dh7=0,dhh=0,b_mode=3,b_stemp=20.0,b_shum=0,alert=255,`f_rate=B`,f_dir=1,`b_f_rate=B`,b_f_dir=1,dfr1=A,dfr2=A,`dfr3=B`,dfr4=A,dfr5=A,dfr6=A,dfr7=B,dfrh=0,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0,dmnd_run=0,en_demand=0"

**Power Off**
/set_control_info?pow=0&mode=3&stemp=20.0&shum=0&f_rate=B&f_dir=1

**After Power Off**
/get_control_info 
ret=OK,pow=0,mode=3,`adv=0`,stemp=20.0,shum=0,dt1=20.0,dt2=20.0,dt3=20.0,dt4=20.0,dt5=20.0,dt6=20.0,dt7=20.0,dh1=0,dh2=0,dh3=0,dh4=0,dh5=0,dh6=0,dh7=0,dhh=0,b_mode=3,b_stemp=20.0,b_shum=0,alert=255,`f_rate=A`,f_dir=1,`b_f_rate=A`,b_f_dir=1,dfr1=A,dfr2=A,'dfr3=A`,dfr4=A,dfr5=A,dfr6=A,dfr7=B,dfrh=0,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0,dmnd_run=0,en_demand=0"

**Power On**
/set_control_info?pow=1&mode=3&stemp=20.0&shum=0&f_rate=A&f_dir=1

| Parameter | After Set Mode Cool On | After Power Off |
|-----------|------------|-----------|
| pow       | 1          | 0         |
| adv       | 12         | 0         |
| f_rate    | B          | A         |
| b_f_rate  | B          | A         |

<hr>
[image]

After Power Off -> Power On

[image]

revk, 2024-06-09:
Silence is a separate issue, S21 does not report silent/quiet/night mode.

We have a setting to always assume you want/have silent rather than auto.

Nick-Sills, 2024-06-09:
I'm not sure I understand.  'f_rate` = B  is silence.  I'm not in Spain at the moment so I'm doing this remotely and can't hear the fan. I could have sworn it was working. If so, why are we resetting 'f_rate` and 'b_f_rate` to A?

revk, 2024-06-09:
This is explained in various places in the documentation.

The underlying messages to the Daikin over S21 do not have a distinction.

We send the Daikin B and it comes back and says yes mode is A. Yes it sets silent but reports back that it is auto. Several attempts to "guess" which it is are not very good.

If anyone ever finds a flag or message to actually find that it is silent, that would be great. But nobody has yet.

There is a setting in the Faikin to always work as silent mode instead of auto mode, which may help.

Nick-Sills, 2024-06-09:
OK, thanks for that. Can you help me understand a bit more?

If I set 'Silence' and 'Eco' then change the Fan mode from 'Auto' to 'Cool' then both 'Silence' and 'Auto' persist, so it's good. Is it just that a Power Off command changes the Fan mode to Auto on S21 (whereas in reality it's still silent).

The power off command actually contains the correct fan mode, so I wouldn't call this a guess.

Nick-Sills, 2024-06-09:
Also it's `dfrx` array that's caching the Fan mode for each operational mode, and that's getting reset on power off.

Nick-Sills, 2024-06-09:
Aren't we caching the `dfrx` array now because of this fix?

https://github.com/revk/ESP32-Faikin/issues/348

Nick-Sills, 2024-06-09:
After a power off, could we lookup the `f_rate` and `b_f_rate` in the `dfrx` array  depending on the operational mode.?

revk, 2024-06-09:
Off should not be losing fan rate setting (other than auto/silent), that puzzles me.

revk, 2024-06-09:
Unless your Daikin is sending something different when off, we could get S21 debug and find out.

Nick-Sills, 2024-06-09:
> Off should not be losing fan rate setting (other than auto/silent), that puzzles me.

Yes, It is losing silent, and It's also losing the Preset Mode  `adv` parameter, 

**My question is:**
We fixed a problem with Fan Modes and Preset modes by populating **and caching** the `dfr1` to `dfr7` array, and the `adv` parameter in this issue: https://github.com/revk/ESP32-Faikin/issues/348. And that includes 'Silent' and 'Auto'

I understand that we can't get the Fan Mode from S21, but in other cases we're maintaining a cache for persistence. **Isn't it possible to do the same when the Power Off - `/set_control_info?pow=0&mode=3&stemp=20.0&shum=0&f_rate=B&f_dir=1` is received?** Also the correct  `f_rate` value is sent with the `pow=0`. 

I understand it's not perfect, because the state of the fan rate could be changed by a remote or the Onecta app so the cache could be stale, but persisting the last fan rate that was set (Silence, Auto or otherwise) in the http (or MQTT) is 1000 x better than it currently is, and is actually a very good outcome.  

( I tried the Onecta app, and confirmed that 'Indoor Quiet/Silent' doesn't update the HA Climate entity, but my question still stands ) 

Can you fire the same URLs at your end, to see what I mean? 

Fan Rate - Silence
/set_control_info?pow=0&mode=7&stemp=20.0&shum=0&f_rate=B&f_dir=0

Preset - Eco
/set_special_mode?spmode_kind=2&set_spmode=1

Swing Mode Vertical and Set Mode Cool
/set_control_info?pow=1&mode=3&stemp=20.0&shum=0&f_rate=B&f_dir=1

After Power On with Set Mode Cool
/get_control_info
ret=OK,pow=1,mode=3,adv=12,stemp=20.0,shum=0,dt1=20.0,dt2=20.0,dt3=20.0,dt4=20.0,dt5=20.0,dt6=20.0,dt7=20.0,dh1=0,dh2=0,dh3=0,dh4=0,dh5=0,dh6=0,dh7=0,dhh=0,b_mode=3,b_stemp=20.0,b_shum=0,alert=255,f_rate=B,f_dir=1,b_f_rate=B,b_f_dir=1,dfr1=A,dfr2=A,dfr3=B,dfr4=A,dfr5=A,dfr6=A,dfr7=B,dfrh=0,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0,dmnd_run=0,en_demand=0"

Power Off
/set_control_info?pow=0&mode=3&stemp=20.0&shum=0&f_rate=B&f_dir=1

After Power Off
/get_control_info
ret=OK,pow=0,mode=3,adv=0,stemp=20.0,shum=0,dt1=20.0,dt2=20.0,dt3=20.0,dt4=20.0,dt5=20.0,dt6=20.0,dt7=20.0,dh1=0,dh2=0,dh3=0,dh4=0,dh5=0,dh6=0,dh7=0,dhh=0,b_mode=3,b_stemp=20.0,b_shum=0,alert=255,f_rate=A,f_dir=1,b_f_rate=A,b_f_dir=1,dfr1=A,dfr2=A,'dfr3=A`,dfr4=A,dfr5=A,dfr6=A,dfr7=B,dfrh=0,dfd1=0,dfd2=0,dfd3=0,dfd4=0,dfd5=0,dfd6=0,dfd7=0,dfdh=0,dmnd_run=0,en_demand=0"

revk, 2024-06-09:
OK adv is just deduced from the other states, it is not a separate thing we get or store. So losing "silent" loses the adv as a result. So the whole issue is the lack of silent state in S21.

Now "caching" that you set silent may be a thing it could do, but it won't track if you use a remote to change from silent to auto, for example. (Note, if the real Daikin module can tell that it would be interesting as it would mean there is an unknown S21 message, which would be great to find).

At present we don't "cache" that we set silent. Maybe that could be a feature on a setting.

I have done a setting to always do silent instead of auto as plenty of people seem happy with a fan mode set low/high etc but don't need "auto" that is not "silent".

Nick-Sills, 2024-06-09:
> Now "caching" that you set silent may be a thing it could do, but it won't track if you use a remote to change from silent to auto, for example. (Note, if the real Daikin module can tell that it would be interesting as it would mean there is an unknown S21 message, which would be great to find).

Yes, the real Daikin module can tell, but persisting the last fan rate that was set (Silence, Auto or otherwise) in the http (or MQTT) comms is 1000 x better than it currently is, and is actually a very good outcome.

> At present we don't "cache" that we set silent. Maybe that could be a feature on a setting.

That's `B` in the appropriate `dfr1` to `dfr7` array. I perhaps don't understand what is and isn't cached, or what Faikin caching is.

Nick-Sills, 2024-06-09:
It seems that we only lose the persistence of 'Silence' and the Preset mode when  a `pow=0` is received.  Is there any chance you could explain why this command is different, and why it can't be treated like any other operational mode change?

> At present we don't "cache" that we set silent. 
`/get_control_info` after `/set_control_info?pow=1` returns `adv=12`, `f_rate=B`,`b_f_rate=B`, and `dfr3=B`. 

So, isn't that caching silence?  If not, how are we returning Silence when that can't be deduced from S21?

Apologies, for the number of questions. One of us isn't getting it, and it's probably me.

revk, 2024-06-09:
There are a number of underlying S21 messages.

When we send a fan of B we also send temperature and power state in the same message.

We get a response confirming what we set (except it is A not B). We assume that it is B at that stage as that is what we set. We also look at things like the actual fan speed to guess that may be it has been changed to A, but when initially set to B we can assume it is B.

If you change *anything* else in that same message, we have to send all those settings again and get a response. So a change of temp will do that same.

This time the power is off, the fan is off, and we were not actually trying to "change" the fan. So when it reports A this time we assume something has changed it to A. As I say, it is guess work as that message does not differentiate A and B, always reports A.

Now, separately, and in latest code only done a few days ago, it became apparent what the dfr values were, a record of what fan rate was last used in each mode. This is not (as far as I know) anything in S21 messages. So simply remembered by the Daikin WiFi. The Faikin legacy URL code was changed to simply remember what was last reported in each mode. Ie "caching" it and reporting as dfr.

If we could find an S21 message for silent that would solve it all.

We do have a setting to always operate silent not auto, to always send B not A, and always assume A being reported means B. That works for some people.

It may be that I can make a setting to always ignore a report of A if we last set B. Basically assuming nothing else is controlling the aircon. That may do what you want close enough.

revk, 2024-06-09:
I have issued a new beta you can try, with a new settings `noguessnight`

Try that!

closed 2024-06-09 by revk

Nick-Sills, 2024-06-09:
Thanks!

Updated to: Faikin-S3-MINI-N4-R2: f760d47 2024-06-09T11:00:53
Set `noguessnight` on.

No difference I'm afraid. The same `/get_control_info` values are being returned as before, and the Fan and Preset modes are changed when turning on as before. 

I'm not sure what this change is meant to do though?

revk, 2024-06-09:
Ok I may have got the setting wrong, try new beta in a couple of minutes. It should preserve the B you set, i.e. assuming nothing else could have set auto mode instead.

Nick-Sills, 2024-06-09:
If I turn `noguessnight ` off it's retaining Silence. Is that the correct way round?

revk, 2024-06-09:
No, as above 👎 
*Ok I may have got the setting wrong, try new beta in a couple of minutes. It should preserve the B you set, i.e. assuming nothing else could have set auto mode instead.*

Nick-Sills, 2024-06-09:
Faikin-S3-MINI-N4-R2: 1524594 2024-06-09T11:18:29

Yep! That' seems to be  preserving Silence correctly. 👍  I'll run through a few more combinations, but it seems great. 

Can we do the same for `adv` ?

revk, 2024-06-09:
As I say, `adv` is simply deduced, from `powerful`, `econo` and `streamer`, and they are read from the Daikin using separate S21. So we are back to *I would need to see the S21 messages to see why they have changed*, but my guess is that the Daikin is saying they are off.

Nick-Sills, 2024-06-09:
Makes sense, and I've just check with the Daikin WiFi board and that's losing the preset as well, so I guess we can assume that it can't be done. 

I'll continue checking combinations, but we're looking good. 

Perhaps `noguessnight`  should be default 'On'? I guess that defaulting to 'on' might annoy existing users that expect it to behave as it did before, but TBH, I can't think why anyone would expect that, and new users won't be firing issues because they don't realize what `noguessnight` means if it was defaulted to 'On'.

Nick-Sills, 2024-06-09:
Just one more thing...

This issue was raised for MQTT by the OP. I jumped on the thread and add HTTP to the mix, but MQTT still has the same problem. Can the MQTT state topic mirror the HTTP response changes? I think that might work.

revk, 2024-06-09:
The change is at the S21 tracking level, so it will do the same on MQTT - try it.

Nick-Sills, 2024-06-09:
MQTT is always going to 'Heat' when tuned on.

See last post here in case it adds anything:

https://github.com/home-assistant/core/issues/94114

revk, 2024-06-09:
OK what message is HA sending on MQTT to turn it on exactly?

Nick-Sills, 2024-06-09:
Hmm.. It's very difficult to interpret the MQTT stream as I've got so many MQTT messages flying around.  I had a quick pass it looked like this:

Turn on:
Faikin/Villacana AC Back Bedroom/command/mode  - heat

I assumed it was because the current 'status' topic had the an incorrect `mode` node, but I've just re-checked and it's correct, so I've no idea where 'heat' is coming from. 

BTW There are two state topics:

Faikin/Villacana AC Back Bedroom/state/status
Faikin/Villacana AC Back Bedroom/state/ha

This may be correct, but I thought it was odd. Not relevant to the 'heat' issue though. 

I'll keep poking to see if anything else jumps out....

revk, 2024-06-09:
There are several messages sent from the Faikin, the HA status is the one ending `/ha`

Nick-Sills, 2024-06-09:
Still digging but, when the AC is switched off from 'Cool', the `state/ha`  topic is this:

```json
{
  "online": true,
  "target": 20,
  "temp": 24,
  "outside": 27.5,
  "liquid": 15.5,
  "comp": 0,
  "demand": 100,
  "Wh": 853000,
  "fanfreq": 0,
  `"mode": "off"`,
  "fan": "night",
  "swing": "V",
  "preset": "home"
}
```

Should it be like this, as it is in `state/status`? 

```json
{
  "online": true,
  "target": 20,
  "temp": 24,
  "outside": 27.5,
  "liquid": 15.5,
  "comp": 0,
  "demand": 100,
  "Wh": 853000,
  "fanfreq": 0,
  `"mode": "C",`   
  `"power": false`
  "fan": "night",
  "swing": "V",
  "preset": "home"
}
```

revk, 2024-06-09:
I don't think HA understands *on*/*off* as separate from the `mode`, hence `mode` is `off` when *off*.

If you think HA can be configured to understand *on*/*off* as a separate setting, let me know the config, and I'll change it.

Nick-Sills, 2024-06-09:
This might be relevant:

https://www.home-assistant.io/integrations/climate.mqtt/#power_command_topic

revk, 2024-06-09:
Oooh, is that new, OK, that may be something I can change to - I don't have HA here to test though, so I'll have to make some test code for that.

revk, 2024-06-09:
Ok looks like reported back as mode, so presumably still off as mode.

Ok, so I'll just add the power command stuff, and make a beta, try that in a few minutes.

Nick-Sills, 2024-06-09:
OK, I can see that HA is now sending a 'power' command 👍 ,  but the AC isn't turning off:

[image]

revk, 2024-06-09:
Oooh, OK, so close, leave with me a moment.

revk, 2024-06-09:
Ok try new beta in a minute.

Nick-Sills, 2024-06-09:
Bingo! That's done it, and the Fan Mode persists as well. 😄 Thanks, and well done. 

While we're 'in the zone' can I request a couple more, hopefully minor, MQTT things for perfection?

This has been mentioned by me, and others, so it isn't something that an edge case.   Popular FTXM* units, and quite possibly others, don't have a 'Home' preset mode, so in the UI the control bounces around 'None' and 'Home', but settles on 'Home'. It also adds unnecessary text up the Climate display. Would it be possible to have a `nohome` switch to remove it from the list of presets, just leaving 'None', 'Eco' and 'Boost' if set. 

The final thing is a minor visual issue with icons, I think to do with the names of a state. I don't know exactly what it is at the moment but can dig in to find the cause. 

Thanks

revk, 2024-06-09:
OK I'll add `no home`, beta in a minute. Not sure on icons.

Nick-Sills, 2024-06-09:
Thanks. Home is now gone 👍 but selecting 'None' sets 'Eco' mode. Is the mapping incorrect, or is it an HA problem with stale mapping?

revk, 2024-06-09:
I'll have to look when I am back.

revk, 2024-06-09:
I literally look at first letter
```
           jo_bool (s, "econo", *value == 'e');
           jo_bool (s, "powerful", *value == 'b');
```

So None or none should unset both eco and powerful.

What MQTT is sent exactly?

revk, 2024-06-09:
I'll make a new beta that actually has "none" as a preset value, try that.

Nick-Sills, 2024-06-09:
That seems to have fixed it. 👍 Thanks

Finally, I think we might have covered the icon issue before, but I've got better clarity on it now:

HTTP Faikin sets the following hvac_actions for 'Heat' and 'Cool', other operational modes don't set an action, including 'Auto'. It doesn't matter that 'Idle' is never shown, and there's no attempt to set the havac_action for Auto. 

hvac_action: heating
hvac_action: cooling

This shows more distinct icons when either in Heat or Cooling, and setting the same for MQTT would make it consistent. 

HTTP Daikin and Faikin use 'heat_cool' / 'Heat/Cool' rather than 'auto' / "Auto' mode. These are different icon colours in HA. Having an option to set 'heat_cool' would make MQTT and HTTP both Daikin and Faikin consistent.

revk, 2024-06-09:
Any clue where I would put that - we can't get the proper HVAC action, we use one of
```
const char *const hvac_action[] = { "off", "preheating", "heating", "cooling", "drying", "fan", "idle" };
```

Nick-Sills, 2024-06-09:
https://www.home-assistant.io/integrations/climate.mqtt/

action_topic ?

heating for mode heat
cooling for mode cool
empty string otherwise I think

revk, 2024-06-09:
OK we don't have actual mode (as S21 does not say!!!), so we only report when in Faikin auto mode, and that is the list I posted above.

Nick-Sills, 2024-06-09:
I'm not sure I understand. I realize that we don't have the  actual HVAC action, the  HTTP comms spoofs it by forcing Heating for Heat mode and Cooling for Cool mode. Is it possible to do that with an MQTT action_topic ?

revk, 2024-06-09:
The action topic is new, and is for reporting HVAC more, i.e. a live "is it heating or cooling", etc.

We don't get that from S21.

So we cannot really report it at all, but it was added recently for my "Faikin auto" mode, we were are controlling it to force heat or cool as needed to get to target temp, perhaps even based on external BLE temp sensor, etc. In that case we know what we are doing, so can report the HVAC mode.

But in other cases we cannot and report, I think, idle.

Nick-Sills, 2024-06-09:
Understood. I was just suggesting that we fake it by always reporting Heating in Heat mode, and always reporting Cooling in Cool mode. I can see that this what Faikin HTTP is doing if I look at the state attributes for the Faikin HTTP climate entity. 

I'm not using "Faiin Auto" mode.

revk, 2024-06-09:
Not out of the question...

[52 more replies]