revk/xESP32-Faikout #354: [FEATURE] Save Environment-Temperature before restart

xudabit, 2024-06-13:
I'm using a Daikin ATXD connected via S403. For environment-temperature I'm using HomeAssistant which sends the temp via MQTT on every change.

When the Faikin-Module needs to restart (e.g. when saving settings or an OTA-Update) it resets the environment-temperature and uses the home-temperature. This leads to excessive cooling in my case, because the environment-temperature is way lower than the meassured home-temperature from the AC. 

Saving the temp on every change might be harmful for the storage, so I think this could be solved by saving the environment-temperature only before restarting the ESP. For sure this would not work when the ESP was turned off by cutting the power.

revk, 2024-06-14:
What MQTT message are you sending the Faikin?

Could it not just be set as retained so the Faikin gets it on restart from the MQTT server?

xudabit, 2024-06-14:
> What MQTT message are you sending the Faikin?
> 
> Could it not just be set as retained so the Faikin gets it on restart from the MQTT server?

Great... Now I'm feeling dumb :-D

Just kidding. That was a very easy fix for my problem. Tried it and it worked. Thank you!

closed 2024-06-14 by xudabit