revk/xESP32-Faikout #358: [FEATURE] Expose availability topic to HomeAssistant discovery

syphernl, 2024-06-18:
**Daikin hardware**
Not relevant

**Is your feature request related to a problem? Please describe.**
If the AC unit is offline (e.g. due to power cut or wifi connectivity issues) the entities Faikin exposes still appear to be available in HomeAssistant. 
The `HOSTNAME/state/ha` does contain an `online` boolean but this does not appear to have any effect on how the entities show up in HASS.

**Describe the solution you'd like**
HomeAssistant has an `availability_topic` (or `availability`) configuration that can be linked to via the `homeassistant/climate/ID/config`. 
For instance, Zigbee2MQTT configures its autodiscovery of a sensor like this:

```json
   "availability":[
      {
         "topic":"zigbee2mqtt/bridge/state"
      },
      {
         "topic":"zigbee2mqtt/DEVICE_NAME/availability"
      }
   ],
   "availability_mode":"all",
```

For Faikin this will probably be something like this:

```json
   "availability":[
      {
         "topic":"HOSTNAME/state/ha",
         "value_template": "{{value_json.online}}",
         "payload_available": true,
         "payload_not_available": false,  
      }
   ],
   "availability_mode":"all",
```

**Describe alternatives you've considered**
See above

**Additional context**
Add any other context or screenshots about the feature request here.

revk, 2024-06-18:
Yeh, did think of that. We do have a status message but the payload is more complex (when online it has way more info, uptime, and so on) so won't (AFAIK) work. The problem is you can only set one will, as far as I know, and so we would have to turn that off and replace with a simple one for HA.

Can be done. I'll consider it, as an option.

revk, 2024-06-18:
OK I think I can simply do any `availability_topic`, but not 100% sure.

Can you try latest beta and see if that works.

revk, 2024-06-18:
Ok I think latest beta does it now.

revk, 2024-06-18:
OK see it it works, we are definitely doing the state topic and the will, and so on, and HA sees available, but not sure it sees not available??? Anyone see what I have done wrong in the config?

MassiPi, 2024-06-19:
can't see your updated code, but from what you write.. did you implement a LWT or just published something? :)

revk, 2024-06-19:
I had to change the LWT that was built in to the library to be more HA friendly. I'm working on it more now.

revk, 2024-06-19:
OK should be working in next beta in a few minutes.

closed 2024-06-19 by revk

Nick-Sills, 2024-06-25:
Faikin-S3-MINI-N4-R2: 5c7de58 2024-06-24T09:55:56 S21

Not sure when this happened, but MQTT devices are now unavailable:

`homeassistant/climate/34B7DA588E44/config`
```json
  "avty_t": "Faikin/Villacana AC Back Bedroom/state",
  "avty_tpl": "{{value_json.up}}",
```

`Faikin/Villacana AC Back Bedroom/state`
```json
{
  "ts": "2024-06-25T06:53:47Z",
  "id": "34B7DA588E44",
  `"up": 134,`
  "mqtt-up": 130,
  "app": "Faikin",
  "version": "5c7de58",
  "build-suffix": "-S3-MINI-N4-R2",
  "build": "2024-06-24T09:55:56",
  "flash": 4194304,
  "rst": 3,
  "mem": 114892,
  "spi": 2090260,
  "ssid": "Silvia42",
  "bssid": "9483C4494332",
  "rssi": -58,
  "chan": 1,
  "ipv4": "192.168.100.165",
  "online": true,
  "power": false,
  "target": 21,
  "temp": 24,
  "outside": 24.5,
  "liquid": 20.5,
  "Wh": 893700,
  "fanfreq": 0,
  "comp": 0,
  "mode": "off",
  "fan": "auto",
  "streamer": false,
  "swing": "off",
  "preset": "none"
}
```

revk, 2024-06-25:
That is weird, the beta release had a slightly old library, reissuing.

Nick-Sills, 2024-06-26:
Faikin-S3-MINI-N4-R2: 67c96c7 2024-06-26T08:37:58

Working OK now. 👍   

Just an observation. We're no-longer using topic `state/ha`

`config`

```json
  "avty_t": "Faikin/Villacana AC Back Bedroom/state",
  "temp_stat_t": "Faikin/Villacana AC Back Bedroom/state",
  "curr_temp_t": "Faikin/Villacana AC Back Bedroom/state",
  "mode_stat_t": "Faikin/Villacana AC Back Bedroom/state",
```

Not a problem, just noticed that it's changed,