revk/xESP32-Faikout #362: [BUG] Faikin device not coming up. Bootloop

bajo, 2024-06-22:
**Faikin hardware**
I bought 3 of the PCBs about a year ago on Amazon UK.
2 are working fine, but this one is not responding as expected. 
I will attach pictures in a few minutes to make sure the hardware can easily be identified. 

**Daikin hardware**
The ESP32 has the marking ESP32-PICO-MINI-02 on the chip. 
Right now the hardware is only connected to a USB-Serial adapter to a Macbook. 
I spliced in a lab-bench power supply to get stable 3.3V with enough current to properly power the pcb as the logs indicated brownout detector which I was hoping to fix with a stable power. 

**Describe the bug**
I use https://web.esphome.io/ to connect to the serial console. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect the non-S21 connections on the PCB to the USB-Serial adapter. 
2. Connect to the Serial connection on https://web.esphome.io/
3. Click on Logs
4. Accesspoint from the PCB is not created.
5. CPU is reset, System boots again.
6. LED is white, but flickers when the CPU gets reset.
7. Get the logs shown below.

> [12:48:34]x@ets Jul 29 2019 12:21:46
[12:48:34]
[12:48:34]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:48:34]configsip: 271414342, SPIWP:0xee
[12:48:34]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:48:34]mode:DIO, clock div:2
[12:48:34]load:0x3fff0030,len:1336
[12:48:34]load:0x40078000,len:13840
[12:48:34]load:0x40080400,len:4
[12:48:34]ho 8 tail 4 room 4
[12:48:34]load:0x40080404,len:3036
[12:48:34]entry 0x4008058c
[12:48:35]E (1072) RevK: Not updating partition as no ota at 400000
[12:48:35]
[12:48:35]Brownout detector was triggered
[12:48:35]
[12:48:35]ets Jul 29 2019 12:21:46
[12:48:35]
[12:48:35]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:48:35]configsip: 271414342, SPIWP:0xee
[12:48:35]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:48:35]mode:DIO, clock div:2
[12:48:35]load:0x3fff0030,len:1336
[12:48:35]load:0x40078000,len:13840
[12:48:35]load:0x40080400,len:4
[12:48:35]ho 8 tail 4 room 4
[12:48:35]load:0x40080404,len:3036
[12:48:35]entry 0x4008058c
[12:48:37]E (1071) RevK: Not updating partition as no ota at 400000
[12:48:37]E (1321) task_wdt: esp_task_wdt_reset(783): task not found
[12:48:37]
[12:48:37]Brownout detector was triggered
[12:48:37]
[12:48:37]ets Jul 29 2019 12:21:46
[12:48:37]
[12:48:37]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:48:37]configsip: 271414342, SPIWP:0xee
[12:48:37]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:48:37]mode:DIO, clock div:2
[12:48:37]load:0x3fff0030,len:1336
[12:48:37]load:0x40078000,len:13840
[12:48:37]load:0x40080400,len:4
[12:48:37]ho 8 tail 4 room 4
[12:48:37]load:0x40080404,len:3036
[12:48:37]entry 0x4008058c
[12:48:38]E (1071) RevK: Not updating partition as no ota at 400000

**Expected behavior**
Accesspoint is created, so that I can connect and setup up Faikin.

**Additional context**
I tried to build a new image, so that I could flash it onto the PCB, hoping to overcome some bug, but I'm struggling with ESP-IDF on Mac due to to the fact that I can't get libpopt to be detected by CMake. 
Are there any pre-build images I can attempt to flash?

bajo, 2024-06-22:
[image]

bajo, 2024-06-22:
[image]

revk, 2024-06-22:
"Brownout detector was triggered" basically means some sort of hardware issue.

Usually means some sort of short causing the power supply to drop. Nice clear pictures, nothing obvious. But yeh, hardware problem by the sound of it.

bajo, 2024-06-28:
@revk I'm trying with a generic devboard now. 
Flashed it with the Faikin-S1 images from the release folder. My only question now is which GPIO pins to use for RX and TX? I didn't find that in any document to be honest.

revk, 2024-06-28:
Err, you can configure whatever you want. That looks like an older board, does it not say? Oh, it does 26 and 27 inverted by the look of it.

closed 2024-06-28 by revk