revk/xESP32-Faikout #365: [BUG] - {{value_json.target}} in my log continuously

1mfaasj, 2024-06-24:
**Faikin hardware**
Bought from Amazon.co.uk

**Daikin hardware**

- BRP069C4x
- connected the faikin device in the S21 port.

**Describe the bug**
my Home Assistant log is spammed continuously, literaly with the following warning message.

```
Logboek Details (WAARSCHUWING)
Logger: homeassistant.helpers.template
Bron: helpers/template.py:2629
Eerst voorgekomen: 09:58:00 (60 gebeurtenissen)
Laatst gelogd: 10:22:00

Template variable warning: 'dict object' has no attribute 'target' when rendering '{{value_json.target}}'
```

I see in MQTT explorer that "value_json.target" comes from the faikin devices. most likely it has something to do with:
`  "temp_stat_tpl": "{{value_json.target}}",
`

the full:
```
{
  "unique_id": "70041DCE89D0",
  "dev": {
    "ids": [
      "70041DCE89D0"
    ],
    "name": "FaikinZolder1",
    "sw": "4cc882f",
    "mf": "RevK",
    "cu": "http://FaikinZolder1.local/"
  },
  "icon": "mdi:thermostat",
  "~": "command/FaikinZolder1",
  "min_temp": 16,
  "max_temp": 32,
  "temp_unit": "C",
  "temp_step": 0.5,
  "temp_cmd_t": "~/temp",
  "temp_stat_t": "state/FaikinZolder1/ha",
  "temp_stat_tpl": "{{value_json.target}}",
  "curr_temp_t": "state/FaikinZolder1/ha",
  "curr_temp_tpl": "{{value_json.temp}}",
  "mode_cmd_t": "~/mode",
  "mode_stat_t": "state/FaikinZolder1/ha",
  "mode_stat_tpl": "{{value_json.mode}}",
  "payload_on": "1",
  "payload_off": "0",
  "power_command_topic": "~/power",
  "fan_mode_cmd_t": "~/fan",
  "fan_mode_stat_t": "state/FaikinZolder1/ha",
  "fan_mode_stat_tpl": "{{value_json.fan}}",
  "fan_modes": [
    "auto",
    "low",
    "lowMedium",
    "medium",
    "mediumHigh",
    "high",
    "night"
  ],
  "swing_mode_cmd_t": "~/swing",
  "swing_mode_stat_t": "state/FaikinZolder1/ha",
  "swing_mode_stat_tpl": "{{value_json.swing}}",
  "swing_modes": [
    "off",
    "H",
    "V",
    "H+V",
    "C"
  ],
  "pr_mode_cmd_t": "~/preset",
  "pr_mode_stat_t": "state/FaikinZolder1/ha",
  "pr_mode_val_tpl": "{{value_json.preset}}",
  "pr_modes": [
    "eco",
    "boost",
    "home"
  ]
}
```

Can you do something about this? or can you please explain what it means?
I think because Home Assistant doesn't do anything with the attribute, maybe it can be removed?
or convert it into something Home Assistant understands?

revk, 2024-06-24:
`state/FaikinZolder1/ha` should have `target`

What are you seeing for `state/FaikinZolder1/ha` payload?

1mfaasj, 2024-06-24:
```
{
  "online": true,
  "temp": 24,
  "outside": 22,
  "liquid": 24.5,
  "comp": 0,
  "demand": 100,
  "Wh": 268600,
  "fanfreq": 0,
  "mode": "off",
  "fan": "auto",
  "swing": "off",
  "preset": "home"
}
```

revk, 2024-06-24:
Try latest beta anyway, but that does seem odd.  I expect t it is because it is off.

Try turning on, something with a target temp, that should pick up a target temp and then be OK.

1mfaasj, 2024-06-24:
are the beta releases stable? I just switched to beta and updated.
I also turned the AC's on and sets a target temperature. It looks like the error disappeared. thanks!

closed 2024-06-24 by revk

1mfaasj, 2024-06-25:
Hi @revk I switched to beta and noticed a new setting (Set additional HA switches)
it resulted in more similar warnings in the HA log. I just played with the faikin controls but some of them stayed. because of this I switched back to normal release.

revk, 2024-06-25:
I'd recommend the beta normally, just turn off the extra ha switches.

But let me know the actual warnings and I can investigate.