revk/xESP32-Faikout #368: [BUG] - 'dict object' has no attribute

1mfaasj, 2024-06-25:
**Faikin hardware**
Bought from Amazon.co.uk

**Daikin hardware**
- BRP069C4x
- connected the faikin device in the S21 port.

**Describe the bug**
my Home Assistant log reports these warnings, very often. Similar to https://github.com/revk/ESP32-Faikin/issues/365 but now for different attributes. It began actually when you asked me to switch to beta and I enabled Additional HA switches.
now I switched back to 'normal' but the warnings stays.

```
Template variable warning: 'dict object' has no attribute 'power' when rendering '{{value_json.power}}'
Template variable warning: 'dict object' has no attribute 'streamer' when rendering '{{value_json.streamer}}'
```

I already used powerful mode and stream/filter options in the faikin interface:
[image]
but the warnings stays

when I execute
state/FaikinZolder1/ha
it gives me:
```
{
  "online": true,
  "target": 25,
  "temp": 24.5,
  "outside": 24,
  "liquid": 24.5,
  "comp": 0,
  "demand": 100,
  "Wh": 268700,
  "fanfreq": 0,
  "mode": "off",
  "fan": "auto",
  "swing": "off",
  "preset": "home"
}
```

revk, 2024-06-25:
`power` is not `powerful`

But odd. Does it all go away if you turn it on?

1mfaasj, 2024-06-25:
No, I've tried every option

revk, 2024-06-25:
- Ok what status is HA looking for?
- What does the status post have on MQTT?
- Are you on latest beta?

1mfaasj, 2024-06-25:
Hi the warning above disappeared when I clicked on some buttons, but after testing further with beta the mqtt device becomes unavailable in HA. Even after a reboot of the faikin device, HA or completely power off-on doesnt recover it.
After switching back to Normal release it is available again. I did this three times now (normal to beta and visa versa)
Any idea what is causing this?
It works when I use the Faikin interface by its IP address. Only not in HA.
No logs in HA about the unavailable state.

1mfaasj, 2024-06-25:
Looks similar https://github.com/revk/ESP32-Faikin/issues/369

revk, 2024-06-26:
I don't understand how, but the current beta release was built with an old version of the library that broke availability.

1mfaasj, 2024-06-26:
ok let me know if I can test

revk, 2024-06-26:
The beta is done, so yeh, test.

1mfaasj, 2024-06-27:
looks like the issue is resolved with the latest beta (both). thanks!

closed 2024-06-27 by 1mfaasj