revk/xESP32-Faikout #369: [BUG] Device not available in home assistant after Beta 2024-06

geni0602, 2024-06-25:
**Faikin hardware**
ESP32 own bild
**Daikin hardware**
Daikin Perferra s21 port
**Describe the bug**
After update from last Beta, the device is not available in Home Assistant. Mqtt is connected and WEB GUI is ok. Something with availability for home assistant is not correct. Update of stable version resolves the issue 
**To Reproduce**
Steps to reproduce the behavior:
update to last beta, after that the topic state/hostname/ha is not refreshing  

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

revk, 2024-06-26:
I will look again this morning.

A lot of testing was done on this, so I am rather surprised.

Hopefully a simple fix.

geni0602, 2024-06-26:
Let me know, if you need an additional debug. Home assistant version is core 2024.6.0

revk, 2024-06-26:
I don't understand how, but the current beta release was built with an old version of the library that broke availability. New beta in a few minutes. Apologies, very odd indeed.

closed 2024-06-26 by revk

geni0602, 2024-06-26:
It's looking good. Just update from beta and everything is ok.

Thanks.