revk/xESP32-Faikout #371: [BUG] fanrpm not shown in HomeAssistant

andreas-bulling, 2024-06-27:
I just noticed that the fanrpm are not properly sent to HomeAssistant. It's shown as "unknown".

[image]

Firmware: 4cc882f

revk, 2024-06-28:
Can you confirm what you see in the MQTT.

andreas-bulling, 2024-06-28:
The AC was set to cooling mode

[image]

revk, 2024-06-28:
Rather odd if seeing `fanrpm` as `0` and as an array. The array is typical for the normal `.../status` messages the Faikin sends on MQTT every minute. Also HA is looking for `fanfreq`.

Check the topic HA is looking for and post what is in that topic?

andreas-bulling, 2024-06-28:
I should add: Faiking didn't show up at first after opening MQTT Explorer. Only once I turned one of the ACs on did these values appear.

`fanfreq` is also there (didn't screengrab it).

[image]

I am afraid I don't know what you mean/what else I should check. I mean - I configured MQTT for Faikin at some point, it showed up in the MQTT integration in HA, and HA automatically converted all values into corresponding entities... So I assume HA is looking for the topics it has converted itself?!

andreas-bulling, 2024-06-28:
If I may add - and maybe you already have that: But given that you seem to have quite a number of HA users, wouldn't it make sense for you to setup a small HA system to be able to look into/debug these kind of things yourself?

Just because there seems to be quite a bit of time-consuming back and forth as users have to provide information, don't deliver the right or incomplete information, you have to ask again, there can be misunderstandings...

HA is not at all difficult to setup on a small/old RPi for example. Or even in a virtual machine.

I think this would speed up things in a few cases.

revk, 2024-06-28:
I do plan a test HA set up at some point, yet. I've been busy.

But there are so many models, my tests on my units will not get to the bottom of most issues anyway.

revk, 2024-06-28:
OK, Home Assistant makes this easy.

Firstly, it works.

[image]

But in mqtt, selecting the device, you can see MQTT info, which advises what it is looking for.

[image]

And it even shows the messages it has received...

[image]

So, maybe you can check those details for yours...

andreas-bulling, 2024-06-28:
Nice - but where do I need to go in HA to be "in mqtt" ?

revk, 2024-06-28:
Err, lets look!!!

[image]
[image]
[image]
[image]
[image]
[image]

andreas-bulling, 2024-06-28:
I've been using HA for years now and have never clicked on these three dots - thanks a lot!

Sonic-Amiga, 2024-07-03:
One of my users has seen this too... I see you have "fanfreq"; and at some point fanrpm was superseded by fanfreq. Apparently HA caches old data items internally. Consider that a HA feature. I am not using HA myself, so can't give any advice, sorry.

BTW, the newest Faikin allows the user to choose between rpm and frequency (in Hz) in advanced settings.

revk, 2024-07-04:
It could be cached in the MQTT server as a retained message - restarting MQTT may clear.

closed 2024-07-04 by revk