revk/xESP32-Faikout #375: Faikin & original wifi adapter simultaneously

funkypoly, 2024-07-02:
I am trying to find a way be able to use my faikin and the original brp069b41 adapter at the same time.
Reason is that i want to take benefit of the external BLE thermometer and also be able to control my unit from the Onecta application remotely.

FYI, my unit is an FTXM25N. 

So, i checked the board where the S21 & S403 are and it seems that there is a board (KRP067A41E) that gives the S21 port.  
You can find some pictures here : https://minisplitsystemsdirect.com/products/daikin-krp067a41e-interface-adaptor
This interface connects to a parent PCB via the S403 connector. There is a picture available in the following link : https://hvacdirect.com/media/hvac/pdf/KRP067A41E-Submittal.pdf

The only free connector i can find is the S16, as the S21 is being used by the original Daikin WLAN adapter.

Any other ideas??? Do you know anything about this S16 4pin connector?

TIA!

revk, 2024-07-03:
Why do you want to use the Daikin app? What's the advantage?

Sonic-Amiga, 2024-07-03:
This is a serial point-to-point interface; it's not intended for connecting more than two devices (AC and controller).

It's theoretically possible to implement pass-through mode for Faikin+ board (with two connectors); not sure if @revk would want to do it. Lots of work with unclear benefit.

And it's probably not (legally) possible to make Faikin compatible with Onecta because that involves a cloud, which, you know... proprietary.

funkypoly, 2024-07-04:
> Why do you want to use the Daikin app? What's the advantage?

Hi revk,

As i mentioned in my previous message, i am looking for a way to control the unit remotely.
I have seen people with different units that are able to use both devices and that's the reason i started this topic.

revk, 2024-07-04:
Ah, I see. Depending on your internet connection you may be able to set up port forwarding or some such on your router, or even a VPN. I'm somewhat used to proper IP addresses (IPv4 and IPv6) and a proper firewall so I don't have that problem.

I also think home assistant has a cloud service which may be a simpler solution.

geni0602, 2024-07-12:
I can confirm. Home assistant and MQTT works like a charm. I have four remote units connected remote to my mqtt broker. No problems at all.

funkypoly, 2024-07-18:
So... I was googling about the S16 port that can be found on the KRP067A41E board. It's a JEM-A connector

Here is some useful information : https://community.openenergymonitor.org/t/hack-my-heat-pump-and-publish-data-onto-emoncms/2551/105?page=6
Now i have to find some time to validate the information ... and do some measuring.

Sonic-Amiga, 2024-08-09:
Hello @funkypoly !

I have recently acquired this very BRP adapter for reverse engineering.
FYI, the adapter is actually not cloud-only. It implements a HTTP-based API, which is accessible on the local network. OpenHab supports it; i would expect HomeAssistant to also do so because that's also quite a large mature product.

So if you want to keep using original app, you probably don't need Faikin. @revk Sorry to say that; i understand that you probably don't want your product to be de-advertised.

In Faikin's defense i can say that it's also perfectly compatible with that API; that's why i joined the project. Thank @revk very much for great piece of software! Daikin did a real hard job to make BRP unavailable in Russia.

closed 2024-08-18 by revk