revk/xESP32-Faikout #384: [BUG] Faikin can't connect to Wifi

imdos, 2024-07-09:
**Faikin hardware**
2024-02-25
5060634238151

**Daikin hardware**
S21 lead, with dupont cables. The original lead doesn't fit. Indoor device is: FTXF35C5V1B

**Describe the bug**
I try to add the Faikin to the wifi I have for IOT devices. This is a 2,4Ghz only network. Tried both the original included firmware (Faikin-S3-MINI-N4-R2: 4945ab0 2024-02-28T15:54:01) and even flashed the latest current stable version. 

From Windows with esptool: .\esptool.exe -p COM8 write_flash 0x0 ..\..\Downloads\Faikin-S3-MINI-N4-R2-bootloader.bin 0x8000 ..\..\Downloads\partition-table.bin 0xd000 ..\..\Downloads\ota_data_initial.bin 0x10000 ..\..\Downloads\Faikin-S3-MINI-N4-R2.bin

Version is now: 4cc882f 2024-06-10

**To Reproduce**
Steps to reproduce the behavior:
The device is blinking green. And shows a wifi network. 

1. Connect to the open wifi ( Faikin-<MAC> )
2. Add all information.
3. press save

**Expected behavior**
That the device will connect to the wifi. 

**Additional context**
I see the page loading and after a while get the error: WiFi did not connect, try again. 

P.S. the wifi password is 39 characters long (upper/lower/number random).

revk, 2024-07-09:
Well, 39 characters is valid, also difficult to type accurately.

It looks like a password of the full 63 characters should be supported.

I have seen reports of issues with SSIDs containing spaces.

A good test would be to set up a simpler SSID with simpler password and confirm if that works. If it does then try and work out if it is the SSID or password that is causing the issue.

At the end of the day I cannot rule out some stupid issue in the ESP WiFI library code. If you find a limitation, it can be raised as an issue on the GitHub for ESP IDF I expect.

imdos, 2024-07-09:
I am sure the password is accurate. I just copy and paste it from my password manger (bitwarden) and therefore don't expect any problem with that. 

There isn't a space either in the SSID; since it is just "IOT2". Without the quotes of course. 

I just managed to connect to the guest network which has only 11 characters, all lowercase and numbers.   

Could it be a problem that the buffer isn't big enough to contain all the information (e.g. SSID/mqtt settings/web password/timezone).

revk, 2024-07-09:
Ah, I may have found it, D'Oh. One moment.

revk, 2024-07-09:
Yeh, it is that set up screen that does not allow more than 32 chars, my bad. Elsewhere it allows more - if you could get it on a wifi you could probably then set the correct password on normal settings pages.

But I'll issue a new beta now.

closed 2024-07-09 by revk

imdos, 2024-07-09:
Great; managed to get it on the correct wifi and set the correct timezone from the settings. But that is always cumbersome since I had to connect my laptop wired to do so.