revk/xESP32-Faikout #393: [FEATURE] Make temperature step configurable

jamofer, 2024-07-16:
**Daikin hardware**
Current temperature step:
* X50: 0.1
* S21: 0.5
* CN_WIRED: 1

**Describe the solution you'd like**
Allow to configure temperature step in the configuration page and expose it in the HA discovery response.

**Additional context**
My HVAC uses X50 with a 0.1ºC temperature step, and I always use steps of 1ºC. If I could change it through configuration, it would better suit my usage. Maybe it could be restricted to use only multiples of the current protocol.

revk, 2024-07-16:
The steps are based on what the hardware supports. Surely you just set what temp you want, and if you want whole degrees you just set whole degrees? Sorry if I am missing the point.

revk, 2024-07-16:
If we are just talking HA, it may be I could adjust - can this not be done in HA itself?

jamofer, 2024-07-16:
My current use case:
I use Faikin through MQTT in Home Assistant. It uses current MQTT discovery response for setting the temperature step for the frontend and with the _vainilla_ thermostat card you cannot customize it. So, if you click **+** and **-** buttons it will only change 0.1ºC for every click.

If you receive and update from the MQTT broker while you are increasing/decreasing it will reset to the current target temperature making it sometimes a little painful.

> If we are just talking HA, it may be I could adjust - can this not be done in HA itself?

I'm agree that it would be better HA thermostat card allowed to customize it.

MuzzaMcD, 2024-07-16:
I agree with this suggested request. In the 'Homeassistant MQTT Climate Config' sent from Faikin is the  "temp_step" value.
When temp_step was not included in the config it defaulted to 1.  When temp_step was added to the config in a recent? version of Faikin this made it a little finicky to alter the set temperature **via the slider or the plus or minus button**. Not too painful for 0.5 increments but for 0.1 increments would be quite difficult. 

I have recently set up a hack for my HA climate definition by copying (manually building) the Fakin delivered MQTT config into my MQTT yaml. 
A bit like this:

```
 climate:
    - unique_id: "ACBed3eHack"
      icon: "mdi:thermostat"
      availability_topic: "state/AC_Bed3"
      availability_template: "{{value_json.up}}"
      payload_available: true
      payload_not_available: false
      current_temperature_topic: "state/AC_Bed3"
      current_temperature_template: "{{value_json.temp}}"
      mode_state_topic: "state/AC_Bed3"
      mode_state_template: "{{value_json.mode}}"
      mode_command_topic: "command/AC_Bed3/mode"
      action_topic: "state/AC_Bed3/myHack"
      action_template: "{{value_json.action}}"
      temperature_state_topic: "state/AC_Bed3"
      temperature_state_template: "{{value_json.target}}"
      temperature_command_topic: "command/AC_Bed3/temp"
      min_temp: 12
      max_temp: 32
      temperature_unit: "C"
      temp_step: 1                 #    <-----------
      payload_on: '1'
      payload_off: '0'
      power_command_topic: "command/AC_Bed3/power"
      device:
        name: AC_Bed3e
        identifiers:
          - "ACBed3eHack"
        suggested_area: Bed3
        manufacturer: "Muzza"
```

As an aside, Im also using this hack to override the 'action' based on my interpretation of when I consider the unit in 'idle' mode. Refer 'action_topic' above ... But that's another story. :)

commit deb1d6a8 (2024-07-16): Fix #393
closed 2024-07-16 by revk