revk/xESP32-Faikout #400: [BUG] MQTT Topic for "setting"

lorbetzki, 2024-07-21:
Hello,

i'm not really sure if this is a bug, but in your advanced.md i found this section
[image]

but i cannot send setting/hostname i need to send setting/ID. If i enable "prefixhost" in the advanced settings it turns to

ID/setting, but i expect hostname/setting like the image said on the "prefixhost"-line

[image]

And now to the initial question, is it a bug or is it intentional? 

Best regards

revk, 2024-07-22:
The hostname should work. What did you set the hostname to?

I can retest here, but most of my stuff uses the hostname.

Bear in mind if prefixapp is set it is appname/hostname

lorbetzki, 2024-07-22:
HI,

I apologize, I expressed myself in a misleading way.

I can send a json like {dark: true} to hostname/setting (in my case, ac-dach/setting), but if i send without payload, i received all the existing settings to id/topic

MQTT Explorer shows
[image]

but i expected to receive the existing settings to hostname/setting

Best Regards

revk, 2024-07-23:
Ah, I see. That may be a documentation omission. That is deliberate.

Unlike `command` and `info`/`status` messages which have a clear direction, the `setting` is used to set settings, but it did not fit well to make this setting dump response something different. As such the response is the same as the setting command. Not ideal design, I agree. But only matters when two devices have the same hostname, for example, as each will see the response as address to them to change settings.

I'\ll update documentation.

closed 2024-07-23 by revk